RoboStack/ros-noetic #161: Missing icon rqt-graph on windows

npnmos, 2021-09-13:
Hello, 
I have just installed rqt-graph by running `conda install ros-noetic-rqt-graph`. The output was: 

```
(noetic) C:\Users\name>conda install ros-noetic-rqt-graph
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\giraudi\AppData\Local\miniforge3\envs\noetic

  added / updated specs:
    - ros-noetic-rqt-graph

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    graphviz-2.38              |       hfd603c8_2        29.3 MB
    pydot-1.4.1                |   py38haa95532_0          43 KB
    ros-noetic-qt-dotgraph-0.4.2|   py38h4b9bc1a_5          27 KB  robostack
    ros-noetic-qt-gui-0.4.2    |   py38h4b9bc1a_5         100 KB  robostack
    ros-noetic-rqt-graph-0.4.14|   py38h4b9bc1a_5          57 KB  robostack
    ros-noetic-rqt-gui-0.5.2   |   py38h4b9bc1a_5         124 KB  robostack
    ros-noetic-rqt-gui-py-0.5.2|   py38h4b9bc1a_5          12 KB  robostack
    ------------------------------------------------------------
                                           Total:        29.7 MB

The following NEW packages will be INSTALLED:

  graphviz           pkgs/main/win-64::graphviz-2.38-hfd603c8_2
  pydot              pkgs/main/win-64::pydot-1.4.1-py38haa95532_0
  ros-noetic-qt-dot~ robostack/win-64::ros-noetic-qt-dotgraph-0.4.2-py38h4b9bc1a_5
  ros-noetic-qt-gui  robostack/win-64::ros-noetic-qt-gui-0.4.2-py38h4b9bc1a_5
  ros-noetic-rqt-gr~ robostack/win-64::ros-noetic-rqt-graph-0.4.14-py38h4b9bc1a_5
  ros-noetic-rqt-gui robostack/win-64::ros-noetic-rqt-gui-0.5.2-py38h4b9bc1a_5
  ros-noetic-rqt-gu~ robostack/win-64::ros-noetic-rqt-gui-py-0.5.2-py38h4b9bc1a_5

The following packages will be UPDATED:

  openssl                                 1.1.1k-h2bbff1b_0 --> 1.1.1l-h2bbff1b_0

Proceed ([y]/n)? y

Downloading and Extracting Packages
ros-noetic-qt-gui-0. | 100 KB    | ############################################################################ | 100%
ros-noetic-rqt-gui-0 | 124 KB    | ############################################################################ | 100%
ros-noetic-rqt-graph | 57 KB     | ############################################################################ | 100%
pydot-1.4.1          | 43 KB     | ############################################################################ | 100%
graphviz-2.38        | 29.3 MB   | ############################################################################ | 100%
ros-noetic-qt-dotgra | 27 KB     | ############################################################################ | 100%
ros-noetic-rqt-gui-p | 12 KB     | ############################################################################ | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
```
When I run rqt, I do not see the icons:
[image]
I don't know if it is an issue with my installation or if it is a possible bug. Just wanted to let you know
Thank you

Tobias-Fischer, 2021-09-13:
Do you have any ideas @traversaro @ooeygui?

traversaro, 2021-09-13:
> 
> 
> Do you have any ideas @traversaro @ooeygui?

We had a similar issue in https://github.com/robotology/robotology-superbuild/issues/871 . However, in this case both `rqt` and `rqt_graph.exe` are installed in `Library\bin`, so no workaround should be necessary.

traversaro, 2021-09-13:
Just to avoid problems, could you try to install it in an environment in which the conda-forge channel has top priority and also `graphviz` and `pydot` are installed with conda-forge ?

Tobias-Fischer, 2021-09-13:
Agreed that the non-conda-forge packages might be an issue (especially if `qt` is also from the main channel). @npnmos please follow our instructions in the README and put robostack and conda-forge at the top of your priority list with strict priority.

npnmos, 2021-09-13:
I had no success. I have tried creating a new environment as following:
```
(base) C:\Users\giraudi\Documents\prystine>conda create -n noetic2
(base) C:\Users\giraudi\Documents\prystine>conda activate noetic2

(noetic2) C:\Users\giraudi\Documents\prystine>conda config --env --add channels conda-forge
Warning: 'conda-forge' already in 'channels' list, moving to the top

(noetic2) C:\Users\giraudi\Documents\prystine>conda config --env --add channels robostack

(noetic2) C:\Users\giraudi\Documents\prystine>conda config --env --set channel_priority strict

(noetic2) C:\Users\giraudi\Documents\prystine>conda install ros-noetic-rqt-graph
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\giraudi\AppData\Local\miniforge3\envs\noetic2

  added / updated specs:
    - ros-noetic-rqt-graph

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    apr-1.7.0                  |       he38c35c_5          47 KB  conda-forge
    boost-cpp-1.74.0           |       h5b4e17d_4        16.0 MB  conda-forge
    bzip2-1.0.8                |       h8ffe710_4         149 KB  conda-forge
    cairo-1.16.0               |    hb19e0ff_1008         2.3 MB  conda-forge
    cmake-3.21.2               |       h39d44d4_0        12.1 MB  conda-forge
    distro-1.5.0               |     pyh9f0ad1d_0          20 KB  conda-forge
    docutils-0.17.1            |   py38haa244fe_0         767 KB  conda-forge
    expat-2.4.1                |       h39d44d4_0         279 KB  conda-forge
    font-ttf-dejavu-sans-mono-2.37|       hab24e00_0         388 KB  conda-forge
    font-ttf-inconsolata-3.000 |       h77eed37_0          94 KB  conda-forge
    font-ttf-source-code-pro-2.038|       h77eed37_0         684 KB  conda-forge
    font-ttf-ubuntu-0.83       |       hab24e00_0         1.9 MB  conda-forge
    fontconfig-2.13.1          |    h1989441_1005         1.7 MB  conda-forge
    fonts-conda-ecosystem-1    |                0           4 KB  conda-forge
    fonts-conda-forge-1        |                0           4 KB  conda-forge
    freetype-2.10.4            |       h546665d_1         489 KB  conda-forge
    fribidi-1.0.10             |       h8d14728_0          63 KB  conda-forge
    getopt-win32-0.1           |       h8ffe710_0          19 KB  conda-forge
    gettext-0.19.8.1           |    h1a89ca6_1005         5.0 MB  conda-forge
    gmock-1.10.0               |       h2d74725_7         9.0 MB  conda-forge
    graphite2-1.3.13           |             1000          97 KB  conda-forge
    graphviz-2.49.0            |       hefbd956_0         1.1 MB  conda-forge
    gtest-1.10.0               |       h2d74725_7         5.6 MB  conda-forge
    gts-0.7.6                  |       h7c369d9_2         188 KB  conda-forge
    harfbuzz-2.9.1             |       hc601d6f_0         881 KB  conda-forge
    icu-68.1                   |       h0e60522_0        16.3 MB  conda-forge
    intel-openmp-2021.3.0      |    h57928b3_3372         3.0 MB  conda-forge
    jbig-2.1                   |    h8d14728_2003          45 KB  conda-forge
    lerc-2.2.1                 |       h0e60522_0         133 KB  conda-forge
    libapriconv-1.2.2          |       h8ffe710_5         719 KB  conda-forge
    libblas-3.9.0              |     11_win64_mkl         4.5 MB  conda-forge
    libcblas-3.9.0             |     11_win64_mkl         4.5 MB  conda-forge
    libdeflate-1.7             |       h8ffe710_5          61 KB  conda-forge
    libffi-3.3                 |       h0e60522_2          37 KB  conda-forge
    libgd-2.3.2                |       h138e682_0         448 KB  conda-forge
    libglib-2.68.4             |       h1e62bf3_0         3.4 MB  conda-forge
    liblapack-3.9.0            |     11_win64_mkl         4.5 MB  conda-forge
    libpng-1.6.37              |       h1d00b33_2         724 KB  conda-forge
    libtiff-4.3.0              |       h0c97f57_1         1.1 MB  conda-forge
    libwebp-1.2.1              |       h57928b3_0          11 KB  conda-forge
[... 189 lines cut ...]
  tbb                conda-forge/win-64::tbb-2021.3.0-h2d74725_0
  tinyxml2           conda-forge/win-64::tinyxml2-8.0.0-h39d44d4_1
  ucrt               conda-forge/win-64::ucrt-10.0.20348.0-h57928b3_0
  vc                 conda-forge/win-64::vc-14.2-hb210afc_5
  vs2015_runtime     conda-forge/win-64::vs2015_runtime-14.29.30037-h902a5da_5
  wheel              conda-forge/noarch::wheel-0.37.0-pyhd8ed1ab_1
  xorg-kbproto       conda-forge/win-64::xorg-kbproto-1.0.7-hcd874cb_1002
  xorg-libice        conda-forge/win-64::xorg-libice-1.0.10-hcd874cb_0
  xorg-libsm         conda-forge/win-64::xorg-libsm-1.2.3-hcd874cb_1000
  xorg-libx11        conda-forge/win-64::xorg-libx11-1.7.2-hcd874cb_0
  xorg-libxau        conda-forge/win-64::xorg-libxau-1.0.9-hcd874cb_0
  xorg-libxdmcp      conda-forge/win-64::xorg-libxdmcp-1.1.3-hcd874cb_0
  xorg-libxext       conda-forge/win-64::xorg-libxext-1.3.4-hcd874cb_1
  xorg-libxpm        conda-forge/win-64::xorg-libxpm-3.5.13-hcd874cb_0
  xorg-libxt         conda-forge/win-64::xorg-libxt-1.2.1-hcd874cb_2
[... cut ...]
  xorg-xproto        conda-forge/win-64::xorg-xproto-7.0.31-hcd874cb_1007
  xz                 conda-forge/win-64::xz-5.2.5-h62dcd97_1
  yaml               conda-forge/win-64::yaml-0.2.5-he774522_0
  zlib               conda-forge/win-64::zlib-1.2.11-h62dcd97_1010
  zstd               conda-forge/win-64::zstd-1.5.0-h6255e5f_0

Proceed ([y]/n)? y
```
When I run rqt, I don't see the icons. I confirm you that the first time I had mixed packages from the standard channel and conda-forge, but this time they are all from either robostack or conda-forge.

Tobias-Fischer, 2021-09-13:
Looking at the code: https://github.com/ros-visualization/rqt_graph/blob/60e4fa87e8490e71e3dc4a953b2c28914867aaec/src/rqt_graph/ros_graph.py#L175-L190 which uses `QIcon.fromTheme` and the QT documentation https://doc.qt.io/qt-5/qicon.html#fromTheme it states that:
```
Note: By default, only X11 will support themed icons. In order to use themed icons on Mac and Windows, you will have to bundle a compliant theme in one of your themeSearchPaths() and set the appropriate themeName().
```
I am not sure where this should happen - somewhere in the code?

Actually, I just had a look and found [this PR](https://github.com/ros-visualization/rqt_graph/pull/46). Might be worth giving a shot - @npnmos are you comfortable creating a PR here with the changes contained in that PR as a patch?

Tobias-Fischer, 2021-09-14:
Actually I wonder what happens if you `conda install -c conda-forge adwaita-icon-theme`? Does that fix the issue?

npnmos, 2021-09-14:
I am temporarily very busy these days. Maybe Friday i will let you know
whether this last suggestion works or not. If not, i will look into that PR
you mentioned before.
Thank you

Tobias-Fischer, 2021-11-22:
Closing due to inactivity - feel free to reopen @lorenzo-gi

closed 2021-11-22 by Tobias-Fischer