RoboStack/ros-noetic #162: error retrieving package ros-noetic-smach-ros-2.5.0-py38he9ab703_6.tar.bz2

tadteo, 2021-09-14:
Hi,

I am trying to install ros-noetic-desktop-full but every time I try I got the error 500.

```

https://conda.anaconda.org/robostack/linux-64/ros-noetic-smach-ros-2.5.0-py38he9ab703_6.tar.bz2

Tobias-Fischer, 2021-09-14:
Hi @tadteo - that's a weird one. I can download the file without issues. What happens if you try `wget https://conda.anaconda.org/robostack/linux-64/ros-noetic-smach-ros-2.5.0-py38he9ab703_6.tar.bz2`? Does that work?

tadteo, 2021-09-14:
Hi @Tobias-Fischer,
Thank you for the fast reply!
 
When I tried downloading that specific packet 2 hours ago was not working. I have just retried now and the installation worked without a problem.
Probably a temporary server error.
 
Thank you again for your time and wonderful project!

closed 2021-09-14 by tadteo