RoboStack/ros-noetic #164: Turtlebot3 Teleop

robogeekcanada, 2021-09-16:
I started to play around with RoboStack. I think is very promising. The ROS installation was not very straight forward in Windows until I read carefully to use command prompt. I was also able to run `turtlesim` without much problems, however running into issues with turtlebot3. I am able to launch gazebo, start the teleop node, check that is publishing correctly, but turtebot3 doesn't move in Gazebo. Unfortunately there is no error I can share, it just doesn't work. If anyone has come across this problem, open to suggestions. Alternatively if there is tutorial to follow or things to try I'd appreciate it. Thank you

Tobias-Fischer, 2021-09-16:
Hi @robogeekcanada - any chance you could try the same in Linux and report whether it works there? Just trying to pinpoint down the issue.

robogeekcanada, 2021-09-16:
I will definitely try. Thank you for prompt response.

robogeekcanada, 2021-09-16:
It works! :)

I started from scratch again:

```
RoboStack (for ROS Noetic)
$ conda create -n robostackenv python=3.8
$ conda config --env --add channels conda-forge
$ conda config --env --add channels robostack
$ conda config --env --set channel_priority strict

$ conda install -c robostack ros-noetic-desktop-full
# on Windows, install Visual Studio 2017 or 2019 with C++ support 
# see https://docs.microsoft.com/en-us/cpp/build/vscpp-step-0-installation?view=msvc-160

$ conda install vs2019_win-64
# reload environment to activate required scripts before running anything
# on Windows, please restart the Anaconda Prompt / Command Prompt!
$ conda deactivate robostackenv
$ conda activate robostackenv

Turtlebot3 installation:
$ conda install -c robostack ros-noetic-turtlebot3
$ conda install -c robostack ros-noetic-turtlebot3-gazebo
$ conda install -c robostack ros-noetic-turtlebot3-fake
```

Notice that the installation to work:
$ conda install -c robostack ros-noetic-desktop-full

For some reason yesterday it was working `conda install ros-noetic-desktop-full but not anymore`. Very excited that this works in Windows. I will still try in Ubuntu.

closed 2021-09-16 by robogeekcanada