RoboStack/ros-noetic #165: Issues with RViz

robogeekcanada, 2021-09-16:
When trying to load `turtlebot3_fake` in `rviz`, the meshes are transparent. I read the other issues in this repo and tried:

`mamba upgrade ros-noetic-rviz`

but no success.  This is what I get:

[image]

I also noticed, when Rviz opens it opens a small windows, so I presume this issue might have to be related to OGRE.

[image]

When I tried to run SLAM with gmapping, there are more errors as it looks a few dlls are not working. However, I want to make sure I get this issue sorted first, then work on the dlls.

Tobias-Fischer, 2021-09-16:
Could you please provide the output of `conda list`?

robogeekcanada, 2021-09-16:
```
# packages in environment at C:\ProgramData\Miniconda3\envs\robostackenv:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        41.0                 h57928b3_0    conda-forge
apr                       1.7.0                he38c35c_5    conda-forge
assimp                    5.0.1                hc2aa0de_6    conda-forge
atk-1.0                   2.36.0               h7222f49_4    conda-forge
bcrypt                    3.2.0            py38h294d835_1    conda-forge
boost                     1.74.0           py38h1266d08_3    conda-forge
boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
bullet                    3.17                 h57928b3_0    conda-forge
bullet-cpp                3.17                 h60cbd38_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
catkin_pkg                0.4.23             pyhd8ed1ab_1    conda-forge
certifi                   2021.5.30        py38haa244fe_0    conda-forge
cffi                      1.14.6           py38hd8c33c5_1    conda-forge
cmake                     3.21.2               h39d44d4_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h4324990_2    conda-forge
cryptography              3.4.7            py38hd7da0ea_0    conda-forge
curl                      7.79.0               h789b8ee_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
docutils                  0.17.1           py38haa244fe_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.9                h8d14728_0    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flann                     1.9.1             h7dfa0ef_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.5.0               h83ca5bb_1    conda-forge
gdk-pixbuf                2.42.6               h1c5aac7_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1006    conda-forge
glib                      2.68.4               h0e60522_1    conda-forge
glib-tools                2.68.4               h0e60522_1    conda-forge
gmock                     1.10.0               h2d74725_7    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  2.49.0               hefbd956_0    conda-forge
gtest                     1.10.0               h2d74725_7    conda-forge
gtk3                      3.24.29              h9a703b0_1    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
harfbuzz                  2.9.1                hc601d6f_0    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
hicolor-icon-theme        0.17                 h57928b3_2    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
intel-openmp              2021.3.0          h57928b3_3372    conda-forge
jasper                    2.0.14               h77af90b_2    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsoncpp                   1.9.4                h2d74725_3    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kiwisolver                1.3.2            py38hbd9d945_0    conda-forge
[... 397 lines cut ...]
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                h301d43c_0    conda-forge
xorg-libxfixes            5.0.3             h301d43c_1004    conda-forge
xorg-libxi                1.7.10               h301d43c_0    conda-forge
xorg-libxinerama          1.1.4             hfa435f1_1001    conda-forge
xorg-libxpm               3.5.13               h301d43c_0    conda-forge
[... cut ...]
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

Tobias-Fischer, 2021-09-16:
The output looks all fine.

Could you please confirm the exact steps you take to start the simulation? It would be great if you could add all steps you take, as well as the output in the terminal window(s).

On Linux everything works fine. I am a bit suspicious about the transform errors that one can see in your screenshot; I think they are the issue (they do not occur for me). What happens if you change the "Fixed Frame" in the Global Options from `base_footprint` to `base_link`?

robogeekcanada, 2021-09-16:
Thank you for helping! 2 errors I realized I made:

1. I didn't launch `roslaunch turtlebot3_bringup turtlebot3_remote.launch` so no `/tf`
2. As you mentioned base_footprint was selected.
[image]

Working now :) No errors either when launching. I am wondering now if the problem was because I had Gazebo open at the same time, I will verify this as well. 

On a separate topic, I tried to install but it's complaining incompatibility with Python 3.8 and needs 3.6
 $ conda install -c robostack ros-noetic-laser-proc

If ok with you, let's close this issue.

Tobias-Fischer, 2021-09-16:
Great! Let's close this. Please open a separate issue re ros-noetic-laser-proc.

Tobias-Fischer, 2021-09-16:
PS: The best way of launching should be `roslaunch turtlebot3_fake turtlebot3_fake.launch` as far as I know.

closed 2021-09-16 by Tobias-Fischer

Tobias-Fischer, 2021-09-16:
I just rebuild `ros-noetic-laser-proc`, you should be able to install it in an hour or so.

robogeekcanada, 2021-09-18:
Thank you @Tobias-Fischer.  I tested the installation of `ros-noetic-laser-proc` in Windows at no issues.