RoboStack/ros-noetic #167: Turtlebot3 SLAM issues Windows

robogeekcanada, 2021-09-18:
When testing Turtlebot3 SLAM a few errors pop up:

```
$ set TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_slam turtlebot3_slam.launch
```
[image]

[image]

[image]

[image]

In summary the following `dlls` not found:

utils.dll, sensor_range.dll, sensor_odometry.dll, scanmatcher.dll

robogeekcanada, 2021-09-18:
Doing a search in my directory the dlls are installed in 2 places:

```
C:\ProgramData\Miniconda3\envs\robostackenv\Library\lib\openslam_gmapping

C:\ProgramData\Miniconda3\pkgs\ros-noetic-openslam-gmapping-0.2.1-py38h4b9bc1a_5\Library\lib\openslam_gmapping
```

So for some reason ROS is not detecting them

robogeekcanada, 2021-09-18:
Not sure this is the correct way to fix the problem but it works 😄. I copied the missing dlls into the `bin` folder of my environment.

`C:\ProgramData\Miniconda3\envs\robostackenv\Library\bin`

Tobias-Fischer, 2021-09-18:
That's a bug in openslam_gmapping. Let's repopen to keep track of it.

Tobias-Fischer, 2021-09-19:
Fixed thanks to https://github.com/ros-perception/openslam_gmapping/pull/34

closed 2021-09-19 by Tobias-Fischer