RoboStack/ros-noetic #173: Cannot find ros-noetic-vrpn-client-ros

benjaminabruzzo, 2021-10-01:
Installation was mostly seamless, but this particular package is missing.  Others work (such as ros-noetic-std-msgs, which was installed without issue).  Is there any way to add additional packages to the https://conda.anaconda.org/robostack/linux-64 ?

`
PackagesNotFoundError: The following packages are not available from current channels:

  - ros-noetic-vrpn-client-ros

Current channels:

  - https://conda.anaconda.org/robostack/linux-64
  - https://conda.anaconda.org/robostack/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://conda.atoti.io/linux-64
  - https://conda.atoti.io/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
`

Tobias-Fischer, 2021-10-02:
Hi, please check out https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md#adding-new-packages-via-pull-requests

benjaminabruzzo, 2021-10-03:
Thanks, working on it.  

The instructions failed on the step 'Run vinca to generate the recipe by executing vinca --multiple'

I'm looking at the patch process now, but I'm not familiar with what I need to put into the patch file.

wolfv, 2021-10-03:
It's enough to send a PR with your changes and the CI should hopefully take care of it :)

Tobias-Fischer, 2021-11-22:
Closing due to inactivity. Feel free to reopen if you have questions @benjaminabruzzo

closed 2021-11-22 by Tobias-Fischer