RoboStack/ros-noetic #175: Error when running catkin_make where 'ros/ros.h' is not found

cyuanqi920, 2021-10-05:
Hi, I have installed ros with Roscore running with no issues.
System:
OSX Big Sur

When compiling the package https://github.com/f1tenth/f1tenth_simulator here, it says
 'ros/ros.h' not found, is there any resolve for this?

[image]

cyuanqi920, 2021-10-10:
Hey Tobias, I managed to run the node after doing what you said, but there's another error now. By any chance do you know how I can solve it? What happens was the RVIZ visualiser wasn't working properly.
[image]
[image]

Tobias-Fischer, 2021-10-10:
What's the output of conda list?

cyuanqi920, 2021-10-10:
this is a really long list..
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cyuanqi920, 2021-10-10:
I tried running 'rosrun rviz rviz' and things work fine, could it be a problem of https://github.com/f1tenth/f1tenth_simulator, something inside the config file of their rviz which is incompatible to Mac?

the file path to the rviz config file is f1tenth_simulator/launch/simulator.rviz

Tobias-Fischer, 2021-10-10:
Hi @cyuanqi920, yes, it could well be that the config is somehow incompatible, but I don't know enough about rviz. I suggest to start from a "plain" rviz view and then add the display tools that are listed in https://github.com/f1tenth/f1tenth_simulator/blob/master/launch/simulator.rviz one-by-one manually.