RoboStack/ros-noetic #176: error when sourcing custom ws

schnellerblitz99, 2021-10-11:
Hey,

whenever I want to source my custom workspace (I want to import a custom ros message in my python script), the program has trouble to import any packages already installed in my conda env (tensorflow in my case). Without sourcing it works fine but then it doesnt find my custom message types.

You have an idea how to solve this issue?

Thanks for sharing!

schnellerblitz99, 2021-10-15:
Hey, sorry for the late reply. But thanks for your help! Indeed I was building my workspace outside of my conda environment. Rebuilding it solved the issue.
Thanks, great project!

Karthikeyanc2, 2021-11-22:
Hi @schnellerblitz99 ,

I am facing the same problem.

Following are the commands that I used:
```
conda activate conda_ros
catkin_make
source ~/devel/setup.bash
roslaunch all_sensors all_sensors.launch
```

Here conda_ros is the environment with ROS from robostack.
all_sensors and custom_messages are my packages that I just built.

I am able to run the nodes but the nodes are not able to find the custom_messages package. But rospack find is able to find them after sourcing. 

Also note, I do not even have a ros environment outside this conda_ros, so there is no way things are getting confused inside. I am obviously missing something here. Can you please help me out?

Thanks,
Karthik

closed 2021-11-22 by Tobias-Fischer