RoboStack/ros-noetic #184: Installing ros-noetic-rosbridge-server misses bson dependency

ruben-arts, 2021-11-02:
After installing rosbridge-server with Conda, you will get issues with the bson dependency.

When trying to run the rosbridge-server you will be missing the bson module. 
<details>
  <summary>Error log:</summary>
  
```
 Log opened.
 Traceback (most recent call last):
   File "/home/mamba/bson-test/.env/lib/rosbridge_server/rosbridge_websocket", line 48, in <module>
     from rosbridge_server import ClientManager
   File "/home/mamba/bson-test/.env/lib/python3.8/site-packages/rosbridge_server/__init__.py", line 2, in <module>
     from .websocket_handler import RosbridgeWebSocket
  File "/home/mamba/bson-test/.env/lib/python3.8/site-packages/rosbridge_server/websocket_handler.py", line 49, in <module>
     from rosbridge_library.rosbridge_protocol import RosbridgeProtocol
   File "/home/mamba/bson-test/.env/lib/python3.8/site-packages/rosbridge_library/rosbridge_protocol.py", line 34, in <module>
     from rosbridge_library.protocol import Protocol
   File "/home/mamba/bson-test/.env/lib/python3.8/site-packages/rosbridge_library/protocol.py", line 43, in <module>
     from rosbridge_library.util import json, bson
   File "/home/mamba/bson-test/.env/lib/python3.8/site-packages/rosbridge_library/util/__init__.py", line 19, in <module>
     import bson
ModuleNotFoundError: No module named 'bson'
```
</details>

This [issue](https://github.com/RobotWebTools/rosbridge_suite/issues/198) is known and the fix would be installing the `pymongo` library.

So the workaround for now is installing pymongo : 
```
pip install pymongo
# OR 
conda install -c conda-forge pymongo
```

Could this be added as a dependency?

wolfv, 2021-11-02:
Hi, thanks for discovering this! Yes, indeed, that seems like the proper fix!

We can add dependencies in `patch/dependencies.yaml` file. Do you want to send a PR?

ruben-arts, 2021-11-02:
Oh after making that PR I see I miss read your suggestion. Would it be enough to add a instance of rosbridge-library to the `dependencies.yaml` ?

wolfv, 2021-11-02:
yes, just adding that to dependencies.yaml should be enough. I think your way also unfortunately doesn't work because we generate the recipes _before_ we're patching them :)

PR #186 (2021-11-02): patch: add pymongo as dep of rosbridge_library
This patch should fix the issues that come from having the wrong bson module.

I added it as a Run dependency as it should be installed when running the module but not sure if that is enough? Reading [this](https://docs.conda.io/projects/conda-build/en/latest/resources/define-metadata.html) still confuses me too much.

I'm not sure how to test this locally? Very willing to do so if anyone could explain!

closes #184
closed 2021-11-02 by Tobias-Fischer