RoboStack/ros-noetic #187: Rviz not starting

daniele-gusmini, 2021-11-08:
When I launch Rviz from terminal nothing happen, there is no new window only an exec saying "Not responding"
In terminal it shows:

[ INFO] [1636370703.034942000]: rviz version 1.14.10
[ INFO] [1636370703.034974000]: compiled against Qt version 5.12.9
[ INFO] [1636370703.034982000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1636370703.064988000]: Forcing OpenGl version 0.

I am on macOS Monterey

traversaro, 2021-11-08:
Hi @daniele-gusmini, can you provide the output of the `conda list` and `conda info` commands? Thanks!

daniele-gusmini, 2021-11-08:
Of course, here `conda list`

`# Name                    Version                   Build  Channel
aom                       3.2.0                he49afe7_2    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.0.1                h1224e73_6    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py38h96a0964_2    conda-forge
boost                     1.74.0           py38hb0f0857_4    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_0    conda-forge
certifi                   2021.10.8        py38h50d1736_1    conda-forge
cffi                      1.15.0           py38h1a44b6c_0    conda-forge
cmake                     3.21.3               hdbd2f3e_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
cryptography              35.0.0           py38h56c4533_2    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               ha13b53f_2    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.18             py38h50d1736_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.1                he49afe7_0    conda-forge
ffmpeg                    4.4.1                h79e7b16_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               h7d96ac2_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.6               h2e6141f_0    conda-forge
gettext                   0.19.8.1          hd1a6beb_1008    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
glib                      2.70.0               hcf210ce_1    conda-forge
glib-tools                2.70.0               hcf210ce_1    conda-forge
gmock                     1.10.0               h926bf3e_7    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.6.13               h756fd2b_1    conda-forge
gpgme                     1.16.0               he49afe7_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  2.49.3               h77de9ca_0    conda-forge
gtest                     1.10.0               h926bf3e_7    conda-forge
gtk2                      2.24.33              h675d97a_1    conda-forge
gts                       0.7.6                hccb3bdf_2    conda-forge
harfbuzz                  2.9.1                h159f659_1    conda-forge
hdf5                      1.10.6          nompi_hc5d9132_1114    conda-forge
icu                       68.2                 he49afe7_0    conda-forge
ilmbase                   2.5.5                hfab91a5_0    conda-forge
jasper                    1.900.1           h636a363_1006    conda-forge
jbig                      2.1               h0d85af4_2003    conda-forge
jpeg                      9d                   hbcb3906_0    conda-forge
jxrlib                    1.1                  h35c211d_2    conda-forge
kernel-headers_linux-64   2.6.32              he073ed8_15    conda-forge
kiwisolver                1.3.2            py38h12bbefe_1    conda-forge
krb5                      1.19.2               hcfbf3a7_3    conda-forge
lame                      3.100             h35c211d_1001    conda-forge
lcms2                     2.12                 h577c468_0    conda-forge
lerc                      3.0                  he49afe7_0    conda-forge
libapr                    1.7.0                h0d85af4_5    conda-forge
libapriconv               1.2.2                h0d85af4_5    conda-forge
libaprutil                1.6.1                h664449b_5    conda-forge
[... 342 lines cut ...]
 populated config files : /Users/danielegusmini/Developer/Environments/miniforge3/.condarc
                          /Users/danielegusmini/Developer/Environments/miniforge3/envs/RosNoetic/.condarc
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.9.7.final.0
       virtual packages : __osx=12.0.1=0
                          __unix=0=0
                          __archspec=1=x86_64
[... cut ...]
          package cache : /Users/danielegusmini/Developer/Environments/miniforge3/pkgs
                          /Users/danielegusmini/.conda/pkgs
       envs directories : /Users/danielegusmini/Developer/Environments/miniforge3/envs
                          /Users/danielegusmini/.conda/envs
               platform : osx-64
             user-agent : conda/4.10.3 requests/2.26.0 CPython/3.9.7 Darwin/21.1.0 OSX/12.0.1
                UID:GID : 501:20
             netrc file : None
           offline mode : False
`

bart02, 2021-11-09:
Confim! 
macOS Monterey also :)

bart02, 2021-11-09:
@daniele-gusmini check message above

Tobias-Fischer, 2021-11-09:
Ah gosh, this is giving me a headache ;). Not sure what's causing this .. I can't see any relevant changes between 1.14.8 and 1.14.10. Also nothing has changed in the build of the packages as far as I can tell ..

Any ideas @wolfv?

bart02, 2021-11-09:
Maybe there are some relevant changes in dependencies because many packages were reinstalled while rviz downgrading.

traversaro, 2021-11-10:
@bart02 can you provide (`conda list`) the working environment? Thanks!

daniele-gusmini, 2021-11-10:
Thank you everyone, 
I have solved downgrading the version

mfkenson, 2021-11-10:
+1
I experienced the same on MacOS 12.0.1 (21A559).

`mamba install ros-noetic-rviz==1.14.8` works.

mfkenson, 2021-11-10:
> @bart02 can you provide (`conda list`) the working environment? Thanks!

after downgrade to ros-noetic-rviz==1.14.8
```
# packages in environment at /opt/anaconda3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.0.1                h1224e73_6    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py38h96a0964_2    conda-forge
boost                     1.74.0           py38hb0f0857_4    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
bzip2                     1.0.8                hc929b4f_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.3.0                h0d85af4_0    conda-forge
ca-certificates           2021.10.26           hecd8cb5_2  
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_0    conda-forge
catkin_tools              0.7.1              pyhd8ed1ab_0    conda-forge
cctools                   949.0.1             hd9211c8_21    conda-forge
cctools_osx-64            949.0.1             h6407bdd_21    conda-forge
certifi                   2021.10.8        py38h50d1736_1    conda-forge
cffi                      1.14.6           py38h9688ba1_0    conda-forge
clang                     11.1.0               h694c41f_1    conda-forge
clang-11                  11.1.0          default_he082bbe_1    conda-forge
clang_osx-64              11.1.0               hb91bd55_4    conda-forge
clangxx                   11.1.0          default_he082bbe_1    conda-forge
clangxx_osx-64            11.1.0               h7e1b574_4    conda-forge
cmake                     3.5.0                         3    conda-forge
compiler-rt               11.1.0               h654b07c_0    conda-forge
compiler-rt_osx-64        11.1.0               h8c5fa43_0    conda-forge
compilers                 1.3.0                h694c41f_0    conda-forge
console_bridge            1.0.1                ha1b3eb9_0    conda-forge
cryptography              35.0.0           py38h56c4533_2    conda-forge
cxx-compiler              1.3.0                h940c156_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               ha13b53f_2    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.18             py38h50d1736_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.1                he49afe7_0    conda-forge
ffmpeg                    4.3.2                h4dad6da_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.3.0                h6818c3e_0    conda-forge
freeimage                 3.18.0               h446b61c_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.6               h2e6141f_0    conda-forge
gettext                   0.19.8.1          h7937167_1005    conda-forge
gfortran                  9.3.0               h768ea0c_14    conda-forge
gfortran_impl_osx-64      9.3.0               h9cc0e5e_23    conda-forge
gfortran_osx-64           9.3.0               h18f7dce_14    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
glib                      2.68.4               he49afe7_0    conda-forge
glib-tools                2.68.4               he49afe7_0    conda-forge
gmock                     1.10.0               h926bf3e_7    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.6.13               h756fd2b_1    conda-forge
gpgme                     1.15.1               he49afe7_0    conda-forge
graphite2                 1.3.13            h12caacf_1001    conda-forge
graphviz                  2.49.0               h77de9ca_0    conda-forge
[... 321 lines cut ...]
unixodbc                  2.3.9                hd863c23_0    conda-forge
urdfdom                   2.3.5                h940c156_0    conda-forge
urdfdom_headers           1.0.5                h879752b_2    conda-forge
wheel                     0.37.0             pyhd3eb1b0_1  
x264                      1!161.3030           h0d85af4_1    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h01d97ff_0    conda-forge
xorg-libsm                1.2.3             h1de35cc_1000    conda-forge
[... cut ...]
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.5                h1de35cc_0  
yaml                      0.2.5                haf1e3a3_0    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zlib                      1.2.11               h1de35cc_3  
zstd                      1.5.0                h582d3a0_0    conda-forge
zziplib                   0.13.69              hbbe82c9_1    conda-forge
```

Tobias-Fischer, 2022-02-02:
Could someone please try with the latest build (needs new environment with python=3.9 @mfkenson @bart02 @daniele-gusmini)?

Tobias-Fischer, 2022-02-02:
It works fine on my M1 - closing here unless I hear otherwise.

closed 2022-02-02 by Tobias-Fischer

ps2575, 2022-04-24:
I got same ERROR message on rviz.
i'm using M1 Macbook / docker - ubuntu:20.04
and cannot run rviz... 
Is there any solution?? plz,,,

traversaro, 2022-04-24:
@ps2575 are you using apt ROS packages or conda/RoboStack ROS packages? This issue tracker is related only to  conda/RoboStack ROS packages.

ps2575, 2022-04-24:
> 

i don' think so (just a beginner of ROS)
i set ubuntu container on my Docker, installed ROS,

ps2575, 2022-04-24:
> using

> @ps2575 are you using apt ROS packages or conda/RoboStack ROS packages? This issue tracker is related only to conda/RoboStack ROS packages.

> 
@traversaro 
i don' think so (just a beginner of ROS)
i set ubuntu container on my Docker, installed ROS,
and other gui runs well using x11 (such as turtlesim)
[image]

Tobias-Fischer, 2022-04-24:
How did you install ROS?

ps2575, 2022-04-24:
> How did you install ROS?
on my ubuntu 20.04 contatiner i typed the line below 
(i've done something more following my lecture note, but below is the one i think you wanna know)

sudo apt install ros-noetic-desktop-full

ps2575, 2022-04-24:
i got 2 Error to solve my problem 
[image]

there are some replies that to remove one below 
[image]

should i try this ?

Tobias-Fischer, 2022-04-24:
Apologies but this repository is only concerned with the RoboStack installations of ROS - see the readme. You can try to install ROS in this way on your Mac without using a virtual machine. Or please reach out to the ROS forums for help.

ps2575, 2022-04-24:
> 죄송하지만 이 리포지토리는 ROS의 RoboStack 설치에만 관련되어 있습니다. 읽어보기를 참조하세요. 가상 머신을 사용하지 않고 이러한 방식으로 ROS를 Mac에 설치할 수 있습니다. 또는 도움이 필요하면 ROS 포럼에 연락하십시오.
@Tobias-Fischer 
umm,, do you mean rviz things cannot be run on mac docker setting?
if so, should i have to try Parallels or Robostack(?) to run rviz (or any other gui acceleration of ROS) ? 

anyway, Thanks for answering,,I have a ROS practice test tomorrow and I was so hopeless..

Tobias-Fischer, 2022-04-25:
I don't know about Docker or parallels.

You can try RoboStack - it should work as explained in the readme :)

soumith, 2022-05-09:
I can confirm that out of the box on OSX, the latest rviz package is hanging without output, and `1.14.8` seems to be working.
I can provide any debug, output or other details to resolve this

rsrinivasan1, 2022-05-10:
Hi, how do I downgrade rviz?

Tobias-Fischer, 2022-05-10:
mamba install ros-noetic-rviz=1.14.8

rsrinivasan1, 2022-05-10:
Thanks, I tried this with conda, but got the following error:

PackagesNotFoundError: The following packages are not available from current channels:

  - ros-noetic-rviz==1.14.8

I ended up manually installing the file and putting it in the right folder. Works now!

ravi1754, 2022-09-28:
@rsrinivasan1 
Facing the same issue , can you tell what do you mean by  installed manually and putting in the right folder?

traversaro, 2023-05-17:
Hello @fedor-chervinskii, can you open a new issue following the issue template? Thanks!

fedor-chervinskii, 2023-05-17:
@traversaro here you go, thanks for trying to figure it out!

yeyanan93, 2023-07-16:
> > 
> 
> @traversaro
> 
> i don' think so (just a beginner of ROS) i set ubuntu container on my Docker, installed ROS, and other gui runs well using x11 (such as turtlesim) [image]

@ps2575    Do you have solve the problem?

traversaro, 2023-07-16:
> > > 
> > 
> > @traversaro
> > 
> > i don' think so (just a beginner of ROS) i set ubuntu container on my Docker, installed ROS, and other gui runs well using x11 (such as turtlesim) [image]
> 
> @ps2575    Do you have solve the problem?

Hello @yeyanan93 , can you open a new issue following the template? Thanks!