RoboStack/ros-noetic #188: Rviz not opening

vikramadityafcb, 2021-11-12:
I'm currently operating a macOS Big Sur 11.5.2
When I try to open rviz I keep running into this issue: 

[ INFO] [1636746775.697671000]: rviz version 1.14.8
[ INFO] [1636746775.697739000]: compiled against Qt version 5.12.9
[ INFO] [1636746775.697749000]: compiled against OGRE version 1.10.12 (Xalafu)

And I get the following pop-up error: Could not contact ROS master at [http://localhost:11311], retrying...

How do I solve this? I already tried conda install ros-noetic-rviz==1.14.8 as mentioned in a previous thread.

traversaro, 2021-11-12:
Are you running roscore before runing rviz? That error says that for some reason roscore is not reachable.

vikramadityafcb, 2021-11-12:
I am

traversaro, 2021-11-12:
Is roscore starting correctly or it is exiting immediately? Are other ROS commands just as `rostopic list` working correctly or not?

vikramadityafcb, 2021-11-12:
[image]

They aren't, my bad. How do I resolve this?

traversaro, 2021-11-12:
Just to clarify, `rostopic list` should run in a separate terminal, not in the one in which you launched `roscore`.

vikramadityafcb, 2021-11-12:
Hi, 
Sorry for the dumb questions. I'm just starting out with ROS. I went to complete basics and tried to setup the environment using 
(robostackenv) adi@nichols-ve703-00735 ~ % source /Users/adi/opt/miniconda3/envs/robostackenv/setup.bash

However, I get the following error: /Users/adi/opt/miniconda3/envs/robostackenv/setup.bash:.:8: no such file or directory: /Users/adi/setup.sh

To check this, I ran 
(robostackenv) adi@nichols-ve703-00735 ~ % printenv | grep ROS
which outputted 
ROS_ROOT=/Users/adi/opt/miniconda3/envs/robostackenv/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
ROS_DISTRO=noetic
ROS_ETC_DIR=/Users/adi/opt/miniconda3/envs/robostackenv/etc/ros
ROS_PACKAGE_PATH=/Users/adi/opt/miniconda3/envs/robostackenv/share
ROS_PYTHON_VERSION=3.8
ROS_VERSION=1
ROS_OS_OVERRIDE=robostack:osx

Just a bit confused why this isn't working. Have been sitting around this basic issue for about 3 hours now. :(

Thank you so much for your help though!! Really appreciate it!

Tobias-Fischer, 2021-11-12:
There is no need to `source /Users/adi/opt/miniconda3/envs/robostackenv/setup.bash`.

Terminal 1:
```
conda activate robostackenv
roscore
````

Terminal 2:
```
conda activate robostackenv
rviz
```

Should do the job.

closed 2021-11-14 by vikramadityafcb

alex-lechner, 2023-04-26:
@Tobias-Fischer I tried opening rviz with 

```sh
conda activate robostackenv
rviz
```

but I keep getting the error 

```sh
dyld[13517]: symbol not found in flat namespace '__Py_NoneStruct'
[1]    13517 abort      rviz
```

I could not find any solution in the issue section of the repo (neither closed nor open issues) and Google & ChatGPT also weren't helpful.

I even tried to set the environment variable `DYLD_LIBRARY_PATH` with
```sh
export DYLD_LIBRARY_PATH=/Users/me/miniconda3/envs/robostackenv/lib/python3.9
```

but that did not fix it.

I'm on MacOSX 13.3.1 and have followed the installation instructions in your `README`.

Do you have any ideas about what might be missing?

Tobias-Fischer, 2023-04-26:
Could you please create a new issue and fill the issue template? I have an idea but want to see the versions that you are using.

thanks, Tobi

angsylvest, 2023-04-26:
> ```shell
> symbol not found in flat namespace '__Py_NoneStruct'
> ```

I am also experiencing the same issue.