RoboStack/ros-noetic #189: sudo: apt-get: command not found

vikramadityafcb, 2021-11-14:
Newbie trying to learn ROS. Tried running `$ sudo apt-get install ros-noetic-ros-tutorials` from [http://wiki.ros.org/ROS/Tutorials/NavigatingTheFilesystem](http://wiki.ros.org/ROS/Tutorials/NavigatingTheFilesystem) but it resulted in the following error: 

sudo: apt-get: command not found

Screenshots attached below for your reference.

Thank you so much for helping out!!

[image]
[image]

wolfv, 2021-11-14:
Apt get is the package manager on Ubuntu. Since you are using macOS and robostack you should use mamba or conda. The command should otherwise look the same:

Mamba install ros-noetic-ros-tutorials

vikramadityafcb, 2021-11-14:
Thank you so much! Was stuck on this for a long time 😅. I really really appreciate that you took out time to answer such a simple question!

closed 2021-11-15 by vikramadityafcb