RoboStack/ros-noetic #190: Encountered problems while solving:   - nothing provides requested ros-noetic-desktop

RishabhGarg108, 2021-11-15:
I have a M1 macbook air (MacOS 12.0.1). I was following the installation instructions and in the following step
```
mamba install ros-noetic-desktop
```
I encountered the error
```
Looking for: ['ros-noetic-desktop']

robostack/noarch         [====================] (00m:00s) Done
robostack/osx-arm64      [====================] (00m:00s) Done
conda-forge/osx-arm64    [====================] (00m:01s) Done
conda-forge/noarch       [====================] (00m:06s) Done

Pinned packages:
  - python 3.8.*

Encountered problems while solving:
  - nothing provides requested ros-noetic-desktop
```
I seek some help with the installation. Thanks!

traversaro, 2021-11-15:
Hi @RishabhGarg108 ! Not all packages are available on all platforms, you can see which packages are available on M1 aka `osx-arm64` on https://robostack.github.io/noetic.html . In particular, on M1 several packages that depend on qt are currently missing, as qt is not available on M1 on conda-forge.

traversaro, 2021-11-15:
In the meanwhile, if you want to just test ROS's command line packages, you can try to install `ros-noetic-ros-base`. An alternative is also to run `os-64` packages under Rosetta2.

Tobias-Fischer, 2021-11-15:
Hi @RishabhGarg108,
Indeed `ros-noetic-desktop` is not yet natively available on M1 as @traversaro said. On the conda-forge side we are very close to getting `qt` working for the M1 though, it should just be a matter of another week or two - please stay tuned.

For instructions to use Rosetta, please see https://github.com/RoboStack/ros-noetic/issues/156

Please also watch https://github.com/RoboStack/ros-noetic/issues/156 to get updates - closing this issue here.

closed 2021-11-15 by Tobias-Fischer