RoboStack/ros-noetic #191: rtabmap modelets?

nosense345, 2021-11-15:
Taking first steps with ROS using Robostack (I don't necessarily know what I'm doing but I'm ambitious)

I've installed ros-noetic-rtabmap 0.20.9 using mamba.
I would like to use the rtabmap_ros/point_cloud_xyz nodelet but that type turns out to be undeclared.

Neither rosrun nor me (looking through the files on my system) can find these nodelets (the source is here: https://github.com/introlab/rtabmap_ros/tree/master/src/nodelets). I _think_ that maybe they should go into  lib/librtabmap_plugins.

How can I get access to these nodelets?

Thank you for your help,
Klaas

Tobias-Fischer, 2021-11-15:
Hi Klaas,
Many thanks for opening this issue, and all the best for your ROS endeavours. We're glad you've chosen RoboStack to experiment around :).

For these nodelets, you need to `mamba install ros-noetic-rtabmap-ros` (mind the `-ros` at the end). Give it a go and let us know if that fixes your issues.

Best, Tobi

nosense345, 2021-11-22:
I get the following output from mamba:

Pinned packages:
  - python 3.8.*

Encountered problems while solving:
  - nothing provides requested ros-noetic-rtabmap-ros

wolfv, 2021-11-22:
Did you add the the robostack channel (-c robostack -c conda-forge)?

Tobias-Fischer, 2021-11-22:
I just built + uploaded the ros-noetic-rtabmap-ros package - it should work now. Note that no GUI support is available at the moment. Feel free to reopen if it doesn't work.

closed 2021-11-22 by Tobias-Fischer