RoboStack/ros-noetic #192: Turtlesim graphics not showing up

vikramadityafcb, 2021-11-16:
When I try to run `rosrun turtlesim turtlesim_node` I get the following output 

> [ INFO] [1637036965.024629000]: Starting turtlesim with node name /turtlesim
[ INFO] [1637036965.029622000]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]

However, while it initiates a turtlesim_node.exec, I cannot see the turtle and the other graphics as shown in this [http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics](http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics).

How can I solve this? 

Thank you so much for your help!!

Tobias-Fischer, 2021-11-16:
Hi @vikramadityafcb, is this on osx? If so, let's please stick to https://github.com/RoboStack/ros-noetic/issues/158

If it's on another platform, please let us know here.

vikramadityafcb, 2021-11-16:
It's on OSX. My bad, I did not see the previous thread. 
Thank you for the quick response!

closed 2021-11-16 by vikramadityafcb