RoboStack/ros-noetic #193: Cannot get Moveit to launch tutorial simulation

Mechazo11, 2021-11-16:
Dear RoboStack developers,

Firstly, congratulations on integrating ROS with the Anaconda environment. 

I followed the installation tutorial as depicted here https://github.com/RoboStack/ros-noetic and can get both Rviz and Gazebo to launch. 

Note that, when creating the conda environment, I intentionally used Python 3.7 in hopes of tackling compatibility issues with Pytorch library which I plan to use later on.

After successfully building the conda environment as discussed, I created a ``catkin_ws'' workspace and initialized it with catkin_make which ran just fine.

The article https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3 talks both MoveIT and Franka robot being available in neotic-desktop-full installation. However, when I invoked "roslaunch panda_moveit_config demo.launch rviz_tutorial:=true", the system is reporting the following

"RLException: [demo.launch] is neither a launch file in package [panda_moveit_config] nor is [panda_moveit_config] a launch file name.

What is the correct procedure to get both moveit and franka emika panda package to be installed correctly into the anaconda environment?
 
With best,
Azmyin Md. Kamal,
Doctoral Students, Dept. of ME, LSU

Tobias-Fischer, 2021-11-16:
Hi @Mechazo11,

Many thanks for getting in touch and for the bug report!

You need to install the franka / panda packages separately, they are not included in `ros-noetic-desktop-full`. 

For panda, please `mamba install ros-noetic-moveit-resources-panda-description ros-noetic-moveit-resources-panda-moveit-config ros-noetic-panda-moveit-config`

For franka, please see a list of available packages on https://robostack.github.io/noetic.html

Let us know if this helps :).

Mechazo11, 2021-11-16:
Sure, I will try it out and let you know as soon as I am able. I was wondering, would I be able to install ROS neotic on a ubuntu 18.04 using RoboStack?

Tobias-Fischer, 2021-11-16:
Yes - any recent-ish Linux will do :)

Tobias-Fischer, 2021-11-16:
Oh, also .. we only build for Python 3.8, so please use these builds (the other builds would be very very old). Pytorch (and TensorFlow) is now available on conda-forge (CPU+GPU) for various Python versions (including 3.8), so you should not have any issues with that :).

Mechazo11, 2021-11-17:
Hello,

I was able to launch RViz and start the tutorial simulation for MoveIT from Anaconda in a 18.04 Ubuntu machine using your suggested packages.

However, after multiple attempts in updating the packages and manually installing missing packages from https://robostack.github.io/noetic.html I am getting this error when I invoke Moveit's MotionPlanning tool to start the turotial.

The error message is given as follows

[ERROR] [1637128813.802814424]: PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.  Error: Failed to load library /home/az/anaconda3/envs/ros_neotic/lib/libmoveit_motion_planning_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmoveit_semantic_world.so.1.1.0: cannot open shared object file: No such file or directory)

Is the moveit_rviz_plugin package fully supported in RobotStack?

Tobias-Fischer, 2021-11-17:
COuld you please provide the output of `conda list`?

Mechazo11, 2021-11-17:
nspr                      4.32                 h9c3ff4c_1    conda-forge
nss                       3.72                 hb5efdd6_0    conda-forge
numpy                     1.21.4           py38he2449b9_0    conda-forge
octomap                   1.9.7                h4bd325d_0    conda-forge
ogre                      1.10.12              hf655ca9_7    conda-forge
olefile                   0.46               pyh9f0ad1d_1    conda-forge
openal-soft               1.20.1               hc9558a2_0    conda-forge
openexr                   2.5.5                hf817b99_0    conda-forge
openh264                  2.1.1                h780b84a_0    conda-forge
openjpeg                  2.4.0                hb52868f_1    conda-forge
openssl                   1.1.1l               h7f98852_0    conda-forge
orocos-kdl                1.4.0                h9c3ff4c_0    conda-forge
osrf_pycommon             0.2.1              pyhd8ed1ab_0    conda-forge
packaging                 21.2               pyhd8ed1ab_1    conda-forge
pandas                    1.3.4            py38h43a58ef_1    conda-forge
pango                     1.48.10              hb8ff022_1    conda-forge
paramiko                  2.8.0              pyhd8ed1ab_0    conda-forge
partd                     1.2.0              pyhd8ed1ab_0    conda-forge
pcl                       1.11.1               h39b1736_3    conda-forge
pcre                      8.45                 h9c3ff4c_0    conda-forge
pillow                    8.4.0            py38h8e6f84c_0    conda-forge
pip                       21.3.1             pyhd8ed1ab_0    conda-forge
pixman                    0.40.0               h36c2ea0_0    conda-forge
pkg-config                0.29.2            h36c2ea0_1008    conda-forge
poco                      1.10.1               h9c89518_1    conda-forge
pooch                     1.5.2              pyhd8ed1ab_0    conda-forge
poppler                   21.03.0              h93df280_0    conda-forge
poppler-data              0.4.11               hd8ed1ab_0    conda-forge
postgresql                13.3                 h2510834_3    conda-forge
proj                      8.0.1                h277dcde_0    conda-forge
protobuf                  3.15.8           py38h709712a_0    conda-forge
psutil                    5.8.0            py38h497a2fe_2    conda-forge
pthread-stubs             0.4               h36c2ea0_1001    conda-forge
pugixml                   1.11.4               h9c3ff4c_0    conda-forge
py-opencv                 4.5.2            py38hd0cf306_0    conda-forge
pycairo                   1.20.1           py38hf61ee4a_1    conda-forge
pycparser                 2.21               pyhd8ed1ab_0    conda-forge
pycryptodome              3.11.0           py38h80e8405_1    conda-forge
pycryptodomex             3.11.0           py38h497a2fe_1    conda-forge
pydot                     1.4.2            py38h578d9bd_1    conda-forge
pynacl                    1.4.0            py38h497a2fe_3    conda-forge
pyopengl                  3.1.5                      py_0    conda-forge
pyopenssl                 21.0.0             pyhd8ed1ab_0    conda-forge
pyparsing                 2.4.7              pyhd8ed1ab_1    conda-forge
pyqt                      5.12.3           py38h578d9bd_8    conda-forge
pyqt-impl                 5.12.3           py38h0ffb2e6_8    conda-forge
pyqt5-sip                 4.19.18          py38h709712a_8    conda-forge
pyqtchart                 5.12             py38h7400c14_8    conda-forge
pyqtwebengine             5.12.1           py38h7400c14_8    conda-forge
pysocks                   1.7.1            py38h578d9bd_4    conda-forge
python                    3.8.12          hb7a2778_2_cpython    conda-forge
python-dateutil           2.8.2              pyhd8ed1ab_0    conda-forge
python-gnupg              0.4.7              pyhd8ed1ab_0    conda-forge
python-orocos-kdl         1.4.0            py38h709712a_0    conda-forge
python_abi                3.8                      2_cp38    conda-forge
pytz                      2021.3             pyhd8ed1ab_0    conda-forge
pywavelets                1.2.0            py38h6c62de6_0    conda-forge
pyyaml                    6.0              py38h497a2fe_3    conda-forge
qhull                     2020.2               h4bd325d_2    conda-forge
qt                        5.12.9               hda022c4_4    conda-forge
qwt                       6.1.6                h7ec6b3e_0    conda-forge
readline                  8.1                  h46c0cb4_0    conda-forge
requests                  2.26.0             pyhd8ed1ab_0    conda-forge
rhash                     1.4.1                h7f98852_0    conda-forge
ros-distro-mutex          0.1.1                    noetic    robostack
ros-noetic-actionlib      1.13.2           py38h14b2acc_6    robostack
ros-noetic-actionlib-msgs 1.13.1           py38he9ab703_6    robostack
ros-noetic-actionlib-tutorials 0.2.0            py38he9ab703_6    robostack
ros-noetic-angles         1.9.13           py38he9ab703_6    robostack
ros-noetic-audio-common-msgs 0.3.11           py38he9ab703_9    robostack
[... 339 lines cut ...]
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxfixes            5.0.3             h516909a_1004    conda-forge
xorg-libxi                1.7.10               h516909a_0    conda-forge
xorg-libxinerama          1.1.4             h9c3ff4c_1001    conda-forge
xorg-libxmu               1.1.3                h516909a_0    conda-forge
[... cut ...]
yaml                      0.2.5                h516909a_0    conda-forge
yaml-cpp                  0.6.3                he1b5a44_4    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zfp                       0.5.5                h9c3ff4c_7    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

Tobias-Fischer, 2021-11-17:
Could you try `mamba install ros-noetic-moveit-ros-perception=1.1.0`?

Tobias-Fischer, 2021-11-17:
All these things should hopefully go away once we finish #170 ..

Tobias-Fischer, 2021-11-24:
Any update @Mechazo11?

Mechazo11, 2021-11-25:
Hi @Tobias-Fischer apologies for the delay in getting back to you. Was very busy last week. I will test my ROS setup with mamba install ros-noetic-moveit-ros-perception=1.1.0 and get back to you today

Mechazo11, 2021-11-25:
Hi @Tobias-Fischer,

I got MoveIT's tutorial to work with ROS Neotic running on a conda environment after installing the ros-noetic-moveit-ros-perception=1.1.0 package. Attached is a screenshot demonstrating MoveIT running directly off a conda environment. Additionally, my machine is a Ubuntu 18.04 but it is running ROS neotic just fine thanks to your project. I am closing this issue now. Thank you very for your prompt responses in solving this problem. 

[image]

closed 2021-11-25 by Mechazo11