RoboStack/ros-noetic #194: liborocos-kdl missing after moveit upgrade.

capaulson, 2021-11-16:
I updated moveit packages and had issues re-building a catkin workspace. The issue comes back to `liborocos-kdl`. I see packages for melodic, but nothing for noetic. Any ideas on how to fix?

Tobias-Fischer, 2021-11-16:
Hi @capaulson,

Could you please provide some more details? 
- Which issues did you encounter (please provide the stacktrace)? 
- How did you narrow it down to `liborocos-kdl`? 
- Which packages exist for melodic but not noetic? 

orocos-kdl is part of conda-forge (https://github.com/conda-forge/orocos-kdl-feedstock), and is used by RoboStack but not maintained in our repositories.

capaulson, 2021-11-16:
So sorry for the lack of details. 

The error in Catkin: 
```make[2]: *** No rule to make target '***/base/envs/ros/lib/liborocos-kdl.1.5.0.dylib', needed by ***```

There is a liborocos-kdl.dylib in that directory, and generating a symlink for liborocos-kdl.1.5.0.dylib pointing to liborocos-kdl.dylib works, i.e the catkin build doesn't have any errors. 

I'm fairly sure this all comes out of moveit, as that was the only thing I updated.

Tobias-Fischer, 2021-11-17:
Hi @capaulson, could you please try `mamba install orocos-kdl=1.5.0 python-orocos-kdl=1.5.0`? Sorry, this is still an issue on our side.

capaulson, 2021-11-21:
Got it. Thanks!

closed 2021-11-21 by capaulson