RoboStack/ros-noetic #195: Support for Intera-SDK / Sawyer robot

homangab, 2021-11-17:
Thanks for releasing RoboStack! This will be very helpful for the community. I was wondering if it is possible to include packages for the Sawyer robot https://sdk.rethinkrobotics.com/intera/Workstation_Setup#Install_Intera_SDK_Dependencies in the list here https://robostack.github.io/noetic.html?

Tobias-Fischer, 2021-11-17:
Hi @homangab,

It seems like the Sawyer packages have not been released for ROS Noetic, so it's not straightforward for us to do.

You can try building the packages in the same way that the `jackal` packages are built though - please check out e.g. https://github.com/RoboStack/ros-noetic/tree/master/additional_recipes/ros-noetic-jackal-description on how to do that.

homangab, 2021-11-19:
Thanks for the quick reply! That makes sense. I am not very familiar with jackal (:sweat_smile: ), but will try to follow the link you sent.

Tobias-Fischer, 2021-11-19:
If you read through the paper and contributing.md you'll see how to generate most of the recipe.yaml automatically - I think it's just the sources which you'll need to change from a local folder to git if you want everyone to benefit :)