RoboStack/ros-noetic #196: Permission Denied when doing rodeo init on M1 Mac

gy2256, 2021-11-17:
I am following the instruction to install the ROS. However, I encountered the issue 

```
ERROR: cannot create /etc/ros/rosdep/sources.list.d/20-default.list:
	[Errno 13] Permission denied: '/etc/ros/rosdep/sources.list.d/20-default.list'
```

after I ran the `rosdep init` command. I am tempted to use sudo but the comment is asking me not to do so. Is there a fix for this?

Tobias-Fischer, 2021-11-17:
Have you run `conda deactivate` and `conda activate robostackenv`?

Once you have done this, could you report the output of `echo $ROS_ETC_DIR`?

gy2256, 2021-11-17:
Thanks for the reply. Yes, I have followed the instruction deactivate and activate the robostackenv. The output is `/Users/du/miniforge3/envs/robostackenv/etc/ros`

Tobias-Fischer, 2021-11-17:
What's the output of `which rosdep` and `conda list`?

gy2256, 2021-11-18:
Never mind, everything works now after doing the `conda deactivate` and `conda activate robostackenv` again. (Maybe I forget to do it during the first installation?)

For reference, `which rosdep` has the output `/Users/du/miniforge3/envs/robostackenv/bin/rosdep`

and `conda list` has the output
```
# packages in environment at /Users/du/miniforge3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.74.0               h128b0ad_5    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.3.0                h3422bc3_0    conda-forge
ca-certificates           2021.10.8            h4653dfc_0    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_0    conda-forge
catkin_tools              0.7.1              pyhd8ed1ab_0    conda-forge
cctools                   949.0.1             ha9384d2_21    conda-forge
cctools_osx-arm64         949.0.1             h92b3a8a_21    conda-forge
clang                     11.1.0               hce30654_1    conda-forge
clang-11                  11.1.0          default_h0fdd720_1    conda-forge
clang_osx-arm64           11.1.0               h54d7cd3_4    conda-forge
clangxx                   11.1.0          default_h8b1a7b7_1    conda-forge
clangxx_osx-arm64         11.1.0               hb84c830_4    conda-forge
cmake                     3.21.3               h6f0b043_0    conda-forge
compiler-rt               11.1.0               h7d99272_0    conda-forge
compiler-rt_osx-arm64     11.1.0               h103ad0d_0    conda-forge
compilers                 1.3.0                hce30654_0    conda-forge
console_bridge            1.0.1                h260d524_0    conda-forge
cxx-compiler              1.3.0                hc021e02_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18             py38h10201cd_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.1                hbdafb3b_0    conda-forge
fortran-compiler          1.3.0                h630c574_0    conda-forge
gettext                   0.19.8.1          h049c9fb_1008    conda-forge
gfortran                  11.0.1.dev0         h40eb566_14    conda-forge
gfortran_impl_osx-arm64   11.0.1.dev0         hbce66ab_23    conda-forge
gfortran_osx-arm64        11.0.1.dev0         h57527a5_14    conda-forge
gmock                     1.10.0               h260d524_7    conda-forge
gmp                       6.2.1                h9f76cd9_0    conda-forge
gtest                     1.10.0               h260d524_7    conda-forge
icu                       69.1                 hbdafb3b_0    conda-forge
isl                       0.22.1               hb904e53_2    conda-forge
krb5                      1.19.2               hfff258c_3    conda-forge
ld64                      530                 h08716b2_21    conda-forge
ld64_osx-arm64            530                 ha291f0b_21    conda-forge
ldid                      2.1.2                h34ab4f1_3    conda-forge
libclang-cpp11.1          11.1.0          default_h0fdd720_1    conda-forge
libcurl                   7.80.0               h8d07098_0    conda-forge
libcxx                    12.0.1               h168391b_0    conda-forge
libedit                   3.1.20191231         hc8eb9b7_2    conda-forge
libev                     4.33                 h642e427_1    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libgfortran               5.0.0.dev0      11_0_1_hf114ba7_23    conda-forge
libgfortran-devel_osx-arm64 11.0.1.dev0         hf114ba7_23    conda-forge
libgfortran5              11.0.1.dev0         hf114ba7_23    conda-forge
libglib                   2.70.1               h67e64d8_0    conda-forge
libiconv                  1.16                 h642e427_0    conda-forge
libllvm11                 11.1.0               h93073aa_2    conda-forge
libnghttp2                1.43.0               h62a264d_1    conda-forge
libssh2                   1.10.0               h7a5bd25_2    conda-forge
libuv                     1.42.0               h3422bc3_0    conda-forge
libzlib                   1.2.11            hee7b306_1013    conda-forge
llvm-openmp               12.0.1               hf3c4609_1    conda-forge
llvm-tools                11.1.0               h93073aa_2    conda-forge
lz4-c                     1.9.3                hbdafb3b_1    conda-forge
make                      4.3                  he57ea6c_1    conda-forge
mpc                       1.2.1                h309154c_0    conda-forge
mpfr                      4.1.0                h6d7a090_1    conda-forge
ncurses                   6.2                  h9aa5885_4    conda-forge
[... 14 lines cut ...]
rhash                     1.4.1                h27ca646_0    conda-forge
ros-distro-mutex          0.2.0                    noetic    robostack
ros-noetic-catkin         0.8.10           py38hf6adf81_1    robostack
ros-noetic-cpp-common     0.7.2            py38hb56f3aa_1    robostack
ros-noetic-genmsg         0.5.16           py38hf6adf81_1    robostack
ros-noetic-genpy          0.6.15           py38h4f44ae5_1    robostack
ros-noetic-message-runtime 0.4.13           py38hf6adf81_1    robostack
[... cut ...]
tapi                      1100.0.11            he4954df_0    conda-forge
tk                        8.6.11               he1e0b03_1    conda-forge
wheel                     0.37.0             pyhd8ed1ab_1    conda-forge
xz                        5.2.5                h642e427_1    conda-forge
yaml                      0.2.5                h642e427_0    conda-forge
zlib                      1.2.11            hee7b306_1013    conda-forge
zstd                      1.5.0                h861e0a7_0    conda-forge
```

closed 2021-11-18 by gy2256