RoboStack/ros-noetic #197: Compiling ROS project while catkin_make failed

DikZoo, 2021-11-22:
Hi, first of all thank you for integrating ROS with Conda, great work to push forward.
I have followed your instruction to setup Conda env, which called "ROS". However, here I face some problems when compiling project using catkin_make. The details are given below.

System: M1Pro 10-core CPU, macOS Monterey 12.0.1
IDE: CLion
Toolchains: 
- Make: /Users/jiayelin/miniforge3/envs/ROS/bin/make
- C Compiler: /Library/Developer/CommandLineTools/usr/bin/cc
- C++ Compiler: /Library/Developer/CommandLineTools/usr/bin/c++
CMake Environment: 
- CMAKE_PREFIX_PATH=/Users/jiayelin/miniforge3/envs/ROS/
- PATH=/Users/jiayelin/miniforge3/bin:/Users/jiayelin/miniforge3/condabin:/opt/homebrew/bin:/opt/homebrew/sbin:
   /Users/jiayelin/miniforge3/envs/ROS/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
Project for Compiling:
- Fast LOAM: https://github.com/wh200720041/floam
 
**Problem Description**:
In IDE, I can successfully build the project after making several changes to CMakeLists(mainly some PCL module required Qt support which is not available on osx-arm64). But in Terminal, when I use catkin_make under workspace folder, there are two problems: 
- **fatal error: 'ros/ros.h' file not found**
- **runtime library [libboost_program_options.dylib] in /Users/jiayelin/miniforge3/envs/ROS/lib may be hidden by files in:
      /opt/homebrew/lib**
For the first one, I have confirmed that catkin_INCLUDE_DIRS is added to include_directories, and it contains "/Users/jiayelin/miniforge3/envs/ROS/include". Also, roscpp is included in find_package and package.xml.
For the second one, I just don't have any idea how to solve the conflict.

The full **CMakeLists** is given below:
```
cmake_minimum_required(VERSION 2.8.12)
cmake_policy(SET CMP0054 NEW)
project(floam)

set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_CXX_FLAGS "-std=c++14")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

set(Ceres_INCLUDE_DIRS "/opt/homebrew/Cellar/ceres-solver/2.0.0_4/include")
set(Ceres_LIBRARIES "/opt/homebrew/Cellar/ceres-solver/2.0.0_4/lib")

find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  nav_msgs
  sensor_msgs
  roscpp
  rospy
  rosbag
  std_msgs
  tf
  eigen_conversions
)

find_package(Eigen3)
if(NOT EIGEN3_FOUND)
  # Fallback to cmake_modules
  find_package(cmake_modules REQUIRED)
  find_package(Eigen REQUIRED)
  set(EIGEN3_INCLUDE_DIRS ${EIGEN_INCLUDE_DIRS})
  set(EIGEN3_LIBRARIES ${EIGEN_LIBRARIES})  # Not strictly necessary as Eigen is head only
  # Possibly map additional variables to the EIGEN3_ prefix.
else()
  set(EIGEN3_INCLUDE_DIRS ${EIGEN3_INCLUDE_DIR})
endif()
find_package(PCL REQUIRED IO SEGMENTATION FEATURES FILTERS COMMON RECOGNITION)
find_package(Ceres REQUIRED)

include_directories(
  include
  ${catkin_INCLUDE_DIRS}
  ${PCL_INCLUDE_DIRS}
  ${CERES_INCLUDE_DIRS}
  ${EIGEN3_INCLUDE_DIRS}
)
[... 165 lines cut ...]
> 1 error generated.
> make[2]: *** [floam/CMakeFiles/floam_laser_mapping_node.dir/build.make:76: floam/CMakeFiles/floam_laser_mapping_node.dir/src/laserMappingNode.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> 1 error generated.
> make[2]: *** [floam/CMakeFiles/floam_odom_estimation_node.dir/build.make:76: floam/CMakeFiles/floam_odom_estimation_node.dir/src/odomEstimationNode.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> In file included from /Users/jiayelin/Projects/floam_ws/src/floam/src/laserMappingClass.cpp:5:
> /Users/jiayelin/Projects/floam_ws/src/floam/include/laserMappingClass.h:14:10: fatal error: 'pcl_ros/impl/transforms.hpp' file not found
> #include <pcl_ros/impl/transforms.hpp>
>          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 1 error generated.
> make[2]: *** [floam/CMakeFiles/floam_laser_mapping_node.dir/build.make:90: floam/CMakeFiles/floam_laser_mapping_node.dir/src/laserMappingClass.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:297: floam/CMakeFiles/floam_laser_mapping_node.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> In file included from /Users/jiayelin/Projects/floam_ws/src/floam/src/odomEstimationClass.cpp:5:
> /Users/jiayelin/Projects/floam_ws/src/floam/include/odomEstimationClass.h:34:10: fatal error: 'ros/ros.h' file not found
> #include <ros/ros.h>
>          ^~~~~~~~~~~
> 1 error generated.
> make[2]: *** [floam/CMakeFiles/floam_odom_estimation_node.dir/build.make:118: floam/CMakeFiles/floam_odom_estimation_node.dir/src/odomEstimationClass.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:323: floam/CMakeFiles/floam_odom_estimation_node.dir/all] Error 2
> make[1]: *** [CMakeFiles/Makefile2:1493: floam/CMakeFiles/floam_laser_processing_node.dir/all] Error 2
> make: *** [Makefile:146: all] Error 2
> Invoking "make -j10 -l10" failed
Looking forward to your reply! :)

Tobias-Fischer, 2021-11-22:
Hi, please follow the directions in the readme, particularly installing the compilers package and Ceres using conda.

DikZoo, 2021-11-22:
Thank you for your reply. I have followed the directions and install compilers to the env, conda list shows below:

> # packages in environment at /Users/jiayelin/miniforge3/envs/ROS:
> #
> # Name                    Version                   Build  Channel
> apr                       1.7.0                h9b22ae9_5    conda-forge
> atk-1.0                   2.36.0               heb41896_4    conda-forge
> bcrypt                    3.2.0            py38hea4295b_2    conda-forge
> boost                     1.74.0           py38h1e2c3d9_4    conda-forge
> boost-cpp                 1.74.0               h128b0ad_5    conda-forge
> bzip2                     1.0.8                h3422bc3_4    conda-forge
> c-ares                    1.18.1               h3422bc3_0    conda-forge
> c-compiler                1.3.0                h3422bc3_0    conda-forge
> ca-certificates           2021.10.8            h4653dfc_0    conda-forge
> cairo                     1.16.0            hb99c34e_1009    conda-forge
> catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
> catkin_tools              0.7.2              pyhd8ed1ab_0    conda-forge
> cctools                   949.0.1             ha9384d2_21    conda-forge
> cctools_osx-arm64         949.0.1             h92b3a8a_21    conda-forge
> ceres-solver              2.0.0                h586e65e_4    conda-forge
> cffi                      1.15.0           py38hc67bbb8_0    conda-forge
> clang                     11.1.0               hce30654_1    conda-forge
> clang-11                  11.1.0          default_h0fdd720_1    conda-forge
> clang_osx-arm64           11.1.0               h54d7cd3_4    conda-forge
> clangxx                   11.1.0          default_h8b1a7b7_1    conda-forge
> clangxx_osx-arm64         11.1.0               hb84c830_4    conda-forge
> cmake                     3.21.3               h6f0b043_0    conda-forge
> compiler-rt               11.1.0               h7d99272_0    conda-forge
> compiler-rt_osx-arm64     11.1.0               h103ad0d_0    conda-forge
> compilers                 1.3.0                hce30654_0    conda-forge
> console_bridge            1.0.1                h260d524_0    conda-forge
> cryptography              35.0.0           py38h10d4710_2    conda-forge
> cxx-compiler              1.3.0                hc021e02_0    conda-forge
> defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
> distro                    1.6.0              pyhd8ed1ab_0    conda-forge
> docutils                  0.18             py38h10201cd_1    conda-forge
> eigen                     3.4.0                hc021e02_0    conda-forge
> empy                      3.3.4              pyh9f0ad1d_1    conda-forge
> expat                     2.4.1                hbdafb3b_0    conda-forge
> font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
> font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
> font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
> font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
> fontconfig                2.13.1            heb65262_1005    conda-forge
> fonts-conda-ecosystem     1                             0    conda-forge
> fonts-conda-forge         1                             0    conda-forge
> fortran-compiler          1.3.0                h630c574_0    conda-forge
> freetype                  2.10.4               h17b34a0_1    conda-forge
> fribidi                   1.0.10               h27ca646_0    conda-forge
> gdk-pixbuf                2.42.6               hff60771_0    conda-forge
> gettext                   0.19.8.1          h049c9fb_1008    conda-forge
> gflags                    2.2.2             hc88da5d_1004    conda-forge
> gfortran                  11.0.1.dev0         h40eb566_14    conda-forge
> gfortran_impl_osx-arm64   11.0.1.dev0         hbce66ab_23    conda-forge
> gfortran_osx-arm64        11.0.1.dev0         h57527a5_14    conda-forge
> giflib                    5.2.1                h27ca646_2    conda-forge
> glog                      0.5.0                h5c6a83d_0    conda-forge
> gmock                     1.10.0               h260d524_7    conda-forge
> gmp                       6.2.1                h9f76cd9_0    conda-forge
> gpgme                     1.16.0               hbdafb3b_0    conda-forge
> graphite2                 1.3.13            h9f76cd9_1001    conda-forge
> graphviz                  2.49.3               h2097151_0    conda-forge
> gtest                     1.10.0               h260d524_7    conda-forge
> gtk2                      2.24.33              hedeb848_1    conda-forge
> gts                       0.7.6                h4b6d4d6_2    conda-forge
> harfbuzz                  3.1.1                h6efa254_1    conda-forge
> icu                       69.1                 hbdafb3b_0    conda-forge
> isl                       0.22.1               hb904e53_2    conda-forge
> jbig                      2.1               h3422bc3_2003    conda-forge
> jpeg                      9d                   h27ca646_0    conda-forge
[... 181 lines cut ...]
> zstd                      1.5.0                h861e0a7_0    conda-forge

I delete ceres-solver installed by homebrew and reinstall it using `conda install ceres-solver`. However, in terminal catkin_make still can't find ros/ros.h and ceres/ceres.h too.

[... cut ...]
>   floam_laser_mapping_node because there is a cycle in the constraint graph:
> 
>     dir 0 is [/Users/jiayelin/Projects/floam_ws/src/floam/include]
>     dir 1 is [/opt/homebrew/lib]
>       dir 2 must precede it due to runtime library [libboost_program_options.dylib]
>     dir 2 is [/Users/jiayelin/miniforge3/envs/ROS/lib]
>       dir 1 must precede it due to runtime library [/libglog.0.dylib]
> 
>   Some of these libraries may not be found correctly.

So it really confuse me that the env has included in the $PATH, but the header files are still not found.

Tobias-Fischer, 2021-11-22:
Hi, it seems that it still some libraries from homebrew (see dir 1). Have you cleaned your floam_ws and tried again? Note that you can use rosdep to install dependencies. I think pcl is still not installed via conda - try installing this and cleaning your workspace.

DikZoo, 2021-11-23:
Yes, pcl is installed from homebrew and it works fine. After I clean workspace and recompile the project, in IDE it is totally successful even pcl is loaded from brew path. So I guess pcl is not relating to the header problem. 
Now I have pcl installed through Conda so every library this project need comes from Conda. In CLion, the project is build successfully and I can navigate to the header file of **ros, pcl, ceres, boost**. But in Terminal, the header problem is not solved yet, here are error log:
```
/Users/jiayelin/Projects/floam_ws/src/floam/include/lidarOptimization.h:7:10: fatal error: 'ceres/ceres.h' file not found
#include <ceres/ceres.h>
         ^~~~~~~~~~~~~~~
/Users/jiayelin/Projects/floam_ws/src/floam/src/laserProcessingNode.cpp/Users/jiayelin/Projects/floam_ws/src/floam/src/odomEstimationNode.cpp::1414::1010::  fatal errorfatal error: : 'ros/ros.h' file not found'ros/ros.h' file not found

/Users/jiayelin/Projects/floam_ws/src/floam/src/laserMappingNode.cpp#include <ros/ros.h>:
14:         ^~~~~~~~~~~10
#include <ros/ros.h>
         ^~~~~~~~~~~
: fatal error: 'ros/ros.h' file not found
#include <ros/ros.h>
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [floam/CMakeFiles/floam_odom_estimation_node.dir/build.make:90: floam/CMakeFiles/floam_odom_estimation_node.dir/src/lidarOptimization.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [floam/CMakeFiles/floam_laser_mapping_node.dir/build.make:76: floam/CMakeFiles/floam_laser_mapping_node.dir/src/laserMappingNode.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
1 error generated.
make[2]: *** [floam/CMakeFiles/floam_laser_processing_node.dir/build.make:76: floam/CMakeFiles/floam_laser_processing_node.dir/src/laserProcessingNode.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [floam/CMakeFiles/floam_odom_estimation_node.dir/build.make:76: floam/CMakeFiles/floam_odom_estimation_node.dir/src/odomEstimationNode.cpp.o] Error 1
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/src/laserProcessingClass.cpp:4:
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/include/laserProcessingClass.h:7:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_cloud.h:47:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/PCLHeader.h:3:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/memory.h:46:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/type_traits.h:40:
/Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_struct_traits.h:42:10: fatal error: 'boost/mpl/assert.hpp' file not found
#include <boost/mpl/assert.hpp>  // for BOOST_MPL_ASSERT_MSG
         ^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/src/laserMappingClass.cpp:5:
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/include/laserMappingClass.h:9:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_cloud.h:47:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/PCLHeader.h:3:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/memory.h:46:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/type_traits.h:40:
/Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_struct_traits.h:42:10: fatal error: 'boost/mpl/assert.hpp' file not found
#include <boost/mpl/assert.hpp>  // for BOOST_MPL_ASSERT_MSG
         ^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/src/odomEstimationClass.cpp:5:
In file included from /Users/jiayelin/Projects/floam_ws/src/floam/include/odomEstimationClass.h:13:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_cloud.h:47:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/PCLHeader.h:3:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/memory.h:46:
In file included from /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/type_traits.h:40:
/Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_struct_traits.h:42:10: fatal error: 'boost/mpl/assert.hpp' file not found
#include <boost/mpl/assert.hpp>  // for BOOST_MPL_ASSERT_MSG
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
```
As you can see, `ceres, ros, boost` header is not found, but `pcl` is not report error(I guess this is hidden by current error). while their header's absolute paths are:

- **ceres header**: /Users/jiayelin/miniforge3/envs/ROS/include/ceres/ceres.h
- **ros header**: /Users/jiayelin/miniforge3/envs/ROS/include/ros/ros.h
- **boost/mpl/assert**: /Users/jiayelin/miniforge3/envs/ROS/include/boost/mpl/assert.hpp
- **pcl/point_cloud**: /Users/jiayelin/miniforge3/envs/ROS/include/pcl-1.11/pcl/point_cloud.h

And in floam_ws, after activate ROS env, the output of `echo $PATH` is:
/Users/jiayelin/miniforge3/envs/ROS/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/Users/jiayelin/miniforge3/condabin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin

I can be sure that all dependencies needed are installed. It is strange that pcl,ceres installed through homebrew can be found correctly by CMakeLists, but when they installed through Conda, they can not be found just like ros/ros.h.

**Another question**: is there any way to search package libs only in Conda env by making changes to CMakeLists, because Conda and homebrew both contains boost library, which cause: 
```
runtime library [libboost_system.dylib] in /Users/jiayelin/miniforge3/envs/ROS/lib may be hidden by files in:
  /opt/homebrew/lib
```

Thank you for your patience and all the suggestions!

Tobias-Fischer, 2021-11-23:
See https://github.com/RoboStack/ros-noetic/issues/175#issuecomment-934887036 - the CMakeLists of floam is not correct (same issue as described there).

DikZoo, 2021-11-23:
Thank you! I have tried to remove those lines in CMakeLists, and it works! I do a little search, is this because `set CMAKE_CXX_FLAGS` makes the project limited to **gcc compiler**, while in macOS the compiler is **Clang**?
So, is it means if I want to compile the project in both systems(Linux & macOS) I need to append the CMAKE_CXX_FLAGS instead of set it?

Currently, it seems pcl_conversions and pcl_ros is not supported on M1(so as RViz). Will they be available in future？

Thank you for your advice :)

Tobias-Fischer, 2021-11-23:
Yes to both questions.

Best, Tobi

closed 2021-11-23 by Tobias-Fischer

QuteSaltyFish, 2023-09-05:
Get the same problem today when trying to built floam on ubuntu 20.04. After setting the CMAKE_CXX_FLAGS to append, the project can be built, but the result is totally not working.

Tobias-Fischer, 2023-09-05:
What is not working?

QuteSaltyFish, 2023-09-05:
the floam code generating the wrong mapping result.

Tobias-Fischer, 2023-09-05:
Does it work on the same data without RoboStack?

QuteSaltyFish, 2023-09-05:
yes, I use the same data and build it use the native ros noetic with 20.04 and it works.

QuteSaltyFish, 2023-09-05:
I think the problem may be that PCL require c++14

Tobias-Fischer, 2023-09-14:
Can you please open a new issue @QuteSaltyFish using the issue template, and provide exact steps that are needed to replicate this issue?