RoboStack/ros-noetic #198: error when sourcing custom ws

Karthikeyanc2, 2021-11-22:
Hi,

I am facing a similar problem as shown in [https://github.com/RoboStack/ros-noetic/issues/176].

Following are the commands that I used:

conda activate conda_ros
catkin_make
source ~/devel/setup.bash
roslaunch all_sensors all_sensors.launch

Here conda_ros is the environment with ROS from robostack.
all_sensors and custom_messages are my packages that I just built.

I am able to run the nodes but the nodes are not able to find the custom_messages package. But rospack find is able to find them after sourcing.

Also note, I do not even have a ros environment outside this conda_ros, so there is no way things are getting confused inside. I am obviously missing something here. Can you please help me out?

Thanks,
Karthik

Tobias-Fischer, 2021-11-22:
Any chance you can share a minimum working example that demonstrates this bug?

Karthikeyanc2, 2021-11-22:
Hi @Tobias-Fischer,

Yes, I just created a temp package to share you.
[all_sensors_ws.zip](https://github.com/RoboStack/ros-noetic/files/7580180/all_sensors_ws.zip)

[image]

Thanks,
Karthik

wolfv, 2021-11-22:
@Tobias-Fischer did we patch the PYTHONPATH stuff from catkin?

wolfv, 2021-11-22:
I think the issue is that we need to run something like `export PYTHONPATH=/thinkmate_admin/all_sensors_ws/devel/...:$PYTHONPATH`

where the path needs to correspond to the PYTHON install dir.

wolfv, 2021-11-22:
which we are removing here https://github.com/RoboStack/ros-noetic/blob/45fab57a943c6e6a5972a3ee7fc9d806076fe82a/patch/ros-noetic-catkin.patch#L71-L73 because we don't need / want it for a default conda installation where packages are installed in the proper "python" prefix. however with custom envs we probably need to extend the PYTHONPATH to include the workspace install dirs.

Tobias-Fischer, 2021-11-22:
Hmmm but we try to add it back for the catkin workspace use case just after that, seems like this is not working properly: https://github.com/RoboStack/ros-noetic/blob/45fab57a943c6e6a5972a3ee7fc9d806076fe82a/patch/ros-noetic-catkin.patch#L75-L76

Tobias-Fischer, 2021-11-22:
@Karthikeyanc2: which version of ros-noetic-catkin do you have installed?

wolfv, 2021-11-22:
right, sorry.

Tobias-Fischer, 2021-11-22:
Hi,
I think you are using a very old version of robostack, still running python 3.7 (I found a reference to python 3.7 in your source code in the shebang; please send the output of `conda list`). Please also make sure that you properly set the shebang and install the python script: http://wiki.ros.org/UsingPython3/SourceCodeChanges

So locally for me it works without issues.

Tobias-Fischer, 2021-11-22:
Feel free to reopen if this is still an issue with the latest ros-noetic-catkin (see https://anaconda.org/RoboStack/ros-noetic-catkin/files).

closed 2021-11-22 by Tobias-Fischer