RoboStack/ros-noetic #199: Use the TurtleBot3 tutorial to run in conda virtual environment. Gazebo can't display the robot

nfhe, 2021-11-23:
Hi，man

First of all thank you for integrating ROS with Conda, great work to push forward.
I have followed your instruction to setup Conda env, which called "ROS". However, here I face some problems when
I use TurtleBot3 but Gazebo can't display the robot.The details are given below.

**Problem Description:**
I load the file [in](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#gazebo-simulation) .I have turtlebot3,
turtlebot3_msgs and turtlebot3_simulations in catkin_ws.When I start the gazebo simulation using the tutorial, there are no error warnings, but there is no robot model in the gazebo environment

**Here are my detailed operations**
1.conda activate rostorch
2.cd catkin_ws
3.catkin_make
4.source devel/setup.bash
5.export TURTLEBOT3_MODEL=waffle
6.roslaunch turtlebot3_gazebo turtlebot3_world.launch

**Terminal display**
... logging to /home/he/.ros/log/42dd6c9c-4c0b-11ec-bfb7-8c8caa32777f/roslaunch-he-Lenovo-Legion-R7000P2020H-12831.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://he-Lenovo-Legion-R7000P2020H:43655/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [12844]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 42dd6c9c-4c0b-11ec-bfb7-8c8caa32777f
process[rosout-1]: started with pid [12855]
started core service [/rosout]
process[gazebo-2]: started with pid [12859]
process[gazebo_gui-3]: started with pid [12865]
process[spawn_urdf-4]: started with pid [12871].

**End**

I found two and didn't know where the problem was. I look forward to your reply.

Tobias-Fischer, 2021-11-23:
Is that on Linux?

nfhe, 2021-11-23:
Yes, I use the conda virtual environment installed on Ubuntu 18.04, and then the installation tutorial is configured with ROS neotic.

nfhe, 2021-11-23:
I wanted to train withturtlebot3, but there was no robot model when gazebo started. But rviz  can run.

Tobias-Fischer, 2022-02-09:
Could you please `mamba install ros-noetic-turtlebot3-simulations`, remove the turtlebot packages from your catkin_ws, and try again? It works fine for me on osx, so it would surprise me if it doesn't work on Linux. Feel free to re-open if you still encounter issues.

closed 2022-02-09 by Tobias-Fischer