RoboStack/ros-noetic #200: Build ros-noetic-openni2-camera package / Would directly adding package on the src folder compile?

Mechazo11, 2021-11-27:
Dear RoboStack developers,

Is there any tutorial that teaches the community how to add new packages to RobotStack? I am working on a robot localization project using the Asus Xiton Pro live camera which if I am not mistaken requires the ros-noetic-openni2-camera. As of writing this post, this package appears to be yet build and compiled for the RobotStack project.

On a related note, if we add the ros-noetic-openni2-camera directly a catkin workspace's src folder, would that work with RobotStack?

With best regards,
Mechazo11

adamconkey, 2022-05-03:
I also have this issue. `ros-noetic-openni2-launch` is built but it doesn't seem you can do much with it without `ros-noetic-openni2-camera`.

adamconkey, 2022-05-03:
It is however easy to build it yourself. I just checked out the repo `https://github.com/ros-drivers/openni2_camera.git` and did a `catkin build` and was able to use it no problem.

Tobias-Fischer, 2024-02-29:
As @adamconkey said, it's easy to build locally - it will be really hard to do in CI.