RoboStack/ros-noetic #204: Rviz display problems

momoai96, 2021-12-08:
Hey guys ,
it's me again. 

I'm doing some urdf-stuff right now and found a problem:
Creating cylinder elements with length or radius <1 are not shown in Rviz:

`<?xml version = "1.0"?>
<robot name ="testrobot">
    <link name ="base_link">
        <visual>
            <geometry>
                <cylinder radius="0.2" length="0.6"/>
            </geometry>
        </visual>
    </link>
</robot>`

[image]

If you're increasing your parameters it works fine: 

`<?xml version = "1.0"?>
<robot name ="testrobot">
    <link name ="base_link">
        <visual>
            <geometry>
                <cylinder radius="2" length="6"/>
            </geometry>
        </visual>
    </link>
</robot>`

[image]

Would really appreciate your support! 

Greetings!

Tobias-Fischer, 2021-12-08:
Hiya, do you know whether that's only happening with RoboStack (if so, which platform are you on?)? If it also happens with the default ROS installation please open an issue in the rviz repository.

momoai96, 2021-12-08:
Got the Problem only with my M1 MacBook Pro, earlier my Ubuntu Machine did quiet well!

Tobias-Fischer, 2021-12-09:
Ok - could you please send the output of `conda list`?

Tobias-Fischer, 2022-01-30:
Closing because of inactivity

closed 2022-01-30 by Tobias-Fischer