RoboStack/ros-noetic #205: noetic and pytorch

Veal98, 2021-12-23:
hi~
when i use `rosrun xxx xxx.py`, it tells me that 'no mudule named torch`, but it works well in the cmd  if i input `impot torch`

do u have some suggestions?

tks

Tobias-Fischer, 2021-12-23:
Do you install ROS using RoboStack as per the readme in this repo? What is the output of `conda list`?

Veal98, 2021-12-23:
> Do you install ROS using RoboStack as per the readme in this repo? What is the output of `conda list`?

```bash
(yjc) conda list
# packages in environment at /home/seu/anaconda3/envs/yjc:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main    defaults
_openmp_mutex             4.5                       1_gnu    defaults
blas                      1.0                         mkl    defaults
brotli                    1.0.9                he6710b0_2    defaults
bzip2                     1.0.8                h7b6447c_0    defaults
ca-certificates           2021.10.26           h06a4308_2    defaults
certifi                   2021.10.8        py37h06a4308_0    defaults
cudatoolkit               11.3.1               h2bc3f7f_2    defaults
cycler                    0.11.0             pyhd3eb1b0_0    defaults
dbus                      1.13.18              hb2f20db_0    defaults
empy                      3.3.4                    pypi_0    pypi
expat                     2.4.1                h2531618_2    defaults
ffmpeg                    4.3                  hf484d3e_0    pytorch
fontconfig                2.13.1               h6c09931_0    defaults
fonttools                 4.25.0             pyhd3eb1b0_0    defaults
freetype                  2.11.0               h70c0345_0    defaults
giflib                    5.2.1                h7b6447c_0    defaults
glib                      2.69.1               h5202010_0    defaults
gmp                       6.2.1                h2531618_2    defaults
gnutls                    3.6.15               he1e5248_0    defaults
gst-plugins-base          1.14.0               h8213a91_2    defaults
gstreamer                 1.14.0               h28cd5cc_2    defaults
icu                       58.2                 he6710b0_3    defaults
intel-openmp              2021.4.0          h06a4308_3561    defaults
jpeg                      9d                   h7f8727e_0    defaults
kiwisolver                1.3.1            py37h2531618_0    defaults
lame                      3.100                h7b6447c_0    defaults
lcms2                     2.12                 h3be6417_0    defaults
ld_impl_linux-64          2.35.1               h7274673_9    defaults
libffi                    3.3                  he6710b0_2    defaults
libgcc-ng                 9.3.0               h5101ec6_17    defaults
libgfortran-ng            7.5.0               ha8ba4b0_17    defaults
libgfortran4              7.5.0               ha8ba4b0_17    defaults
libgomp                   9.3.0               h5101ec6_17    defaults
libiconv                  1.15                 h63c8f33_5    defaults
libidn2                   2.3.2                h7f8727e_0    defaults
libpng                    1.6.37               hbc83047_0    defaults
libstdcxx-ng              9.3.0               hd4cf53a_17    defaults
libtasn1                  4.16.0               h27cfd23_0    defaults
libtiff                   4.2.0                h85742a9_0    defaults
libunistring              0.9.10               h27cfd23_0    defaults
libuuid                   1.0.3                h7f8727e_2    defaults
libuv                     1.40.0               h7b6447c_0    defaults
libwebp                   1.2.0                h89dd481_0    defaults
libwebp-base              1.2.0                h27cfd23_0    defaults
libxcb                    1.14                 h7b6447c_0    defaults
libxml2                   2.9.12               h03d6c58_0    defaults
lz4-c                     1.9.3                h295c915_1    defaults
matplotlib                3.5.0            py37h06a4308_0    defaults
matplotlib-base           3.5.0            py37h3ed280b_0    defaults
mkl                       2021.4.0           h06a4308_640    defaults
mkl-service               2.4.0            py37h7f8727e_0    defaults
mkl_fft                   1.3.1            py37hd3c417c_0    defaults
mkl_random                1.2.2            py37h51133e4_0    defaults
munkres                   1.1.4                      py_0    defaults
ncurses                   6.3                  h7f8727e_2    defaults
nettle                    3.7.3                hbbd107a_1    defaults
numpy                     1.21.2           py37h20f2e39_0    defaults
numpy-base                1.21.2           py37h79a1101_0    defaults
olefile                   0.46                     py37_0    defaults
opencv-python             4.5.4.60                 pypi_0    pypi
openh264                  2.1.1                h4ff587b_0    defaults
openssl                   1.1.1l               h7f8727e_0    defaults
[... 26 lines cut ...]
zstd                      1.4.9                haebb681_0    defaults
```

and actually, it works well in cmd:
[image]

but it's wrong with `rosrun`
[image]

the code like this:

```python
import sys
import torch
import rospy
import cv2
import numpy as np
import os
import random
from cv_bridge import CvBridge, CvBridgeError
from std_msgs.msg import String
from vision_msgs.msg import Detection2D
from sensor_msgs.msg import Image
from yolo.msg import ImageRequest
```

Tobias-Fischer, 2021-12-23:
You need to install the ros-noetic packages again as per readme - your current environment does not contain any RoboStack packages. Please start from scratch if it doesn't work and follow the readme. The purge command removed all ROS packages, so it cannot work.

closed 2021-12-23 by Tobias-Fischer