RoboStack/ros-noetic #206: How to Port Custom Catkin Packages to RoboStack

aravinho, 2021-12-28:
Thanks for the great work on RoboStack!

I have some private ROS packages I've written for my projects, and in my old workflow, I'd globally install `ros-noetic-desktop`, then create a Catkin workspace, and compile my packages from source within the Catkin workspace, using `catkin build`.

I'm now trying to switch over to RoboStack.  I've created a new Conda env called `ros_env` per the [setup instructions](https://robostack.github.io/GettingStarted.html), and have installed `ros-noetic-desktop` inside the `ros_env`.  I'm now wondering how to compile my custom packages within my newly created `ros_env`.  Do I need to convert my packages into conda packages somehow?  Are there any pointers or tutorials on how to do so?  Thank you very much!

Tobias-Fischer, 2021-12-28:
Hi @aravinho,

Great to hear you want to switch over to RoboStack :)!

There are only very tiny adjustments needed to build your packages in a catkin workspace. In your `ros_env` (which already has `ros-noetic-desktop` installed), make sure that the `catkin_tools`, `compilers` package etc. are also installed (`mamba install catkin_tools compilers`; for other suggestions see the README file in this repo).

You then simply `conda activate ros_env` and run `catkin build` in your catkin workspace. Make sure that you purge the workspace before doing that, though (i.e. delete the `devel`, `install` directories etc.).

Feel free to reopen the issue if you have more questions.

Best, Tobi

closed 2021-12-28 by Tobias-Fischer

aravinho, 2021-12-29:
@Tobias-Fischer that was easy, it worked, thanks!