RoboStack/ros-noetic #212: Issues installing ROS Noetic via mamba on M1 Mac mini

skweesel, 2022-01-28:
Hello. I'm following the instructions on the README. I'm trying to install ros-noetic-desktop on an M1 mac mini. When get to the `mamba install ros-noetic-desktop` step I get 
```
Looking for: ['ros-noetic-desktop']

conda-forge/osx-arm64    Using cache
conda-forge/noarch       Using cache
pkgs/main/noarch         [====================] (00m:00s) No change
pkgs/main/osx-arm64      [====================] (00m:00s) No change
pkgs/r/osx-arm64         [====================] (00m:00s) 404 Failed
pkgs/r/noarch            [====================] (00m:00s) No change
robostack/noarch         [====================] (00m:00s) No change
robostack/osx-arm64      [====================] (00m:00s) No change

Pinned packages:
  - python 3.8.*

Encountered problems while solving:
  - nothing provides requested ros-noetic-desktop
  ```
I've never used conda or mamba as a package manager. I don't know what I'm doing wrong. Here's additional info if it's relevant:
```
 conda env list
# conda environments:
#
base                     /Users/eos/mambaforge
robostackenv          *  /Users/eos/mambaforge/envs/robostackenv
                         /Users/eos/opt/miniconda3
```

Tobias-Fischer, 2022-01-28:
Hi, please see https://github.com/RoboStack/ros-noetic/issues/156

closed 2022-01-28 by Tobias-Fischer

skweesel, 2022-01-28:
Boy you're quick! I'll follow that ticket and report back if that worked for me.

skweesel, 2022-01-28:
I think it worked! These are the steps I followed. The main difference than the ticket referenced is that I had to specify python=3.8 
```
CONDA_SUBDIR=osx-64 conda create -n rosetta python=3.8   # create a new environment called rosetta with intel packages.
conda activate rosetta
python -c "import platform;print(platform.machine())"
conda env config vars set CONDA_SUBDIR=osx-64  # make sure that conda commands in this environment use intel packages
conda deactivate rosetta
conda activate rosetta
echo "CONDA_SUBDIR: $CONDA_SUBDIR"  #Now continue from the README
conda config --env --add channels conda-forge #included here just in case you haven't already previously done this
conda config --env --add channels robostack
conda config --env --set channel_priority strict
mamba install ros-noetic-desktop #Or mamba install ros-noetic-desktop-full
```

Thank you so much @Tobias-Fischer

Tobias-Fischer, 2022-01-29:
Nice! Fyi we have native packages for ROS2 galactic, and we could offer them here as well. All it needs is a bit of time. If interested in helping out let me know

skweesel, 2022-01-29:
Thanks @Tobias-Fischer. I think I'm an intermediate ROS user/dev and I'm not sure what I could help with but I could help with testing instructions and trying to build from your sources if that's useful to this project