RoboStack/ros-noetic #215: Command Prompt Still not working

SoumyajitProjects, 2022-02-01:
After entering with no spaces now the terminal displays: 

 -package ros-noetic-desktop-1.5.0-py39heb88ffd_15 requires ros-noetic-viz, but none of the providers can be installed

traversaro, 2022-02-01:
Can you report the command you are trying to run?

SoumyajitProjects, 2022-02-01:
so essentially I've tried to install ROS-Noetic on my Mac using the prompts giving but for someone reason whenever I enter in **mamba install ros-noetic-desktop** it does not install it as it displays an that an error has occurred.

SoumyajitProjects, 2022-02-02:
**# if you don't have mamba yet, install it first in the base environment (not needed when using mambaforge):
conda install mamba -c conda-forge

conda create -n robostackenv python=3.9 -c conda-forge
conda activate robostackenv
# this adds the conda-forge channel to the new created environment configuration 
conda config --env --add channels conda-forge
# and the robostack channels
conda config --env --add channels robostack-experimental
# and the robostack channel
conda config --env --add channels robostack
# it's very much advised to use strict channel priority
conda config --env --set channel_priority strict

mamba install ros-noetic-desktop

# optionally, install some compiler packages if you want to e.g. build packages in a catkin_ws:
mamba install compilers cmake pkg-config make ninja

# on linux and osx (but not Windows) you might want to:
mamba install catkin_tools
# on Windows, install Visual Studio 2017 or 2019 with C++ support 
# see https://docs.microsoft.com/en-us/cpp/build/vscpp-step-0-installation?view=msvc-160

# only on linux, if you are having issues finding GL/OpenGL, also do:
mamba install mesa-libgl-devel-cos7-x86_64 mesa-dri-drivers-cos7-x86_64 libselinux-cos7-x86_64 libxdamage-cos7-x86_64 libxxf86vm-cos7-x86_64 libxext-cos7-x86_64 xorg-libxfixes

# on Windows, install the Visual Studio command prompt via Conda:
mamba install vs2019_win-64

# note that in this case, you should also install the necessary dependencies with conda/mamba, if possible

# reload environment to activate required scripts before running anything
# on Windows, please restart the Anaconda Prompt / Command Prompt!
conda deactivate
conda activate robostackenv

# if you want to use rosdep, also do:
mamba install rosdep
rosdep init  # note: do not use sudo!
rosdep update**

That is essentially what I was talking about. So basically after I installed anaconda I also installed miniconda, miniforge and mambaforge. I went on anaconda, opened up miniconda form the enviroments and from there I opened up the terminal from which I started to use the installation process which I sent above. The issue is after I enter **mamba install ros-noetic-desktop**, it that it cannot recognize mamba and ROS won't install.

traversaro, 2022-02-02:
> So basically after I installed anaconda I also installed miniconda, miniforge and mambaforge.

Ok, I guess this could be one of the problems. Tipically, you just install one out of `anaconda`, `miniconda`, `miniforge` and `mambaforge`, not all of them. My guess is that in the environment that you created, the `defaults` channel has an high precedence w.r.t. to the `conda-forge` one, and this is creating problems.

However, to avoid having a lot of cross-talking problem, I first suggest to uninstall all the installers that you installed (i.e. `anaconda`, `miniconda`, `miniforge` and `mambaforge`) and just install one of them. If  you do not have a preference, go for `mambaforge` .

SoumyajitProjects, 2022-02-02:
That seemed to have worked. Is there any way to check or is there a way to open ROS after?

Tobias-Fischer, 2022-02-02:
Great to hear! You can now go for standard ROS tutorials: http://wiki.ros.org/ROS/Tutorials

closed 2022-02-02 by Tobias-Fischer