RoboStack/ros-noetic #216: Python 3.6 support removed from ros-noetic-desktop=1.5.0

adammacudzinski, 2022-02-01:
Previously our team was using the following env config file for our ROS noetic system:

`name: dei_env
channels:
  - robostack
  - conda-forge
dependencies:
  - python=3.6
  - ros-noetic-desktop=1.5.0`

But recently running `mamba env create` has started failing with the following error: `Encountered problems while solving:
  - package ros-noetic-desktop-1.5.0-py39h6fdeb60_13 requires python_abi 3.9.*, but none of the providers can be installed`

Any idea what can be done to reenable python3.6 support?

wolfv, 2022-02-01:
Hey, yes, unfortunately we removed some old packages from the channel. We can try to restore them, but I would suggest to move to Python 3.8 or 3.9. Would that work for you?

adammacudzinski, 2022-02-01:
No unfortunately we're bound to Python3.6 for the immediate future. If you can re-upload these to the channel temporarily (or just send them over directly), we can copy them over to our internal artifactory, and won't have to rely on robostack.

Tobias-Fischer, 2022-02-01:
Hi @adammacudzinski - sorry for causing these issues, I would not have thought that anyone is still using these really old packages (to be honest I thought they broke a long time ago).

If you still have the packages installed on a machine, you can recover them from the `/path/to/mambaforge/pkgs` folder. Does that help? If you have them around and want to put them on a `robostack-old` channel or something like that, this would be great.

adammacudzinski, 2022-02-01:
Unfortunately I don't think we have these around still since our CICD env is stateless. Would you be able to rebuild and re-upload?

Tobias-Fischer, 2022-02-01:
A rebuild takes considerable time unfortunately - I'm still doing the Python 3.9 builds. So unfortunately I cannot build them (maybe someone else offers help), but you could build them yourself quite easily. Maybe someone else got the Python 3.6 packages around and could help out (@baszalmstra @ruben-arts)

baszalmstra, 2022-02-02:
No sorry we only ever used the Python 3.8 packages so I wont have those cached anywhere either.

Speaking of the global rebuild does it target Python3.9 only?

Tobias-Fischer, 2022-02-02:
Yep, only py39.

wolfv, 2022-02-02:
Hi @adammacudzinski I have just started to upload python 3.6 packages from an backup to a new channel (`robostack-old`)

Can you refer to that channel? https://anaconda.org/robostack-old/repo

Tobias-Fischer, 2022-02-02:
Thanks @wolfv - it seems like the upload finished, so closing here. Feel free to reopen @adammacudzinski

adammacudzinski, 2022-02-04:
@wolfv @Tobias-Fischer Thanks for your help on this! I just confirmed that it works on our end.