RoboStack/ros-noetic #218: Struggling with ros-noetic-turtlebot3 between arch/python versions

sylarevan, 2022-02-02:
Hello,

I'm trying to install `ros-noetic-turtlebot3` on my mac M1. But with a "native" arm64 environment, this package is not available so far. So I switched to a rosetta/osx-64 environment. But now I get :
- with python 3.9:  `ros-noetic-desktop` installs fine, but then:`ros-noetic-turtlebot3-1.2.5-py38h7e910e5_5 requires python 3.8.*, but none of the providers can be installed`
- with python 3.8: `ros-noetic-desktop` can not be installed: `ros-noetic-desktop-1.5.0-py38hb43b470_13 requires ros-noetic-joint-state-publisher-gui, but none of the providers can be installed`

Is this just a matter of time before `ros-noetic-turtlebot3` is available with python 3.9 (on osx-64 arch at least)? I do not need any specific python version for my use case, the installation is for students who will be working with very  basic ros nodes and functionalities on a turtlebot3 burger.

Tobias-Fischer, 2022-02-04:
It's now available for osx-arm64. Could you please try again? Feel free to reopen if it doesn't work.

sylarevan, 2022-02-04:
Hi @Tobias-Fischer 

I can confirm `ros-noetic-turtlebot3` now installs fine for the arm64 architecture. Not tested yet though.
There is still an issue with the osx-64 architecture, as it still depends on python 3.8, which itself does not permit the `ros-noetic-desktop` installation as per the error messages above. So I'm not able to install the very same environment on an intel mac for now.

Tobias-Fischer, 2022-02-04:
No gazebo for osx-arm64 at the moment, see our gitter for a problem description if you want to help out.

turtlebot3 is now available for osx-64. To get the gazebo packages and other packages that you also want, please open a PR.

closed 2022-02-04 by Tobias-Fischer

sylarevan, 2022-02-04:
@Tobias-Fischer Thanks for all!
My students are still reporting some issues installing ros-noetic-desktop-full on osx64 arch. I will take a look first and come back (maybe on gitter then) if needed.

Tobias-Fischer, 2022-02-04:
For desktop-full please continue using python=3.8 - I just tried `mamba create -n testtest --dry-run ros-noetic-desktop-full python=3.8` and that works. For python=3.9 we're still working on some issues related to gazebo (see https://github.com/conda-forge/gazebo-feedstock/issues/57).

sylarevan, 2022-02-04:
I just ran this quick test (on a M1 mac):
```
CONDA_SUBDIR=osx-64 conda create -n robot_rosetta_3.8 python=3.8 -c conda-forge
conda activate  robot_rosetta_3.8
conda env config vars set CONDA_SUBDIR=osx-64
conda deactivate
conda activate  robot_rosetta_3.8
conda config --env --add channels conda-forge
conda config --env --add channels robostack
conda config --env --add channels robostack-experimental
conda config --env --set channel_priority strict
mamba install ros-noetic-desktop-full
```

This results in `package ros-noetic-desktop-full-1.5.0-py38hb43b470_13 requires ros-noetic-desktop, but none of the providers can be installed`.

Trying `mamba install ros-noetic-desktop` instead results in `package ros-noetic-desktop-1.5.0-py38hb43b470_13 requires ros-noetic-joint-state-publisher-gui, but none of the providers can be installed` which was the issue my students raised.

Tobias-Fischer, 2022-02-04:
I think that's a bug in mamba @wolfv

Tobias-Fischer, 2022-02-05:
Out of interest, could you try the same commands but without `conda config --env --set channel_priority strict` please?

Tobias-Fischer, 2022-02-05:
And another version, where instead of `mamba install ros-noetic-desktop-full` try `mamba install ros-noetic-desktop-full python=3.8`

sylarevan, 2022-02-05:
> Out of interest, could you try the same commands but without `conda config --env --set channel_priority strict` please?

This works well. As soon as I set channel priority with `conda config --env --set channel_priority strict` I get the error I mentionned before.

If I now use `mamba install ros-noetic-desktop-full python=3.8` **with** the channel priority set to strict, the same error appears. Without strict priority it works weel too.

wolfv, 2022-02-05:
Hm, I'd be curious what happens if you run `CONDA_SUBDIR=osx-64 conda create -n robot_rosetta_3.8 python=3.8 "openssl=1.1.1*" -c conda-forge`

It may be the case that mamba would have to downgrade openssl. 

@Tobias-Fischer maybe we should make installation a one-liner instead? Installing `mamba create -n python=3.8 ros-noetic-desktop-full -c robostack -c conda-forge --override-channels` might find a solution in a more straight forward way?

We can then still configure the channels into the environments config file (or instruct how to configure them globally).

wolfv, 2022-02-05:
Or -- another idea -- maybe the problem is the strict priority!
I just saw we have a `gazebo` package in the robostack channel now (although only for linux-aarch64). Do we also have other packages that are also in conda-forge? Because with strict prio, the conda-forge versions are completely forbidden.

Do we also have our own `qt` in robostack or robostack-experimental?

wolfv, 2022-02-05:
Btw. I tested locally.

This doesn't work `CONDA_SUBDIR=osx-64 mamba create -n sjsjs ros-noetic-desktop-full python=3.8 -c robostack -c robostack-experimental -c conda-forge --strict-channel-priority --override-channels`

Thsi works:

`CONDA_SUBDIR=osx-64 mamba create -n sjsjs ros-noetic-desktop-full python=3.8 -c robostack -c conda-forge --strict-channel-priority --override-channels`

So my hunch is that some package in robostack-experimental are overwriting some essential dependency otherwise found in conda-forge.

wolfv, 2022-02-05:
So robostack-experimental contains `pyqt`, `sip` and some other packages. I guess it's better to not mix `robostack` and `robostack-experimental` at this point.

wolfv, 2022-02-05:
Another way to circumvent the problem is to use `--no-channel-priority`

e.g. `CONDA_SUBDIR=osx-64 mamba create -n sjsjs ros-noetic-desktop-full python=3.8 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels`

Tobias-Fischer, 2022-02-05:
Happy to change the readme to that one liner if you are @wolfv

sylarevan, 2022-02-05:
Well, at least I can confirm that without the strict priority, everything is working fine on a mac M1 with Rosetta, including Turtlebot 3 simulations on Gazebo. I am very happy to have found RoboStack, we will certainly use it now instead of our Ubuntu VM approach for our ROS teaching unit. Last step: test everything on Windows! Anyway, thanks for your work.

wolfv, 2022-02-05:
Thanks for the kind words and sorry for the trouble @sylarevan ! I hope it continues to work well for you!