RoboStack/ros-noetic #220: Scene and environment renders on gazebo but the robot arm doesn't

tohsin, 2022-02-06:
I am trying to run the simple arm package from the udacity ros course.
`https://github.com/udacity/simple_arm_01.git simple_arm`
The scene shows up but the  robot arm won't render in gazebo.
i get this warning at the end, not sure if thats the problem but i have checked how to fix it and the solutions didnt work
`Controller Spawner couldn't find the expected controller_manager ROS interface`
I use a mac

wolfv, 2022-02-06:
You may miss the `ros-noetic-gazebo-ros-control` package?  

https://answers.ros.org/question/256820/warn-controller-spawner-couldnt-find-the-expected-controller_manager-ros-interface/

tohsin, 2022-02-06:
Yeah i tried that with `mamba install ros-noetic-gazebo-ros-control` the arm  still wont render and the warning is still there

Tobias-Fischer, 2022-09-26:
Is that still an issue @tohsin? Which platform are you on?

Tobias-Fischer, 2022-10-20:
Closing here as we don't have sufficient information. Feel free to reopen @tohsin

closed 2022-10-20 by Tobias-Fischer