RoboStack/ros-noetic #222: Turtlesim node not rendering but running

ibkvictor, 2022-02-08:
# The Turtlesim package is not displaying a window #192
I noticed while running the bash commands from the [official ROS tutorial](http://wiki.ros.org/ROS/Tutorials).
```bash
rosrun turtlesim turtlesim_node
```

**Similar issues** were observed for the packages:
rqt_console and rqt_logger_level

---
## Environment info
     active environment : ros_env
    active env location : /Users/user/miniconda/envs/ros_env
            shell level : 2
       user config file : /Users/user/.condarc
 populated config files : /Users/user/miniconda/envs/ros_env/.condarc
          conda version : 4.11.0
    conda-build version : not installed
         python version : 3.9.5.final.0
       virtual packages : __osx=10.16=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/user/miniconda  (writable)
      conda av data dir : /Users/user/miniconda/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-experimental/osx-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/user/miniconda/pkgs
                          /Users/user/.conda/pkgs
       envs directories : /Users/user/miniconda/envs
                          /Users/user/.conda/envs
               platform : osx-64
             user-agent : conda/4.11.0 requests/2.25.1 CPython/3.9.5 Darwin/20.4.0 OSX/10.16
                UID:GID : 501:20
             netrc file : None
           offline mode : False

Tobias-Fischer, 2022-02-08:
`conda list`?

ibkvictor, 2022-02-08:
`Conda list`
---
```
# packages in environment at /Users/user/miniconda/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        41.0                 h694c41f_1    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.0.1                h1224e73_7    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py38h96a0964_2    conda-forge
boost                     1.74.0           py38hb0f0857_5    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
brotli                    1.0.9                h0d85af4_6    conda-forge
brotli-bin                1.0.9                h0d85af4_6    conda-forge
bullet-cpp                3.17                 h06619e7_1    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.3.0                h0d85af4_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1              hd9211c8_5    conda-forge
cctools_osx-64            973.0.1              hb41342b_5    conda-forge
certifi                   2021.10.8        py38h50d1736_1    conda-forge
cffi                      1.15.0           py38h1a44b6c_0    conda-forge
cfitsio                   3.470                h01dc385_7    conda-forge
clang                     11.1.0               h694c41f_1    conda-forge
clang-11                  11.1.0          default_he082bbe_1    conda-forge
clang_osx-64              11.1.0               hc8532ff_6    conda-forge
clangxx                   11.1.0          default_he082bbe_1    conda-forge
clangxx_osx-64            11.1.0               h2c55bc3_6    conda-forge
cmake                     3.21.3               hdbd2f3e_0    conda-forge
compiler-rt               11.1.0               h654b07c_0    conda-forge
compiler-rt_osx-64        11.1.0               h8c5fa43_0    conda-forge
compilers                 1.3.0                h694c41f_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
cppzmq                    4.8.1                hf09c4ce_0    conda-forge
cryptography              36.0.1           py38h56c4533_0    conda-forge
curl                      7.81.0               hf45b732_0    conda-forge
cxx-compiler              1.3.0                h940c156_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.9.5                he0c73e0_5    conda-forge
dbus                      1.13.6               h811a1a6_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py38h50d1736_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.9                h0d85af4_0    conda-forge
expat                     2.4.4                he49afe7_0    conda-forge
fcl                       0.6.1                h31ff308_3    conda-forge
ffmpeg                    4.3.2                h4dad6da_1    conda-forge
flann                     1.9.1             h4541d04_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.94              h10f422b_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.29.1           py38h96a0964_0    conda-forge
fortran-compiler          1.3.0                h6818c3e_0    conda-forge
freeimage                 3.18.0               h7d96ac2_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
freexl                    1.0.6                h0d85af4_0    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.5.0               hc02bdab_1    conda-forge
gdbm                      1.18                 h8a0c380_2    conda-forge
[... 468 lines cut ...]
xorg-randrproto           1.5.0             h0d85af4_1001    conda-forge
xorg-recordproto          1.14.2            h0d85af4_1002    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-util-macros          1.19.3               h35c211d_0    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xineramaproto        1.2.1             h0d85af4_1001    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.5                haf1e3a3_1    conda-forge
[... cut ...]
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.11            h9173be1_1013    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

`Conda list`
---
```
[ INFO] [1644326881.864937000]: Starting turtlesim with node name /turtlesim
[ INFO] [1644326881.875156000]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```
** Appears to create a window but does not run**

[image]

Tobias-Fischer, 2022-02-08:
Hi @ibkvictor - please update to ros-distro-mutex=0.3 and python=3.9. You are still using an old build, which is known to have this issue.

closed 2022-02-08 by Tobias-Fischer