RoboStack/ros-noetic #224: Request: Rebuild ros-noetic-jackal-simulator, gazebo etc. for linux-64 Python3.8

mharradon, 2022-02-09:
Thank you!

Tobias-Fischer, 2022-02-09:
Unfortunately this is not going to happen. We have it for Python 3.9 - what's wrong with these?.

If you still have the packages somewhere, I'm happy to upload them.

mharradon, 2022-02-09:
Currently I have a bunch of recipes that can't resolve anymore. I don't have upper limits on any of my dependencies, so if this can't be rebuilt I'll have to try to figure out what other dependencies need to be updated and probably push fixes upstream.

Tobias-Fischer, 2022-02-09:
Don't get me wrong - they can be rebuilt, and you can quite easily do it locally following the instructions in https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md but we can't do it here due to time constraints ..

Let us know if we can help with upstream patches.

closed 2022-02-09 by Tobias-Fischer

mharradon, 2022-02-09:
I have narrowed down the conflict to an incompatibility between `tensorflow-gpu` and `ros-noetic-jackal-simulator` on linux-64. I don't know if that's of any significant priority to your team, but FYI.

```
(base) root@fe701a50d3cd:/# mamba create --dry-run --name ros2 tensorflow-gpu ros-noetic-jackal-simulator

... MAMBA BANNER ...

Looking for: ['tensorflow-gpu', 'ros-noetic-jackal-simulator']

conda-forge/linux-64     Using cache
conda-forge/noarch       Using cache
pkgs/main/linux-64       Using cache
pkgs/main/noarch         Using cache
pkgs/r/linux-64          Using cache
pkgs/r/noarch            Using cache
robostack/noarch         [====================] (00m:00s) No change
robostack/linux-64       [====================] (00m:00s) No change
Encountered problems while solving:
  - nothing provides __cuda needed by tensorflow-2.7.0-cuda102py310hcf4adbc_0
  ```
  
Installing `tensorflow-gpu` directly resolves fine as well as `tensorflow-gpu` with `ros-noetic-desktop-full`. Channel solve is set to flexible. I've dealt enough with solve errors to know not to read into the `__cuda` error too much.

I am pursuing building the linux-x64 python3.8 version of `ros-noetic-jackal-`.

wolfv, 2022-02-09:
@mharradon that's not a problem with our packages, it's a problem with your system. Do you have a NVIDIA GPU?
What is the output of `nvidia-smi`?

wolfv, 2022-02-09:
Or can you post the outptu of `conda/mamba info`?

Tobias-Fischer, 2022-02-09:
I think you might need to specify python=3.9

mharradon, 2022-02-09:
conda info:

```
     active environment : base
    active env location : /var/overmind/miniconda
            shell level : 1
       user config file : /root/.condarc
 populated config files : /root/.condarc
          conda version : 4.11.0
    conda-build version : 3.21.8
         python version : 3.7.4.final.0
       virtual packages : __linux=4.19.76=0
                          __glibc=2.23=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /var/overmind/miniconda  (writable)
      conda av data dir : /var/overmind/miniconda/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/linux-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /var/overmind/miniconda/pkgs
                          /root/.conda/pkgs
       envs directories : /var/overmind/miniconda/envs
                          /root/.conda/envs
               platform : linux-64
             user-agent : conda/4.11.0 requests/2.27.1 CPython/3.7.4 Linux/4.19.76-linuxkit ubuntu/16.04.7 glibc/2.23
                UID:GID : 0:0
             netrc file : None
           offline mode : False     
 ```
 
 This is in a Docker container without a physical GPU currently. Typically I am able to install tensorflow-gpu without a physical GPU attached (nvidia drivers etc. are installed). E.g. I can still create an environment just with `tensorflow-gpu`. This gets me: 
 
   `+ tensorflow-gpu                  2.6.2  cuda110py39h5b0ac8e_1   conda-forge/linux-64       29 KB`
   
I will try pinning python3.9.

mharradon, 2022-02-09:
`mamba create --dry-run --name ros2 python=3.9 tensorflow-gpu ros-noetic-jackal-simulator` has the same result.

mharradon, 2022-02-09:
I am currently attempting updating my Docker image from CUDA-10.0 to CUDA-11.2.2. My guess is that __cuda detection doesn't work on 10.0. The Docker image is based on `nvidia/cudagl:11.2.2-devel-ubuntu16.04`.

wolfv, 2022-02-09:
You can use `CONDA_CUDA_OVERRIDE=11.2 mamba ...` or other ways to specify this environment variable if you _really_ want the GPU version (whihc is 1.8Gb larger in size than the `cpu` version). 

Here is a docs PR that addresses this: https://github.com/conda-forge/conda-forge.github.io/pull/1601/files

mharradon, 2022-02-09:
Thanks! I may give that a try. My use case is building Docker images that will be run on a GPU but they may be built in a variety of places (e.g. locally, CI/CD).

Is GPU detection literally detection of attached GPU hardware? Or does it detect the existence of nvidia drivers / CUDA libraries? The latter might cover most of the need for that override.

wolfv, 2022-02-09:
With conda they do check for the `cuda.so` https://github.com/conda/conda/blob/35cd47952fe694b49375a72bdbaf7834570fdc2d/conda/common/cuda.py#L5

I think using an override is the safe bet in these cases.