RoboStack/ros-noetic #225: Missing icons on rqt applications on Ubuntu 20.04

agtbaskara, 2022-02-09:
I notice that when running rqt apps, like rqt_bag most of the icons are missing.

I am running a fresh installation of Ubuntu 20.04

Here is a screenshot of rqt_bag with missing icons:

[image]

For now, the temporary solution for me is to install `tango-icon-theme` on the system manually using `sudo apt install tango-icon-theme`

I reference this for the temporary solution [https://answers.ros.org/question/372890/rqt-button-not-show-icon/](url)

Here is a screenshot of rqt_bag after installing `tango-icon-theme` : 

[image]

Notice that there is still some icons are still missing like the thumbnail icon

Tobias-Fischer, 2022-02-09:
Can you try mamba install -c conda-forge adwaita-icon-theme?

agtbaskara, 2022-02-09:
I have tried that before finding the `tango-icon-theme` solution, and It doesn't help in my case.

Probably we need to create a new conda package for `tango-icon-theme` or set some system variable in the conda ros env, so it uses `adwaita-icon-theme` as default?

Tobias-Fischer, 2022-02-09:
How about `conda install -c robostack-experimental ros-galactic-tango-icons-vendor`?

agtbaskara, 2022-02-09:
I have tested it, and it doesn't work (I uninstall the tango-icon-theme from the system and reboot the system before testing)

wolfv, 2022-02-09:
weird, in your second screenshot icons that existed in teh first screenshot have gone missing.

agtbaskara, 2022-02-09:
Probably because `tango-icon-theme` itself is incomplete

agtbaskara, 2022-02-09:
Interestingly I tried to install `gnome-icon-theme` by using `sudo apt install gnome-icon-theme` and all the icons are now appears

Here is a screenshot of rqt_bag after installing `gnome-icon-theme` :
[image]

wolfv, 2022-02-09:
Great! This is really good to know.
We need to figure out if `adwaita-icon-theme` on conda-forge covers the same icons or what's missing.

agtbaskara, 2022-02-09:
I already tried to install `adwaita-icon-theme` from conda-forge before (`mamba install -c conda-forge adwaita-icon-theme`)
But the icons are still missing (same as my first screenshot)

Probably this is because rqt use `gnome-icon-theme` rather than `adwaita-icon-theme` as default icon

Tobias-Fischer, 2022-04-03:
This also happens on other platforms unfortunately - and there's no way workaround on osx

Tobias-Fischer, 2023-02-18:
Xref https://github.com/conda-forge/qt-main-feedstock/issues/120
@hmaarrfk has done some great work for better system integration of qt, I wonder if this will help here?

hmaarrfk, 2023-02-18:
i'm not so sure. I haven't really hit any of these issues myself. sometimes starting things from the terminal really helps.

Tobias-Fischer, 2023-02-19:
Do you know if this is an issue on windows as well @traversaro?

Tobias-Fischer, 2023-02-19:
I’m trying to build the old gnome icon theme to see if this fixes the issue: https://github.com/conda-forge/staged-recipes/pull/22077

Tobias-Fischer, 2023-02-20:
Alright .. after going through all the trouble in building gnome-icon-theme, I found that this does not fix the issue after all. However, it sent me off the right path. Basically, QIcon isn't looking in the right directories by default. I patched python-qt-binding to look into the conda_prefix: https://github.com/RoboStack/ros-noetic/commit/e94f566d4972dd128fb684856df837c4e4edf4b8

Then all that is needed is installing the proper theme, which I enforce in https://github.com/RoboStack/ros-noetic/commit/3b40e3df393011378720d73c601de0bb4070d3aa (thankfully the themes are available for all platforms).

This fixes the issue locally on my MacOS machine. Could you please try it on your Linux machine @agtbaskara? You might need to recreate the env, I've overwritten the ros-noetic-python-qt-binding package with a new one that contains the patch.

Closing here, feel free to reopen.

closed 2023-02-20 by Tobias-Fischer

hmaarrfk, 2023-02-20:
ah good find!

traversaro, 2026-05-11:
Interesting, depending on which version of `adwaita-icon-theme` version did this change, I guess this problem may also affect ROS-O people (https://github.com/ros-o), we may check if they have some useful patch.