RoboStack/ros-noetic #227: rviz error: Failed to validate dynamic code signature of accessing process

sylarevan, 2022-02-09:
Hello,

rviz does not run after a clean `ros-noetic-desktop-full` install with python 3.9 on osx64 arch.
Launching rviz from the terminal results in:
```
[ INFO] [1644422490.110950000]: rviz version 1.14.13
[ INFO] [1644422490.110981000]: compiled against Qt version 5.12.9
[ INFO] [1644422490.110989000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1644422490.137024000]: Forcing OpenGl version 0.
```
and that's all. The icon in the dock indicates "program is not responsing".
After a quick look inside the Console.app, I can find some intriguing errors about code signature:
```
erreur	17:06:36.223314+0100	tccd	Failed to validate dynamic code signature of accessing process <TCCDProcess: identifier=<ID of InvalidCode>, pid=37618, auid=501, euid=501, binary_path=/usr/local/Caskroom/mambaforge/base/envs/robot_3.9/bin/rviz>: #-67062: Error Domain=NSOSStatusErrorDomain Code=-67062 "(null)"
erreur	17:06:36.230954+0100	tccd	Failed to validate dynamic code signature of accessing process <TCCDProcess: identifier=<ID of InvalidCode>, pid=37618, auid=501, euid=501, binary_path=/usr/local/Caskroom/mambaforge/base/envs/robot_3.9/bin/rviz>: #-67062: Error Domain=NSOSStatusErrorDomain Code=-67062 "(null)"
```

But I don't know if this is the root cause of rviz apparently crashing. 
Launching  rviz with the verbose option does not give any more information.

wolfv, 2022-02-09:
hmm. it sounds liek a codesigning issue. 

What happens if you do `codesign -s - -f /usr/local/Caskroom/mambaforge/base/envs/robot_3.9/bin/rviz`?
That signs the binary it with an ad-hoc certificate.

sylarevan, 2022-02-09:
After signing rviz with `codesign -s - -f /opt/homebrew/Caskroom/mambaforge/base/envs/robot_rosetta_3.9/bin/rviz` rviz still refuses to run.
In Console.app, I now get the following error messages (note that the ones related to code signature are not there anymore):
```
erreur	20:30:16.621433+0100	diagnostics_agent	Error reading receipt: Error Domain=NSCocoaErrorDomain Code=256 "The file “receipt” couldn’t be opened." UserInfo={NSFilePath=/opt/homebrew/Caskroom/mambaforge/base/envs/robot_rosetta_3.9/bin/rviz/Contents/_MASReceipt/receipt, NSUnderlyingError=0x135646080 {Error Domain=NSPOSIXErrorDomain Code=20 "Not a directory"}}
erreur	20:30:16.638295+0100	rviz	error finding potential wrapper bundle for node <FSNode 0x6000010bdce0> { isDir = ?, path = <private> }: Error Domain=NSOSStatusErrorDomain Code=-10811 UserInfo={_LSLine=1579, _LSFunction=<private>}
erreur	20:30:16.638380+0100	rviz	error finding potential wrapper bundle for node <FSNode 0x6000010bdce0> { isDir = ?, path = <private> }: Error Domain=NSOSStatusErrorDomain Code=-10811 UserInfo={_LSLine=1579, _LSFunction=<private>}
erreur	20:30:16.638427+0100	rviz	error finding potential wrapper bundle for node <FSNode 0x6000010bdce0> { isDir = ?, path = <private> }: Error Domain=NSOSStatusErrorDomain Code=-10811 UserInfo={_LSLine=1579, _LSFunction=<private>}
```

wolfv, 2022-02-09:
Can you start rviz with verbose flags?

wolfv, 2022-02-09:
Did you also try the non-rosetta one?

It might be interesting to try this snippet:

```
find $CONDA_PREFIX -name ".dylib" -exec codesign -s - -f {} \;
find $CONDA_PREFIX -name ".so" -exec codesign -s - -f {} \;
```

to sign all libraries in the environment.

sylarevan, 2022-02-09:
> It might be interesting to try this snippet:
> ```
> find $CONDA_PREFIX -name ".dylib" -exec codesign -s - -f {} \;
> find $CONDA_PREFIX -name ".so" -exec codesign -s - -f {} \;
>```
> to sign all libraries in the environment.

Just tested, no changes.

> Did you also try the non-rosetta one?

Nope, as `desktop-full` was not available. It seems it is now the case :v: ... and I can confirm it works with the native version. 
Nevetheless, I tried on our intel mac computer this afternoon (so, no Rosetta), the error was exactly the same and I was not able to launch rviz.

sylarevan, 2022-02-11:
Back on my osx64 mac, I tried this time to launch rviz from `rosrun rviz rviz` and I again had an error message about code signature concerning this time `/usr/local/Caskroom/mambaforge/base/envs/robot_3.9/lib/rviz/rviz`. After signing it with `codesign` I finally get quite the same error message about missing "receipt":

`Error reading receipt: Error Domain=NSCocoaErrorDomain Code=256 "The file “receipt” couldn’t be opened." UserInfo={NSFilePath=/usr/local/Caskroom/mambaforge/base/envs/robot_3.9/lib/rviz/rviz/Contents/_MASReceipt/receipt, NSUnderlyingError=0x7fe866211bc0 {Error Domain=NSPOSIXErrorDomain Code=20 "Not a directory"}}`

(I also signed `/opt/homebrew/Caskroom/mambaforge/base/envs/robot_rosetta_3.9/bin/rviz` to be sure).

Tobias-Fischer, 2022-04-07:
Hmm very odd. On my M1 Mac, I am able to run the osx-64 (i.e. Intel) rviz package without any issues.

Could you please report your `conda list`? How did you install RoboStack? Can you try a minimal environment with just `mamba create -n testenv ros-noetic-rviz`?

Tobias-Fischer, 2023-01-24:
Closing - new packages are available in the `robostack-staging` channel, if the error still persists, please reopen.

closed 2023-01-24 by Tobias-Fischer

entatic, 2025-03-11:
Rviz still crashing on M1 arm Sonoma. 

➜  ~ mamba install ros-noetic-desktop

Looking for: ['ros-noetic-desktop']

robostack-noetic/osx-arm64 (check zst)              Checked  0.3s
robostack-noetic/noarch (check zst)                 Checked  0.4s
conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
robostack-noetic/noarch                            116.0 B @ 586.0 B/s  0.2s
robostack-noetic/osx-arm64                          57.7kB @ 140.8kB/s  0.4s

(ros_env) ➜  ros_env rviz
[ INFO] [1741732362.730280000]: rviz version 1.14.25
[ INFO] [1741732362.730822000]: compiled against Qt version 5.15.15
[ INFO] [1741732362.730827000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1741732362.756299000]: Forcing OpenGl version 0.
[ INFO] [1741732363.232183000]: Stereo is NOT SUPPORTED
[ INFO] [1741732363.232209000]: OpenGL device: Apple M1
[ INFO] [1741732363.232214000]: OpenGl version: 2.1 (GLSL 1.2).
[1]    86856 killed     rviz

The message in the Console is "CODE SIGNING: cs_invalid_page(0x120048000): p=86856[rviz] final status 0x23000200, denying page sending SIGKILL". (ros_env)

Tobias-Fischer, 2025-03-11:
Hi @entatic - could you please create a new issue and fill all requested information?