RoboStack/ros-noetic #228: Unable to use laser and camera in Gazebo

sylarevan, 2022-02-09:
Hello (again),

I have a strange issue with gazebo and its related laser (`libgazebo_ros_laser`) and camera (`libgazebo_ros_camera`) plugins. While they are correctly integrated inside URDF/gazebo files, both do not seem to be loaded by gazebo at launch. There is no issue for the imu (`libgazebo_ros_imu`) and the controller (`libgazebo_ros_diff_drive`) plugins.

For instance, using `roslaunch turtlebot3_gazebo turtlebot3_autorace_2020.launch` I get (forget about the usual spawning crash):
```
SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [38737]
ROS_MASTER_URI=http://127.0.0.1:11311

setting /run_id to 2ad81b68-89c5-11ec-a4d7-804a145fa150
process[rosout-1]: started with pid [38738]
started core service [/rosout]
process[gazebo-2]: started with pid [38739]
process[gazebo_gui-3]: started with pid [38740]
process[spawn_urdf-4]: started with pid [38745]
[ INFO] [1644424057.808790000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1644424057.809797000]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1644424058.137159000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1644424058.138287000]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1644424059.154249000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ERROR] [1644424059.460505, 116.346000]: Spawn service failed. Exiting.
[ INFO] [1644424059.466605000, 116.346000000]: Physics dynamic reconfigure ready.
Node::Advertise(): Error advertising topic [default/autorace/base_footprint/camera/image]. Did you forget to start the discovery service?
[spawn_urdf-4] process has died [pid 38745, exit code 1, cmd /usr/local/Caskroom/mambaforge/base/envs/robot_3.9/lib/gazebo_ros/spawn_model -urdf -model autorace -x 0.8 -y -1.747 -z 0 -param robot_description __name:=spawn_urdf __log:=/Users/sylar/.ros/log/2ad81b68-89c5-11ec-a4d7-804a145fa150/spawn_urdf-4.log].
log file: /Users/sylar/.ros/log/2ad81b68-89c5-11ec-a4d7-804a145fa150/spawn_urdf-4*.log
[ INFO] [1644424061.860900000, 116.346000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1644424061.861737000, 116.346000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1644424061.931314000, 116.346000000]: DiffDrive(ns = //): <tf_prefix> =
[ INFO] [1644424061.932385000, 116.346000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1644424061.934184000, 116.346000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1644424061.936582000, 116.346000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1644424061.937205000, 116.346000000]: DiffDrive(ns = //): Advertise odom on odom
[ INFO] [1644424062.003461000, 116.347000000]: imu plugin missing <frameName>, defaults to <bodyName>
```

The very same command, launched from an Ubuntu VM (but with ROS Noetic installed from APT) results in:
```
SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [7444]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 83428748-898e-11ec-906d-831cee59c044
process[rosout-1]: started with pid [7454]
started core service [/rosout]
process[gazebo-2]: started with pid [7457]
process[gazebo_gui-3]: started with pid [7459]
process[spawn_urdf-4]: started with pid [7467]
[ INFO] [1644400573.641938772]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1644400573.642768245]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1644400573.736290522]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1644400573.737340822]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1644400574.319344540]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1644400574.339049247]: Physics dynamic reconfigure ready.
Node::Advertise(): Error advertising topic [default/autorace/base_footprint/camera/image]. Did you forget to start the discovery service?
[ INFO] [1644400575.308380209, 116.483000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1644400575.313666502, 116.483000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1644400575.825867125, 116.483000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1644400575.826078160, 116.483000000]: Starting Laser Plugin (ns = /)
[ INFO] [1644400575.837676380, 116.483000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1644400575.928633531, 116.483000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1644400575.928790943, 116.483000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1644400575.929846701, 116.483000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1644400575.930332394, 116.483000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1644400575.930683175, 116.483000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1644400575.931963177, 116.483000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1644400575.932196711, 116.483000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1644400575.941963613, 116.483000000]: imu plugin missing <frameName>, defaults to <bodyName>
```
As you can see, Laser and Camera plugins are here correctly launched.
FWIW, I have compared the URDF/Gazebo files in both configurations, and they are completely identical. I have also tested with a usually working personal URDF/Gazebo description and the problem is the same.

But while the camera and laser topics are not available for ROS, camera and laser simulations are correctly performed by Gazebo, as I can see the laser scans and the camera images inside Gazebo.

So far I have not been able to debug anything, and I have not spotted any error message anywhere. Any idea where I could find some indications about this issue? Thanks!

sylarevan, 2022-02-09:
FWIW, I just tried on the native (arm64) version, same problem: no camera or laser topics available.

wolfv, 2022-02-09:
Do you have the log file from Greg spawn_urdf node?

sylarevan, 2022-02-09:
Here is what I have in `~/.ros/log/latest/spawn_urdf-4.log`

```
[rospy.client][INFO] 2022-02-09 21:59:10,038: init_node, name[/spawn_urdf], pid[93473]
[xmlrpc][INFO] 2022-02-09 21:59:10,038: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2022-02-09 21:59:10,038: Started XML-RPC server [http://macsylvain.home:58439/]
[rospy.impl.masterslave][INFO] 2022-02-09 21:59:10,038: _ready: http://macsylvain.home:58439/
[rospy.registration][INFO] 2022-02-09 21:59:10,038: Registering with master node http://localhost:11311
[rospy.init][INFO] 2022-02-09 21:59:10,038: ROS Slave URI: [http://macsylvain.home:58439/]
[rospy.init][INFO] 2022-02-09 21:59:10,038: registered with master
[xmlrpc][INFO] 2022-02-09 21:59:10,038: xml rpc node: starting XML-RPC server
[rospy.rosout][INFO] 2022-02-09 21:59:10,039: initializing /rosout core topic
[rospy.rosout][INFO] 2022-02-09 21:59:10,042: connected to core topic /rosout
[rospy.simtime][INFO] 2022-02-09 21:59:10,042: initializing /clock core topic
[rospy.simtime][INFO] 2022-02-09 21:59:10,043: connected to core topic /clock
[rosout][INFO] 2022-02-09 21:59:10,044: Loading model XML from ros parameter robot_description
[rosout][INFO] 2022-02-09 21:59:10,047: Waiting for service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2022-02-09 21:59:10,264: topic[/rosout] adding connection to [/rosout], count 0
[rosout][INFO] 2022-02-09 21:59:11,259: Calling service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2022-02-09 21:59:11,299: topic[/clock] adding connection to [http://macsylvain.home:58450/], count 0
[rosout][INFO] 2022-02-09 21:59:12,228: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2022-02-09 21:59:12,229: signal_shutdown [atexit]
[rospy.internal][INFO] 2022-02-09 21:59:12,230: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2022-02-09 21:59:12,230: topic[/clock] removing connection to http://macsylvain.home:58450/
[rospy.impl.masterslave][INFO] 2022-02-09 21:59:12,230: atexit
```

And in `spawn_urdf-4-stdout.log`:
```
[INFO] [1644440350.044452, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1644440350.047291, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1644440351.259239, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1644440352.228459, 116.507000]: Spawn status: SpawnModel: Successfully spawned entity
```

sylarevan, 2022-02-10:
One precision maybe: the spawn_urdf crash (in the first post) is not the cause of the laser/camera missing topics, most of the time it launches just fine.

I have tested some small things. If I change the definition of the gazebo laser plugin in the gazebo description to 
```
      <plugin filename="libgazebo_ros_laser_xxxxxxxx.so" name="gazebo_ros_lds_lfcd_controller">
        <topicName>scan</topicName>
        <frameName>base_scan</frameName>
      </plugin>
```
where obvioulsy `libgazebo_ros_laser_xxxxxxxx.so` does not exist, there is absolutely no error message anywhere. It just seems the plugin declaration is not taken into account at all (which would explain why there is no /scan topic).

Then, I evaluated the output of xacro used in the `robot_description` parameter:
```
  <param name="robot_description" command="$(find xacro)/xacro --inorder $(find turtlebot3_description)/urdf/turtlebot3_burger_for_autorace_2020.urdf.xacro" />
```
And nothing special here, the plugin definition are still here in the parameter.

Well, in the end, I still do not have any idea why the laser and camera ROS plugins are not taken into account by gazebo. Maybe a gazebo bug?

wolfv, 2022-02-10:
Maybe you can try to give the absolute path to the `libgazebo_ros_laser...so` file?

sylarevan, 2022-02-10:
Unfortunatly, no effect ...
```
<plugin name="gazebo_ros_lds_lfcd_controller" filename="/opt/homebrew/Caskroom/mambaforge/base/envs/robot_3.9/lib/libgazebo_ros_laser.so">
   <topicName>scan</topicName>
   <frameName>base_scan</frameName>
</plugin>
```

wolfv, 2022-02-10:
just making sure: this file exists on your computer, right?
I wonder if `gazebo` hard-codes `.dylib` as extension (which would be the natural one on macOS), or if this is related to other "PREFIX" replacement problems. Maybe @traversaro knows something :)

sylarevan, 2022-02-10:
> just making sure: this file exists on your computer, right?

Yep :sweat_smile: :
```
(robot_3.9) sylar@macsylvain /opt/homebrew/Caskroom/mambaforge/base/envs/robot_3.9/lib> ll -h libgazebo_ros*
-rwxrwxr-x  1 sylar  admin   1,3M  9 fév 21:19 libgazebo_ros_api_plugin.dylib
-rwxrwxr-x  1 sylar  admin   307K  9 fév 21:19 libgazebo_ros_block_laser.dylib
-rwxrwxr-x  1 sylar  admin   262K  9 fév 21:19 libgazebo_ros_bumper.dylib
-rwxrwxr-x  1 sylar  admin   189K  9 fév 21:19 libgazebo_ros_camera.dylib
-rwxrwxr-x  1 sylar  admin   528K  9 fév 21:19 libgazebo_ros_camera_utils.dylib
-rwxrwxr-x  1 sylar  admin   423K  9 fév 21:19 libgazebo_ros_control.dylib
-rwxrwxr-x  1 sylar  admin   303K  9 fév 21:19 libgazebo_ros_depth_camera.dylib
-rwxrwxr-x  1 sylar  admin   368K  9 fév 21:19 libgazebo_ros_diff_drive.dylib
-rwxrwxr-x  1 sylar  admin   266K  9 fév 21:19 libgazebo_ros_elevator.dylib
-rwxrwxr-x  1 sylar  admin   264K  9 fév 21:19 libgazebo_ros_f3d.dylib
-rwxrwxr-x  1 sylar  admin   290K  9 fév 21:19 libgazebo_ros_force.dylib
-rwxrwxr-x  1 sylar  admin   281K  9 fév 21:19 libgazebo_ros_ft_sensor.dylib
-rwxrwxr-x  1 sylar  admin   377K  9 fév 21:19 libgazebo_ros_gpu_laser.dylib
-rwxrwxr-x  1 sylar  admin   250K  9 fév 21:19 libgazebo_ros_hand_of_god.dylib
-rwxrwxr-x  1 sylar  admin   298K  9 fév 21:19 libgazebo_ros_harness.dylib
-rwxrwxr-x  1 sylar  admin   357K  9 fév 21:19 libgazebo_ros_imu.dylib
-rwxrwxr-x  1 sylar  admin   270K  9 fév 21:19 libgazebo_ros_imu_sensor.dylib
-rwxrwxr-x  1 sylar  admin   297K  9 fév 21:19 libgazebo_ros_joint_pose_trajectory.dylib
-rwxrwxr-x  1 sylar  admin   286K  9 fév 21:19 libgazebo_ros_joint_state_publisher.dylib
-rwxrwxr-x  1 sylar  admin   377K  9 fév 21:19 libgazebo_ros_laser.dylib
-rwxrwxr-x  1 sylar  admin   236K  9 fév 21:19 libgazebo_ros_multicamera.dylib
-rwxrwxr-x  1 sylar  admin   302K  9 fév 21:19 libgazebo_ros_openni_kinect.dylib
-rwxrwxr-x  1 sylar  admin   336K  9 fév 21:19 libgazebo_ros_p3d.dylib
-rwxrwxr-x  1 sylar  admin   157K  9 fév 21:19 libgazebo_ros_paths_plugin.dylib
-rwxrwxr-x  1 sylar  admin   318K  9 fév 21:19 libgazebo_ros_planar_move.dylib
-rwxrwxr-x  1 sylar  admin   332K  9 fév 21:19 libgazebo_ros_projector.dylib
-rwxrwxr-x  1 sylar  admin   212K  9 fév 21:19 libgazebo_ros_prosilica.dylib
-rwxrwxr-x  1 sylar  admin   281K  9 fév 21:19 libgazebo_ros_range.dylib
-rwxrwxr-x  1 sylar  admin   334K  9 fév 21:19 libgazebo_ros_skid_steer_drive.dylib
-rwxrwxr-x  1 sylar  admin   161K  9 fév 21:19 libgazebo_ros_template.dylib
-rwxrwxr-x  1 sylar  admin   367K  9 fév 21:19 libgazebo_ros_tricycle_drive.dylib
-rwxrwxr-x  1 sylar  admin   194K  9 fév 21:19 libgazebo_ros_triggered_camera.dylib
-rwxrwxr-x  1 sylar  admin   235K  9 fév 21:19 libgazebo_ros_triggered_multicamera.dylib
-rwxrwxr-x  1 sylar  admin   226K  9 fév 21:19 libgazebo_ros_utils.dylib
-rwxrwxr-x  1 sylar  admin   298K  9 fév 21:19 libgazebo_ros_vacuum_gripper.dylib
-rwxrwxr-x  1 sylar  admin   319K  9 fév 21:19 libgazebo_ros_video.dylib
-rwxrwxr-x  1 sylar  admin   374K  9 fév 21:19 libgazebo_ros_wheel_slip.dylib
```
You can find here those working (`libgazebo_ros_diff_drive` and `libgazebo_ros_imu`) and the others (`libgazebo_ros_laser` and `libgazebo_ros_camera`).

wolfv, 2022-02-10:
I think gazebo might have automatic handling for the extension. might be better to remove `.so` entirely and see if that works. Woudl explain why some of these work and some don't.

traversaro, 2022-02-10:
> I wonder if `gazebo` hard-codes `.dylib` as extension (which would be the natural one on macOS), or if this is related to other "PREFIX" replacement problems. Maybe @traversaro knows something :)

Yes, gazebo takes in input the plugin name with `.so`, and then on macOS automatically changes it to `.dylib` (if it cannot find the `.so` one: https://github.com/osrf/gazebo/blob/13d967c2a5c6c3f761a288f7a352f2ab676e8a98/gazebo/common/Plugin.hh#L198 ).

sylarevan, 2022-02-10:
I forgot to mention I have already tested both extensions `.so` or `.dylib` but this does not change anything.

Tobias-Fischer, 2022-02-11:
Setting the verbose flag leads to:
```
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_laser.so: dlopen(libgazebo_ros_laser.so, 0x0009): tried: '/Users/fischert/mambaforge/envs/robostackenv/lib/libgazebo_ros_laser.so' (no such file), '/Users/fischert/mambaforge/envs/robostackenv/bin/../lib/libgazebo_ros_laser.so' (no such file), 'libgazebo_ros_laser.so' (no such file), '/usr/local/lib/libgazebo_ros_laser.so' (no such file), '/usr/lib/libgazebo_ros_laser.so' (no such file), '/Users/fischert/.ros/libgazebo_ros_laser.so' (no such file), '/usr/local/lib/libgazebo_ros_laser.so' (no such file), '/usr/lib/libgazebo_ros_laser.so' (no such file)
Failed to load plugin /Users/fischert/mambaforge/envs/robostackenv/share/gazebo_plugins/../../lib/libgazebo_ros_laser.dylib: dlopen(/Users/fischert/mambaforge/envs/robostackenv/share/gazebo_plugins/../../lib/libgazebo_ros_laser.dylib, 0x0009): Library not loaded: @rpath/libRayPlugin.dylib
  Referenced from: /Users/fischert/mambaforge/envs/robostackenv/lib/libgazebo_ros_laser.dylib
  Reason: tried: '/Users/fischert/mambaforge/envs/robostackenv/lib/libRayPlugin.dylib' (no such file), '/Users/fischert/mambaforge/envs/robostackenv/lib/libRayPlugin.dylib' (no such file), '/Users/fischert/mambaforge/envs/robostackenv/lib/libRayPlugin.dylib' (no such file), '/Users/fischert/mambaforge/envs/robostackenv/bin/../lib/libRayPlugin.dylib' (no such file), '/usr/local/lib/libRayPlugin.dylib' (no such file), '/usr/lib/libRayPlugin.dylib' (no such file)
```

Quick fix: `ln -s $CONDA_PREFIX/lib/gazebo-11/plugins/libRayPlugin.dylib $CONDA_PREFIX/lib/libRayPlugin.dylib`

Proper fix: I have no idea .. `$CONDA_PREFIX/lib/gazebo-11/plugins/libRayPlugin.dylib` is already contained in `$GAZEBO_PLUGIN_PATH` and `$LD_LIBRARY_PATH`

sylarevan, 2022-02-11:
@Tobias-Fischer :confounded: I didn't think about adding the verbose flag to gazebo server, sorry.
I can reproduce the issue, and the same happens for `libgazebo_ros_laser.dylib`which can not find `libRayPlugin.dylib` and `libgazebo_ros_camera.so` which can not find `libCameraPlugin.dylib`

Indeed, the path to these libraries is in `$GAZEBO_PLUGIN_PATH$` and `$LD_LIBRARY_PATH$`. What is strange is that `libImuSensorPlugin.dylib` is certainly correctly found when loading `libgazebo_ros_imu.dylib` while they all share the same location.

Indeed, adding the two links to the two "missing" libraries makes everything work again.

traversaro, 2022-02-11:
> Setting the verbose flag leads to:
>
> ```
> [Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_laser.so: dlopen(libgazebo_ros_laser.so, 0x0009): tried: [...]

Random guess: can we try to add `$CONDA_PREFIX/lib/gazebo-11/plugins/` to `DYLD_LIBRARY_PATH`?

Tobias-Fischer, 2022-02-11:
I think I tried that and it didn't fix it .. will try again Monday.

sonelu, 2022-03-15:
I actually have the same issue when spawning my robot in Gazebo. I'm on an Intel MacOS under Catalina.

The conda environment is listed [here](https://github.com/RoboStack/ros-noetic/issues/237#issuecomment-1068374984).
The URDF is [here](https://github.com/SYNTHesse-GIT/mh5_common/tree/main/mh5_description/urdf).
The launch file is [here](https://github.com/SYNTHesse-GIT/mh5_common/blob/main/mh5_description/launch/gazebo.launch)

I'm only using `gazebo_ros_control`:
```
<gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
            <robotNamespace>/mh5_robot</robotNamespace>
        </plugin>

       ...
    </gazebo>
```

The log is very similar to the one [above](https://github.com/RoboStack/ros-noetic/issues/228#issuecomment-1034189429).

Tobias-Fischer, 2022-03-15:
Have you tried this workaround? https://github.com/RoboStack/ros-noetic/issues/228#issuecomment-1035689132

sonelu, 2022-03-15:
Where did you put the verbose flag to see the detailed error? (to see which dylibs to link)

Tobias-Fischer, 2022-03-15:
I think it's in the launch file.

sonelu, 2022-03-15:
I've put this in the launch file:
```
<include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="verbose" value="true" />
  </include>
```
and shows certainly more messages from gazebo, but no additional errors.

I've tried adding links as suggested for `libJointControlPlugin.dylib` and `libJointTrajectoryPlugin.dylib` considering the controllers I'm using, but that did not seem to make any difference. If you remember how to get more info from the spawner error, let me know.

traversaro, 2022-08-18:
By looking again at this issue, I suspect I may know what is going on: Gazebo installs `libRayPlugin.dylib` just as a plugin (i.e. a library that is loaded at runtime via ldopen), while then gazebo_ros_laser links it as a regular library, linked at link time. However, RayPlugin is linked without specifying its full path. as usually it is done for C++ library in CMake, see https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_plugins/CMakeLists.txt#L265 .  This that the rpath relative path is not set correctly, and it means that the runtime loader that should find and load libRayPlugin.dylib when libgazebo_ros_laser.dylib is loaded via dlopen, is not able to find it. This the reason the workaround in https://github.com/RoboStack/ros-noetic/issues/228#issuecomment-1035689132 works. Perhaps adding `$CONDA_PREFIX/lib/gazebo-11/plugins` to `DYLD_LIBRARY_PATH` may also work, but it could have unexpected side effects. 

Probably we could understand a bit more what is happening by running: 
~~~
otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib
~~~

but I do not have currently a macOS system to try this on.

traversaro, 2022-08-18:
@sonelu if this is still a problem for you, we can probably understand more also in your case by the output of:
~~~
otool -L $CONDA_PREFIX/lib/libgazebo_ros_control.dylib
~~~

traversaro, 2022-08-18:
By the way, this is the reason the issue is solved on ROS2: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/974 . Probably we could do something similar also on ROS1, i.e. adding the plugin directory to `DYLD_LIBRARY_PATH`.

sylarevan, 2022-08-20:
> Probably we could understand a bit more what is happening by running:
> 
> ```
> otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib
> ```
> 
> but I do not have currently a macOS system to try this on.

Here is the output of this command:
```
(robot_3.9) sylar@macsylvain ~> otool -L $CONDA_PREFIX/lib/libgazebo_ros_control.dylib
/opt/homebrew/Caskroom/mambaforge/base/envs/robot_3.9/lib/libgazebo_ros_control.dylib:
	@rpath/libgazebo_ros_control.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libgazebo_physics.11.dylib (compatibility version 11.0.0, current version 11.10.1)
	@rpath/libgazebo_common.11.dylib (compatibility version 11.0.0, current version 11.10.1)
	@rpath/libboost_thread.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libsdformat9.9.dylib (compatibility version 9.0.0, current version 9.3.0)
	@rpath/libignition-common3-graphics.3.dylib (compatibility version 3.0.0, current version 3.13.2)
	@rpath/libcontroller_manager.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtransmission_interface_parser.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/liburdf.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libclass_loader.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
	@rpath/libroslib.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libroscpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosconsole.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libroscpp_serialization.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librostime.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libconsole_bridge.1.0.dylib (compatibility version 1.0.0, current version 0.0.0)
	@rpath/libignition-math6.6.dylib (compatibility version 6.0.0, current version 6.10.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

Hope it helps!

Tobias-Fischer, 2023-01-25:
> By the way, this is the reason the issue is solved on ROS2: [ros-simulation/gazebo_ros_pkgs#974](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/974) (related issue: [ros-simulation/gazebo_ros_pkgs#970](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/970)). Probably we could do something similar also on ROS1, i.e. adding the plugin directory to `DYLD_LIBRARY_PATH`. I guess this is possible via [`catkin_add_env_hooks`](https://github.com/ros/catkin/blob/noetic-devel/cmake/catkin_add_env_hooks.cmake), but I am definetly not a catkin expert.

Can't we do that easily in our activate.sh script in vinca?

Tobias-Fischer, 2023-02-20:
Should be fixed in the latest Gazebo build - this will be available soon

Tobias-Fischer, 2023-02-20:
Closing here

closed 2023-02-20 by Tobias-Fischer

traversaro, 2024-04-07:
According to https://github.com/RoboStack/ros-noetic/issues/459#issuecomment-2040117599 the problem is still there.