RoboStack/ros-noetic #229: rtabmap_ros not working

nosense345, 2022-02-13:
Back to getting rtabmap to work.
versions: 
ros-noetic-rtabmap            0.20.9  py38h448f9f6_6  robostack           
ros-noetic-rtabmap           0.20.14 py38hb4a3afa_12  robostack  

starting rtabmap_ros with roslaunch as shown in the tutorial (http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) for an L515

NODES
  /rtabmap/
    icp_odometry (rtabmap_ros/icp_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

Two errors:

ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: Cannot locate node of type [rtabmapviz] in package [rtabmap_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)

and a bit later 
[ERROR] [1644766486.786762611]: Failed to load nodelet [/rtabmap/rtabmap] of type [rtabmap_ros/rtabmap] even after refreshing the cache: Failed to load library /home/klaas/miniconda3/envs/robostackenv/lib//librtabmap_sync.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/klaas/miniconda3/envs/robostackenv/lib//librtabmap_sync.so: undefined symbol: _ZN7rtabmap7Rtabmap22globalBundleAdjustmentEibif)

I can confirm the first error: rtabmap_ros contains icp_odometry (which is started) but not rtabmapviz (which is not).

What would be the way to fix these?

Thank you very much for your assistance.

Klaas

wolfv, 2022-02-13:
What operating system?

Tobias-Fischer, 2022-04-07:
I just triggered a new build (for python=3.9) with Qt enabled. I can now run rtabmap on my linux-64 machine. Feel free to reopen if this does not work for you.

closed 2022-04-07 by Tobias-Fischer