RoboStack/ros-noetic #235: mac os monterey compiler issue

MrChill, 2022-02-28:
Hi I wanted to compile a workspace and get the following compiler error:

`catkin_make
Base path: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws
Source space: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/src
Build space: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build
Devel space: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/devel
Install space: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/install
Creating symlink "/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/src/CMakeLists.txt" pointing to "/opt/anaconda3/envs/robostackenv/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/install -G Unix Makefiles" in "/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build"
####
-- The C compiler identification is Clang 12.0.0
-- The CXX compiler identification is Clang 12.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /opt/anaconda3/envs/robostackenv/bin/x86_64-apple-darwin13.4.0-clang
-- Check for working C compiler: /opt/anaconda3/envs/robostackenv/bin/x86_64-apple-darwin13.4.0-clang - broken
CMake Error at /opt/anaconda3/envs/robostackenv/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/opt/anaconda3/envs/robostackenv/bin/x86_64-apple-darwin13.4.0-clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/opt/anaconda3/envs/robostackenv/bin/make -f Makefile cmTC_b96af/fast && /opt/anaconda3/envs/robostackenv/bin/make  -f CMakeFiles/cmTC_b96af.dir/build.make CMakeFiles/cmTC_b96af.dir/build
    make[1]: Entering directory '/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_b96af.dir/testCCompiler.c.o
    /opt/anaconda3/envs/robostackenv/bin/x86_64-apple-darwin13.4.0-clang   -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -O2 -pipe -isystem /opt/anaconda3/envs/robostackenv/include  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -MD -MT CMakeFiles/cmTC_b96af.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_b96af.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_b96af.dir/testCCompiler.c.o -c /Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_b96af
    /opt/anaconda3/envs/robostackenv/bin/cmake -E cmake_link_script CMakeFiles/cmTC_b96af.dir/link.txt --verbose=1
    /opt/anaconda3/envs/robostackenv/bin/x86_64-apple-darwin13.4.0-clang -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -O2 -pipe -isystem /opt/anaconda3/envs/robostackenv/include  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/opt/anaconda3/envs/robostackenv/lib -L/opt/anaconda3/envs/robostackenv/lib  CMakeFiles/cmTC_b96af.dir/testCCompiler.c.o -o cmTC_b96af 
    ld: unsupported tapi file type '!tapi-tbd' in YAML file '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libSystem.tbd' for architecture x86_64
    clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [CMakeFiles/cmTC_b96af.dir/build.make:100: cmTC_b96af] Error 1
    make[1]: Leaving directory '/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:127: cmTC_b96af/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/Users/mos7rng/Desktop/Projects/ROS/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
`

should I change the compiler?

Tobias-Fischer, 2022-02-28:
Hi @MrChill,

I haven't seen this one before ;). Could you please `conda activate robostackenv && mamba install tapi`, clean your catkin workspace, and try again?

wolfv, 2022-02-28:
Could you post the output of `conda list`?

MrChill, 2022-02-28:
Hi @wolfv,

```
# packages in environment at /opt/anaconda3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.1                h1224e73_0    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py39h89e85a6_2    conda-forge
boost                     1.74.0           py39ha1f3e3e_5    conda-forge
boost-cpp                 1.74.0               hdbf7018_7    conda-forge
brotli                    1.0.9                h0d85af4_6    conda-forge
brotli-bin                1.0.9                h0d85af4_6    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.4.0                h774085c_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he01c77b_1009    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_1    conda-forge
cctools                   949.0.1             h9abeeb2_23  
cctools_osx-64            949.0.1             hc7db93f_23  
certifi                   2021.10.8        py39h6e9494a_1    conda-forge
cffi                      1.15.0           py39he338e87_0    conda-forge
clang                     12.0.0               hecd8cb5_2  
clang-12                  12.0.0          default_hfcad0e5_2  
clang_osx-64              12.0.0               h7ad7729_2  
clangxx                   12.0.0          default_hfcad0e5_2  
clangxx_osx-64            12.0.0               he0c6f94_2  
cmake                     3.22.2               h4d639be_0    conda-forge
compiler-rt               12.0.0               hd87ff74_2  
compiler-rt_osx-64        12.0.0               hfb26b64_2  
compilers                 1.4.0                h694c41f_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
coreutils                 9.0                  h0d85af4_0    conda-forge
cryptography              36.0.1           py39h209aa08_0    conda-forge
cxx-compiler              1.4.0                hbb4e6a2_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39h6e9494a_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.6                h96cf925_0    conda-forge
ffmpeg                    4.3.2                hbf27d7b_2    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              h10f422b_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.29.1           py39h89e85a6_0    conda-forge
fortran-compiler          1.4.0                h46a1860_0    conda-forge
freeimage                 3.18.0               h7d96ac2_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.6               h2e6141f_0    conda-forge
gettext                   0.19.8.1          hd1a6beb_1008    conda-forge
gfortran                  9.3.0               h768ea0c_15    conda-forge
gfortran_impl_osx-64      9.3.0               h9cc0e5e_23    conda-forge
gfortran_osx-64           9.3.0               h18f7dce_15    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
gmock                     1.10.0               h926bf3e_7    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.6.13               h756fd2b_1    conda-forge
gpgme                     1.16.0               he49afe7_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  2.50.0               h2b5222e_2    conda-forge
[... 335 lines cut ...]
unixodbc                  2.3.9                hd863c23_0    conda-forge
urdfdom                   2.3.5                h940c156_0    conda-forge
urdfdom_headers           1.0.5                h940c156_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
x264                      1!161.3030           h0d85af4_1    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.7.2                h0d85af4_0    conda-forge
xorg-libxau               1.0.9                h35c211d_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zlib                      1.2.11            h9173be1_1013    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

wolfv, 2022-02-28:
@MrChill looks like you are mixing `defaults` and `conda-forge`. (e.g. `tapi` comes from the defaults channel and not the conda-forge one). 

Maybe you can try to recreate the environment but withotu defaults channel (e.g.

`conda/mamba create -n ros ros-noetic-desktop -c robostack -c conda-forge --override-channels`

MrChill, 2022-03-01:
Do you have an idea, where the issue could come from?

Tobias-Fischer, 2022-03-01:
It seems like you didn't install the `compilers` package etc as described in the readme. Could you please install all packages as per readme, clean the workspace and try again?

MrChill, 2022-03-02:
It worked!

`conda create -n ros ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels`

to make sure that rviz is working.

`conda install compilers cmake pkg-config make ninja -c conda-forge --override-channels`

to remove the compiler issue.

Thank you so much for the support. Robostack is amazing!

wolfv, 2022-03-02:
Awesome! Happy to hear that

closed 2022-03-02 by wolfv