RoboStack/ros-noetic #236: ros_control python 3.9

sonelu, 2022-03-14:
The version of Python suggested in the install guide is 3.9.
But some ros packages are not available for 3.9, for instance `ros_control`:
```
# Name                       Version           Build  Channel             
ros-noetic-ros-control          0.19.4  py38h143cb38_3  robostack           
ros-noetic-ros-control          0.19.4  py38h7e910e5_5  robostack           
ros-noetic-ros-control          0.19.5 py38hb43b470_13  robostack 
```

sonelu, 2022-03-14:
The same also for `ros_controllers`:
```
# Name                       Version           Build  Channel             
ros-noetic-ros-controllers          0.18.1  py38h143cb38_3  robostack           
ros-noetic-ros-controllers          0.18.1  py38h7e910e5_5  robostack           
ros-noetic-ros-controllers          0.19.0 py38hb43b470_13  robostack  
```

Tobias-Fischer, 2022-03-14:
Are you okay adding the packages you need to https://github.com/RoboStack/ros-noetic/blob/master/vinca_linux_64.yaml and opening a pull request?

Tobias-Fischer, 2022-03-14:
Added in latest build.

closed 2022-03-14 by Tobias-Fischer

sonelu, 2022-03-15:
Sorry, forgot to mention I'm on MacOS and I think the build skipped them for this OS. Do they need to be added in https://github.com/RoboStack/ros-noetic/blob/master/vinca_osx.yaml too?

Tobias-Fischer, 2022-03-15:
I just triggered the build in https://github.com/RoboStack/ros-noetic/commit/4cb26789759a47fbdb0670c1f6160f07c026eeb9

sonelu, 2022-03-15:
I see `ros-control`
```
# Name                       Version           Build  Channel             
ros-noetic-ros-control          0.19.4  py38h143cb38_3  robostack           
ros-noetic-ros-control          0.19.4  py38h7e910e5_5  robostack           
ros-noetic-ros-control          0.19.5 py38hb43b470_13  robostack           
ros-noetic-ros-control          0.19.5 py39heb88ffd_16  robostack  
```

but not `ros-controllers`
```
# Name                       Version           Build  Channel             
ros-noetic-ros-controllers          0.18.1  py38h143cb38_3  robostack           
ros-noetic-ros-controllers          0.18.1  py38h7e910e5_5  robostack           
ros-noetic-ros-controllers          0.19.0 py38hb43b470_13  robostack 
```

wolfv, 2022-03-15:
@sonelu if you make a PR I'll merge it!

sonelu, 2022-03-15:
@wolfv Tobias already added `ros-controllers` in the vinca file  https://github.com/RoboStack/ros-noetic/blob/06b5343786a8eefe7e0adef8cf9b44e1c7e880b1/vinca_osx.yaml#L31

wolfv, 2022-03-15:
I triggered another build -- we might have missed a certain velodyne package the first time around.

wolfv, 2022-03-15:
@sonelu it should work now, would be nice if you can confirm :)