RoboStack/ros-noetic #238: rosdep get unsupported os robostack

Jack-XHP, 2022-03-16:
Hi I am trying to compile moveit from source with ros noetic and ubuntu 20. I install ros-noetic-desktop-full following the readme guide. Yet when I use  
```
rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
```
I got the following warning 
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_planning: Unsupported OS [robostack]
rviz_visual_tools: Unsupported OS [robostack]
moveit_core: Unsupported OS [robostack]
pilz_industrial_motion_planner: Unsupported OS [robostack]
geometric_shapes: Unsupported OS [robostack]
moveit_ros_manipulation: Unsupported OS [robostack]
moveit_servo: Unsupported OS [robostack]
moveit_ros_visualization: Unsupported OS [robostack]
moveit_ros_planning_interface: Unsupported OS [robostack]
moveit_setup_assistant: Unsupported OS [robostack]
moveit_ros_perception: Unsupported OS [robostack]
srdfdom: Unsupported OS [robostack]
moveit_ros_occupancy_map_monitor: Unsupported OS [robostack]
moveit_kinematics: Unsupported OS [robostack]
moveit_tutorials: Unsupported OS [robostack]

```

Jack-XHP, 2022-03-16:
Here is my conda list 

```
# packages in environment at /home/jx/miniconda3/envs/ros:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h5bd9786_13    conda-forge
adwaita-icon-theme        41.0                 ha770c72_1    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.0.1                hedfc422_7    conda-forge
at-spi2-atk               2.38.0               h0630a04_3    conda-forge
at-spi2-core              2.40.3               h0630a04_0    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
bcrypt                    3.2.0            py38h497a2fe_2    conda-forge
binutils                  2.36.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_2    conda-forge
binutils_linux-64         2.36                 hf3e587d_7    conda-forge
boost                     1.74.0           py38h2b96118_5    conda-forge
boost-cpp                 1.74.0               h312852a_4    conda-forge
brotli                    1.0.9                h7f98852_6    conda-forge
brotli-bin                1.0.9                h7f98852_6    conda-forge
bullet-cpp                3.09                 h1abd341_2    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.4.1                h166bdaf_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cairo                     1.16.0            h6cf1ce9_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
catkin_tools              0.8.4              pyhd8ed1ab_0    conda-forge
certifi                   2021.10.8        py38h578d9bd_1    conda-forge
cffi                      1.15.0           py38h3931269_0    conda-forge
cfitsio                   3.470                hb418390_7    conda-forge
clang-format              13.0.1          default_hc23dcda_0    conda-forge
clang-format-13           13.0.1          default_hc23dcda_0    conda-forge
cmake                     3.22.3               h5432695_0    conda-forge
compilers                 1.4.1                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_0    conda-forge
coreutils                 9.0                  h7f98852_0    conda-forge
coverage                  6.3.2            py38h0a891b7_1    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
cryptography              36.0.1           py38h3e25421_0    conda-forge
curl                      7.82.0               h7bff187_0    conda-forge
cxx-compiler              1.4.1                h924138e_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.10.1               h53b96f2_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py38h578d9bd_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.9                h7f98852_0    conda-forge
expat                     2.4.7                h27087fc_0    conda-forge
fcl                       0.6.1                h2cbc392_3    conda-forge
ffmpeg                    4.3.2                h37c90e5_3    conda-forge
flann                     1.9.1             h2e58136_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              h8e229c2_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.30.0           py38h0a891b7_0    conda-forge
fortran-compiler          1.4.1                h2a4ca65_0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0               h88c329d_7    conda-forge
[... 513 lines cut ...]
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxfixes            5.0.3             h516909a_1004    conda-forge
xorg-libxi                1.7.10               h516909a_0    conda-forge
xorg-libxinerama          1.1.4             h9c3ff4c_1001    conda-forge
xorg-libxmu               1.1.3                h516909a_0    conda-forge
xorg-libxpm               3.5.13               h516909a_0    conda-forge
xorg-libxrandr            1.5.2                h516909a_1    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
[... cut ...]
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.6.3                he1b5a44_4    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Tobias-Fischer, 2022-03-16:
Can you try
`export ROS_OS_OVERRIDE="conda:linux"`
and then run rosdep again?

Tobias-Fischer, 2022-03-16:
And can you please also add the `--verbose` flag to rosdep?

Jack-XHP, 2022-03-16:
Setting conda as os resolve the problem for unsupported os. Now the other problem I get is some conda packages are not picked up by rosdep. I double-check with conda and `pkg-config`, `opengl`, `eigen`, `yaml-cpp` and `tinyxml2` are already installed.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_core: No definition of [liburdfdom-headers-dev] for OS [conda]
rviz_visual_tools: No definition of [libqt5x11extras5-dev] for OS [conda]
geometric_shapes: No definition of [pkg-config] for OS [conda]
moveit_ros_visualization: No definition of [qtbase5-dev] for OS [conda]
moveit_ros_perception: No definition of [opengl] for OS [conda]
moveit_ros_occupancy_map_monitor: No definition of [eigen] for OS [conda]
moveit_ros_planning: No definition of [eigen] for OS [conda]
moveit_ros_manipulation: No definition of [eigen] for OS [conda]
moveit_tutorials: No definition of [eigen] for OS [conda]
moveit_setup_assistant: No definition of [yaml-cpp] for OS [conda]
srdfdom: No definition of [tinyxml2] for OS [conda]
moveit_kinematics: No definition of [liborocos-kdl-dev] for OS [conda]
moveit_ros_planning_interface: No definition of [eigen] for OS [conda]
```

Tobias-Fischer, 2022-03-16:
Thanks - I hadn't thought about that. I pushed a fix which should resolve this. For you to get this fix, please first run
`rosdep init`. It will tell you that it's already initialized and to remove a file if you want to re-initialize (in my case, it was `/Users/fischert/mambaforge/envs/robostackenv/etc/ros/rosdep/sources.list.d/20-default.list`). So remove this file.

Then run `rosdep init` again and subsequently `rosdep update`. This will pull in the updated file (conda.yaml instead of robostack.yaml).

I also created a new ros-noetic-ros-workspace package that sets the ROS_OS_OVERRIDE to `conda:platform` instead of `robostack:platform`, so if you update this package using `mamba update --all` you won't need the manual export anymore.

/cc @wolfv - FYI

Tobias-Fischer, 2022-03-16:
Feel free to reopen if it still doesn't work

closed 2022-03-16 by Tobias-Fischer

JKBehrens, 2022-09-21:
Hi @Tobias-Fischer,
I was just running into this issue on a fresh install of noetic. For now, I hacked installers.py of rosdep (see below) and it worked.

```    
def get_os_name_and_version(self):
        """
        Get the OS name and version key to use for resolution and
        installation.  This will be the detected OS name/version
        unless :meth:`InstallerContext.set_os_override()` has been
        called.

        :returns: (os_name, os_version), ``(str, str)``
        """
        if self.os_override:
            if self.os_override[0].startswith("robostack"):
                return ('conda', self.os_override[-1])
            return self.os_override
        else:
            os_name = self.os_detect.get_name()
            if os_name.startswith("robostack"):
                os_name = 'conda'
            os_key = self.get_os_version_type(os_name)
            os_version = os_key(self.os_detect)
            return os_name, os_version
```
[bug_env.yaml.txt](https://github.com/RoboStack/ros-noetic/files/9620005/bug_env.yaml.txt)

JKBehrens, 2022-09-23:
Thanks for the answer. I migrated everything to python 3.9. However, I have weird issues with mamba not seeing obviously installed packages.

```
mamba install ros-noetic-boost-sml

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.25.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-noetic-boost-sml']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
pkgs/main/linux-64                                            No change
pkgs/r/noarch                                                 No change
[+] 0.1s
pkgs/main/noarch                                              No change
pkgs/r/linux-64                                               No change
robostack/noarch                                              No change
robostack-experimental/linux-64                               No change
robostack/linux-64                                            No change
robostack-experimental/noarch                                 No change

Pinned packages:
  - python 3.9.*

Encountered problems while solving:
  - package ros-noetic-boost-sml-0.1.2-py38h266acf0_9 requires python, but none of the providers can be installed
```
 But python is installed:
```
conda list |grep python
colcon-python-setup-py    0.2.7                      py_0    conda-forge
python                    3.9.13          h9a8a25e_0_cpython    conda-forge
python-dateutil           2.8.2              pyhd8ed1ab_0    conda-forge
python-gnupg              0.5.0              pyhd8ed1ab_0    conda-forge
python-orocos-kdl         1.5.1            py39he80948d_2    conda-forge
python_abi                3.9                      2_cp39    conda-forge
ros-noetic-python-qt-binding 0.4.4           py39h27aa46a_13    robostack
ros-noetic-rviz-python-tutorial 0.11.0          py39h6fdeb60_13    robostack
```

Many other packages had the same problem with other packages, but I could always install them one by one. Only packages dependent on `python` always fail. 

This is my current environment.
[bug_env.yaml.txt](https://github.com/RoboStack/ros-noetic/files/9634647/bug_env.yaml.txt)

Should I open a new issue? Or we solve it here?
Thank you!
Jan

Tobias-Fischer, 2022-09-23:
Hi Jan, these packages have not yet been built for python 3.9. Please open a pull request adding the packages you need to the vinca_*.yaml. If you look through recent pull requests you will find several examples. We can then merge the pull request which will build the packages on the GitHub servers.