RoboStack/ros-noetic #243: error with bondcpp

tohsin, 2022-04-01:
I am trying to run catkin_make and i keep getting this error i have xcode cli tools installed btw any fix?

`CMake Error at /Users/hey/opt/miniconda3/envs/robostackenv/share/bondcpp/cmake/bondcppConfig.cmake:113 (message):
  Project 'bondcpp' specifies
  '/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include' as
  an include dir, which is not found.  It does neither exist as an absolute
  directory nor in
  '${{prefix}}//Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include'.`

traversaro, 2022-04-01:
Thanks for the report! I hope that https://github.com/RoboStack/ros-noetic/pull/244 will fix the issue.

tohsin, 2022-04-01:
Thanks, is there a temporary fix though?

traversaro, 2022-04-01:
I think it should be enough to remove any occurence of /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include from /Users/hey/opt/miniconda3/envs/robostackenv/share/bondcpp/cmake/bondcppConfig.cmake .

Tobias-Fischer, 2022-04-02:
Should be fixed by #244 - feel free to reopen if it is still an issue (please wait an hour or so for the packages to be built).

closed 2022-04-02 by Tobias-Fischer

tohsin, 2022-04-04:
Thanks ,[traversaro](https://github.com/traversaro) solution pretty much fixed it