RoboStack/ros-noetic #245: mamba install catkin_tools failed

LathDevers, 2022-04-06:
System information:

- MacBook Pro (14-inch, 2021)
- Apple M1 Pro
- MacOS Monterey 12.3.1

```
Looking for: ['catkin_tools']

pkgs/main/osx-arm64      [====================] (00m:00s) No change
pkgs/r/noarch            [====================] (00m:00s) No change
pkgs/main/noarch         [====================] (00m:00s) No change
pkgs/r/osx-arm64         [====================] (00m:00s) 404 Failed

Pinned packages:
  - python 3.9.*

Encountered problems while solving:
  - nothing provides requested catkin_tools
 ```

wolfv, 2022-04-06:
You are not using the `conda-forge` channel. I would recommend to add `channels: [conda-forge, robostack]` to your .condarc file, or `-c conda-forge` on the command line.

LathDevers, 2022-04-06:
> You are not using the `conda-forge` channel. I would recommend to add `channels: [conda-forge, robostack]` to your .condarc file, or `-c conda-forge` on the command line.

Thanks

closed 2022-04-06 by LathDevers