RoboStack/ros-noetic #246: kdl-parser-py for Python 3.9

sonelu, 2022-04-12:
Gets, can you please add a build for Python 3.9 for `kdl-parser-py`? Now it's missing:
```
# Name                       Version           Build  Channel             
ros-foxy-kdl-parser            2.4.1  py38h46e7d52_0  robostack           
ros-melodic-kdl-parser          1.13.1  py36h0130604_1  robostack           
ros-melodic-kdl-parser-py          1.13.1  py36h0130604_1  robostack           
ros-noetic-kdl-parser          1.14.1  py38h143cb38_3  robostack           
ros-noetic-kdl-parser          1.14.1 py38h606ea6d_13  robostack           
ros-noetic-kdl-parser          1.14.1  py38h7e910e5_5  robostack           
ros-noetic-kdl-parser          1.14.1 py38hb43b470_10  robostack           
ros-noetic-kdl-parser          1.14.1 py39hf052c5f_14  robostack           
ros-noetic-kdl-parser          1.14.1 py39hf052c5f_15  robostack           
ros-noetic-kdl-parser-py          1.14.1 py38hb43b470_10  robostack           
ros-noetic-kdl-parser-py          1.14.1 py38hb43b470_13  robostack           
ros-noetic-kdl-parser-py          1.14.1  py38hb43b470_7  robostack
```

I'm sorry I this is probably discussed somewhere else, but if this is a simple thing to do with a PR let me know and I will do it.

sonelu, 2022-04-12:
Actually do I just need to add the package in the [vinca_osx.yaml](https://github.com/RoboStack/ros-noetic/blob/e5a20fca7f3882c9713460ab593d8ba826abc6ee/vinca_osx.yaml#L29) file?

Tobias-Fischer, 2022-04-12:
Yes - if you could open a PR that would be great. Please add it in all the vinca files for the various platforms.

sonelu, 2022-04-13:
@Tobias-Fischer thanks for the rebuild.

There is though another problem: it looks like the repo used in the build is not updated from the latest version (from 2020!). This PR: ros/kdl_parser#45 doesn't seem to be package served by RoboStack.

wolfv, 2022-04-13:
Did they properly release it into noetic?

sonelu, 2022-04-13:
They released 1.4.2 and noetic distro update is merged.
Can you try to trigger a new build?

wolfv, 2022-04-13:
You could make a new pr :)

wolfv, 2022-04-14:
I just re-ran the job. Crossing fingers :)

sonelu, 2022-04-14:
I see the jobs have run and the new version is picked up in the [log](https://github.com/RoboStack/ros-noetic/runs/6020251755?check_suite_focus=true#step:16:593) but the package is not showing in `mamba search`:
```
# Name                       Version           Build  Channel             
ros-melodic-kdl-parser-py          1.13.1  py36h0130604_1  robostack           
ros-noetic-kdl-parser-py          1.14.1 py38hb43b470_10  robostack           
ros-noetic-kdl-parser-py          1.14.1 py38hb43b470_13  robostack           
ros-noetic-kdl-parser-py          1.14.1  py38hb43b470_7  robostack           
ros-noetic-kdl-parser-py          1.14.1 py39heb88ffd_16  robostack
```
I haven't seen any errors in the log.

wolfv, 2022-04-14:
The 1.14.2 version appeared 2 minutes ago: https://anaconda.org/RoboStack/ros-noetic-kdl-parser

PR #247 (2022-04-13): Added kdl-parser-py for Python 3.9 in vinca files
Should fix #246