RoboStack/ros-noetic #250: ROS_OS_OVERRIDE points to robostack

mgrrx, 2022-04-15:
rosdep fails resolving dependencies for me. I investigated a bit and for me it looks as if there was an attempt to replace the ROS OS Name "robstack" by "conda" (as we can see in e.g. https://raw.githubusercontent.com/RoboStack/ros-noetic/master/conda.yaml). However `etc/conda/activate.d/ros-noetic-catkin_activate.sh` still points to robostack for me (after a fresh install of MacOS). Replacing robostack with conda allows me to run rosdep as usual.

Tobias-Fischer, 2022-04-16:
What's the output of conda list?

mgrrx, 2022-04-16:
```
# packages in environment at /usr/local/Caskroom/mambaforge/base/envs/ros_env:
#
# Name                    Version                   Build  Channel
anyio                     3.5.0            py38h50d1736_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
asgiref                   3.5.0              pyhd8ed1ab_0    conda-forge
assimp                    5.0.1                h1224e73_7    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.0            py38hed1de0f_3    conda-forge
black                     22.3.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py38hb0f0857_5    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
brotlipy                  0.7.0           py38hed1de0f_1004    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.4.1                h774085c_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
catkin_tools              0.8.5              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             ha54d4a6_10    conda-forge
cctools_osx-64            973.0.1             h609b070_10    conda-forge
certifi                   2021.10.8        py38h50d1736_2    conda-forge
cffi                      1.15.0           py38h1a44b6c_0    conda-forge
charset-normalizer        2.0.12             pyhd8ed1ab_0    conda-forge
clang                     12.0.1               h694c41f_4    conda-forge
clang-12                  12.0.1          default_he082bbe_4    conda-forge
clang_osx-64              12.0.1               h633439f_8    conda-forge
clangxx                   12.0.1          default_he082bbe_4    conda-forge
clangxx_osx-64            12.0.1               hdb584c0_8    conda-forge
click                     8.1.2            py38h50d1736_0    conda-forge
cmake                     3.23.1               h35a7dd9_0    conda-forge
compiler-rt               12.0.1               he01351e_0    conda-forge
compiler-rt_osx-64        12.0.1               hd3f61c9_0    conda-forge
compilers                 1.4.1                h694c41f_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
coreutils                 9.0                  h0d85af4_0    conda-forge
coverage                  6.3.2            py38hed1de0f_2    conda-forge
cryptography              36.0.2           py38h826b3c8_1    conda-forge
cxx-compiler              1.4.1                hbb4e6a2_0    conda-forge
dataclasses               0.8                pyhc8e2a94_3    conda-forge
dbus                      1.13.6               h811a1a6_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py38h50d1736_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.4.1                h46a1860_0    conda-forge
freeimage                 3.18.0               haf7e230_8    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.8               hb161b9c_0    conda-forge
gettext                   0.19.8.1          hd1a6beb_1008    conda-forge
gfortran                  9.3.0               h768ea0c_15    conda-forge
gfortran_impl_osx-64      9.3.0               h9cc0e5e_23    conda-forge
gfortran_osx-64           9.3.0               h18f7dce_15    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
gmock                     1.11.0               hbb4e6a2_0    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gpgme                     1.15.1               he49afe7_0    conda-forge
[... 237 lines cut ...]
urllib3                   1.26.9             pyhd8ed1ab_0    conda-forge
uvicorn                   0.17.6           py38h50d1736_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
[... cut ...]
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zlib                      1.2.11            h6c3fc93_1014    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

I started noticing this when writing github actions for one project: rosdep was succesful once I added this line here https://github.com/mgrrx/aioros_bridge/actions/runs/2174446232/workflow#L38 . I have the same behavior on my new MacOS installation.

Tobias-Fischer, 2022-04-16:
You are using the old python 3.8 builds - unfortunately there the manual setting of the environment variable (the way you do) is necessary. If you switch over to the python 3.9 builds, this problem will go away.

closed 2022-04-16 by Tobias-Fischer

mgrrx, 2022-04-16:
Oh OK I did not know this. Now I ask myelf why "new" builds target Python 3.9? REP 3 states that noetic targets 3.8 and that's why I would actually prefer using Python 3.8 similar to noetic builds on Ubuntu or Debian (https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025).
What is the reason for the switch to 3.9?

Tobias-Fischer, 2022-04-17:
See https://github.com/RoboStack/ros-noetic/issues/47 .. unfortunately we don't have the resources to build against multiple versions