RoboStack/ros-noetic #251: `rqt_controller_manager` not available despite `ros_control` being installed

sonelu, 2022-04-18:
It looks like `ros-control` build for 3.9 is missing `rqt-controller-manager` that is still left on 3.8, therefore `mamba install ros-noetic-ros-control` will install all the packages _except_ `rqt-controller-manager`:
```
# Name                       Version           Build  Channel             
ros-noetic-rqt-controller-manager          0.19.5 py38hb43b470_11  robostack           
ros-noetic-rqt-controller-manager          0.19.5 py38hb43b470_13  robostack 
```

Should we try a rebuild of `ros-control` or just of `rqt-controller-manager`?

Also, is this a bug in vinca who is skipping `rqt_controller_manager` (because is not listed in the `package.xml`)?

wolfv, 2022-04-18:
I think a rebuild of just that particular package could be enough?

Should it be listed in the direct dependencies? Or is there some other package that should pull it in?

sonelu, 2022-04-18:
I'll put a PR with that package only.

But here is an interesting question: if I clone a metapackage and run a catkin build it will build all the included packages, no matter what dependencies are defined. In our build, when processing a metapackage it will only pull and build the packages that are explicitly listed in the metapackage package.xml. If it is not listed there (as is the case with `rqt_controller_manager`) it will be not built.

Maybe we need to look at how vinca determines the packages so that it's more in line with catkin?

wolfv, 2022-04-18:
If we enable the right options in vinca it should use & rebuild the dependencies listed in the package.xml — that capability is definitely there. I'd need to investigate what and why this failed here.

sonelu, 2022-04-18:
In this case the package `rqt_controller_manager` is not listed in the meta-package's `package.xml`, although it is part of that meta-package. I think it is by design so that rqt doesn't become a dependency for `ros_control`, so that you can install on a physical robot - where you most likely don't need the rqt plugins. It's debatable if this is the best approach or if vinca indeed did something wrong here, probably not. I think we can leave things as they are now.

closed 2022-04-18 by Tobias-Fischer

Tobias-Fischer, 2022-04-18:
Hi @sonelu - many thanks for your continued contributions! Greatly appreciated.

Re the dependencies: I don't think this is a bug in vinca - if the dependency is not declared in the package.xml we have no way to detect it. This wouldn't be any different on a standard Ubuntu installation I don't think. Even a `catkin build ros_control` wouldn't build it, would it?

wolfv, 2022-04-19:
Maybe it works with `catkin` because `rqt_controller_manager` is part of teh [`ros_control`](https://github.com/ros-controls/ros_control) repository and when you clone that repo, part of your workspace? And then `catkin build` will just build all available packages, including `rqt_controller_manager`?

Tobias-Fischer, 2022-04-19:
Yes - that's exactly what happens @wolfv - sorry, should have explicitly written that