RoboStack/ros-noetic #255: Noetic packages downloaded from robostack-old/linux-64 are failing checksum validation

adammacudzinski, 2022-04-27:
Hello,
my team's environment build script just started failing at the conda install step because the packages downloaded from the robostack-old repo are failing with the following error:

```[2022-04-26T19:29:26.188Z] warning  libmamba '/home/luminar/mambaforge/pkgs/ros-noetic-rosgraph-msgs-1.11.3-py36h02824d8_3.tar.bz2' validation failed

[2022-04-26T19:29:26.188Z] ros-noetic-rosgraph-msgs-1.11.3-py36h02824d8_3.tar.bz2 tarball has incorrect size

[2022-04-26T19:29:26.188Z] ros-noetic-resource-retriever-1.12.6-py36h429d66c_3.tar.bz2 tarball has incorrect size

[2022-04-26T19:29:26.188Z] error    libmamba File not valid: file size doesn't match expectation "/home/luminar/mambaforge/pkgs/ros-noetic-resource-retriever-1.12.6-py36h429d66c_3.tar.bz2"```

Any idea what is going on? Or how to fix?

wolfv, 2022-04-27:
I think there was an anaconda.org outage for a couple hours where packages would return `404`: https://anaconda.statuspage.io/

Tobias-Fischer, 2022-05-08:
Closing here - feel free to reopen if it's still an issue.

closed 2022-05-08 by Tobias-Fischer