RoboStack/ros-noetic #258: Missing ros-noetic-perception-pcl

vanem, 2022-05-08:
These packages are not present after installing robostack:
ros-noetic-perception-pcl:
    ros-noetic-pcl-conversions
    ros-noetic-pcl-msgs
    ros-noetic-pcl-ros

Although they appear in the standard distro, and they should be present in robostack according to this list: https://robostack.github.io/noetic.html

Tobias-Fischer, 2022-05-08:
What happens if you try to `mamba install ros-noetic-perception-pcl`?

vanem, 2022-05-08:
I forgot to mention I just installed robostack on Windows. Running:
`mamba install ros-noetic-perception-pcl`
returns:
```
Looking for: ['ros-noetic-perception-pcl']

pkgs/main/win-64                                              No change
anaconda/win-64                                               No change
pkgs/main/noarch                                              No change
anaconda/noarch                                               No change
pkgs/r/noarch                                                 No change
pkgs/r/win-64                                                 No change
pkgs/msys2/noarch                                             No change
pkgs/msys2/win-64                                             No change
conda-forge/noarch                                   8.1MB @   4.6MB/s  2.3s
conda-forge/win-64                                  14.7MB @   4.3MB/s  4.6s

Pinned packages:
  - python 3.9.*

Encountered problems while solving:
  - nothing provides requested ros-noetic-perception-pcl
```
 
I should also mention I tried:
`rosdep install --from-paths src --ignore-src -r -y --rosdistro noetic`
 on the workspace that requires this dependency, but I run into another problem:
`Bad installer [conda]: None of micromamba.exe, mamba.exe, conda.exe was found.`

Tobias-Fischer, 2022-05-08:
I added the perception_pcl packages for build - hopefully they will be up in half an hour or so.

Regarding the rosdep error, please open a separate issue.

closed 2022-05-08 by Tobias-Fischer

vanem, 2022-05-10:
Unfortunately I'm still unable to install perception_pcl, same error. Anything I could try?

wolfv, 2022-05-10:
@vanem hmm, the win-64 package was uploaded here: https://anaconda.org/robostack/ros-noetic-perception-pcl

wolfv, 2022-05-10:
And it resolves for me. Can you post exactly what you tried?

You might have forgotten `-c robostack`? Also I heavily recommend to remove the `defaults` channel.

vanem, 2022-05-10:
You are right. I was not aware that the packages are published on robostack channel. I just followed the example at the end of the FAQ section of the readme, and was not aware you need to specify the channel. 

>  Also I heavily recommend to remove the defaults channel.

I did this by editing "%HOMEPATH%\.condarc", I assume this is what you meant.
After this I was able to dl the package, thanks!

wolfv, 2022-05-10:
great!