RoboStack/ros-noetic #260: rosdep can't find mamba

vanem, 2022-05-09:
On a fresh robostack Windows install, I am able to run mamba directly from the environment, but if I run rosdep on a workspace:
`rosdep install --from-paths src --ignore-src -r -y --rosdistro noetic`
I get:
`Bad installer [conda]: None of micromamba.exe, mamba.exe, conda.exe was found.`

traversaro, 2022-05-13:
Hi @vanem, what do you mean by "if I run rosdep on a workspace"? Even if you are using a catkin workspace, you should always activate the corresponding conda/mamba environment.

vanem, 2022-05-13:
Hi @traversaro. The robostack environment is indeed activated (otherwise the rosdep command would not be found at all). I'm also able to run catkin_make for example and build C++ packages. Here is the complete result of running rosdep:

```
(robostackenv) C:\_work\prj\ros1\ws>rosdep install --from-paths src --ignore-src -r -y

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Bad installer [conda]: None of micromamba.exe, mamba.exe, conda.exe was found.
```

traversaro, 2022-05-13:
Can you list the content of your environment with `mamba list` ? Thanks!

vanem, 2022-05-13:
# packages in environment at c:\_work\app\miniforge3\envs\robostackenv:
#
# Name                    Version                   Build  Channel
apr                       1.7.0                he38c35c_5    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
bcrypt                    3.2.2            py39hb82d6ee_0    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.4.26            haa95532_0    anaconda
cairo                     1.16.0            h15b3021_1010    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
certifi                   2021.10.8        py39hcbf5309_2    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.1               h39d44d4_0    conda-forge
colorama                  0.4.4              pyhd3eb1b0_0    anaconda
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cryptography              36.0.2           py39h7bc7c5c_1    conda-forge
curl                      7.83.0               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39hcbf5309_1    conda-forge
double-conversion         3.2.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
flann                     1.9.1             h075982a_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.33.3           py39hb82d6ee_0    conda-forge
fortran-compiler          1.4.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               h6676e37_8    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gl2ps                     1.4.2                hce2df77_1    anaconda
glew                      2.1.0                hd77b12b_3    anaconda
gmock                     1.11.0               h2d74725_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  2.50.0               had6c3a3_2    conda-forge
gtest                     1.11.0               h2d74725_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
harfbuzz                  4.2.0                hc1763ed_0    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
imath                     3.1.5                h12d4b20_0    conda-forge
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
jasper                    2.0.33               h77af90b_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9e                   h8ffe710_1    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
[... 309 lines cut ...]
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vs2017_win-64             19.16.27032.1        hb4161e2_3    anaconda
vs2019_win-64             19.27.29111          ha89c4d3_2    anaconda
vswhere                   2.8.4                haa95532_0    anaconda
vtk                       9.1.0           qt_py39h1ab545e_207    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge

vanem, 2022-05-13:
I now realize, are the packages suppose to be a mix of robostack and conda-forge channels like this?

traversaro, 2022-05-13:
> I now realize, are the packages suppose to be a mix of robostack and conda-forge channels like this?

Yes, that is totally ok. `conda-forge` is the basis on which `robostack` is built. What is more dangerous is mixing `anaconda` and `conda-forge`, as it seems that it is happening in your environment.

Which commands did you used to build your environment? Ideally it should contain only `conda-forge` and `robostack` packages, not `anaconda` one.

traversaro, 2022-05-13:
I am definitely not an expert, but I guess the problem may be the `.condarc` in user home?

vanem, 2022-05-13:
@traversaro thanks! I cleaned .condarc from user homedir and reinstalled, and that solved the problem

closed 2022-05-13 by vanem

kneave, 2024-04-25:
To add another cause of this for those that find themselves here, it's worth checking if `/etc/ros/rosdep/sources.list.d/20-default.list` exists. I must have run `sudo rosdep init` at some point which was the cause of the `Bad installer` error in my case.