RoboStack/ros-noetic #263: Roscore doesn't work on M1 Apple installation

cmastalli, 2022-05-18:
First of all, I would like to thank you for such kind of project. We need more tools like this in robotics!!!

I am still learning about robot stack, conda and the M1-related stuffs. I installed ros-noetic for osx-arm64 following the instructions here: https://robostack.github.io/GettingStarted.html. The installation seems to be done alright and below you can see the list of installed packages. However, when I run roscore, then I get the following error:
```bash
dyld[38032]: missing symbol called
[rosout-1] process has died [pid 38032, exit code -6, cmd /Users/cm2099/miniconda3/envs/ros/lib/rosout/rosout __name:=rosout __log:=/Users/cm2099/.ros/log/795465a6-d693-11ec-aa28-3e002bd2e4ed/rosout-1.log].
log file: /Users/cm2099/.ros/log/795465a6-d693-11ec-aa28-3e002bd2e4ed/rosout-1*.log
```

Finally, if I enable rosetta, then the error goes away. So, do I have installed osx-arm64 packages? I am a bit confused, any guideline is appreciated.

```bash
Package                                           Version  Build                 Channel                                Size
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

  + apr                                               1.7.0  h9b22ae9_5            conda-forge/osx-arm64                Cached
  + assimp                                            5.2.3  hd77be17_1            conda-forge/osx-arm64                Cached
  + atk-1.0                                          2.36.0  heb41896_4            conda-forge/osx-arm64                Cached
  + bcrypt                                            3.2.2  py39h9eb174b_0        conda-forge/osx-arm64                  41kB
  + boost                                            1.74.0  py39h776e71f_5        conda-forge/osx-arm64                Cached
  + boost-cpp                                        1.74.0  h3afdda6_7            conda-forge/osx-arm64                Cached
  + brotli                                            1.0.9  h1c322ee_7            conda-forge/osx-arm64                Cached
  + brotli-bin                                        1.0.9  h1c322ee_7            conda-forge/osx-arm64                Cached
  + c-ares                                           1.18.1  h3422bc3_0            conda-forge/osx-arm64                Cached
  + cairo                                            1.16.0  hb99c34e_1009         conda-forge/osx-arm64                Cached
  + catkin_pkg                                        0.5.0  pyhd8ed1ab_0          conda-forge/noarch                     51kB
  + certifi                                       2021.10.8  py39h2804cbe_2        conda-forge/osx-arm64                Cached
  + cffi                                             1.15.0  py39h52b1de0_0        conda-forge/osx-arm64                Cached
  + cmake                                            3.23.1  he7c8c24_0            conda-forge/osx-arm64                Cached
  + console_bridge                                    1.0.2  h3e96240_1            conda-forge/osx-arm64                Cached
  + cryptography                                     36.0.2  py39hbe5e4b8_1        conda-forge/osx-arm64                Cached
  + cycler                                           0.11.0  pyhd8ed1ab_0          conda-forge/noarch                   Cached
  + defusedxml                                        0.7.1  pyhd8ed1ab_0          conda-forge/noarch                   Cached
  + distro                                            1.6.0  pyhd8ed1ab_0          conda-forge/noarch                   Cached
  + docutils                                         0.18.1  py39h2804cbe_1        conda-forge/osx-arm64                Cached
  + eigen                                             3.4.0  hc021e02_0            conda-forge/osx-arm64                Cached
  + empy                                              3.3.4  pyh9f0ad1d_1          conda-forge/noarch                   Cached
  + expat                                             2.4.8  h6b3803e_0            conda-forge/osx-arm64                Cached
  + ffmpeg                                            4.3.2  h38cfed3_3            conda-forge/osx-arm64                Cached
  + font-ttf-dejavu-sans-mono                          2.37  hab24e00_0            conda-forge/noarch                   Cached
  + font-ttf-inconsolata                              3.000  h77eed37_0            conda-forge/noarch                   Cached
  + font-ttf-source-code-pro                          2.038  h77eed37_0            conda-forge/noarch                   Cached
  + font-ttf-ubuntu                                    0.83  hab24e00_0            conda-forge/noarch                   Cached
  + fontconfig                                      2.13.94  heb65262_0            conda-forge/osx-arm64                Cached
  + fonts-conda-ecosystem                                 1  0                     conda-forge/noarch                   Cached
  + fonts-conda-forge                                     1  0                     conda-forge/noarch                   Cached
  + fonttools                                        4.33.3  py39h9eb174b_0        conda-forge/osx-arm64                Cached
  + freeimage                                        3.18.0  h6393344_8            conda-forge/osx-arm64                Cached
  + freetype                                         2.10.4  h17b34a0_1            conda-forge/osx-arm64                Cached
  + fribidi                                          1.0.10  h27ca646_0            conda-forge/osx-arm64                Cached
  + gdk-pixbuf                                       2.42.8  hdded42e_0            conda-forge/osx-arm64                Cached
  + gettext                                        0.19.8.1  h049c9fb_1008         conda-forge/osx-arm64                Cached
  + giflib                                            5.2.1  h27ca646_2            conda-forge/osx-arm64                Cached
  + gmock                                            1.11.0  h3e96240_0            conda-forge/osx-arm64                Cached
  + gmp                                               6.2.1  h9f76cd9_0            conda-forge/osx-arm64                Cached
  + gnutls                                           3.6.13  h706517b_1            conda-forge/osx-arm64                Cached
  + gpgme                                            1.16.0  hbdafb3b_0            conda-forge/osx-arm64                Cached
  + graphite2                                        1.3.13  h9f76cd9_1001         conda-forge/osx-arm64                Cached
  + graphviz                                         2.50.0  h53cdc11_2            conda-forge/osx-arm64                Cached
  + gst-plugins-base                                 1.18.5  h5bf8608_3            conda-forge/osx-arm64                Cached
  + gstreamer                                        1.18.5  ha076c35_3            conda-forge/osx-arm64                Cached
  + gtest                                            1.11.0  h3e96240_0            conda-forge/osx-arm64                Cached
  + gtk2                                            2.24.33  h57013de_2            conda-forge/osx-arm64                Cached
  + gts                                               0.7.6  h4b6d4d6_2            conda-forge/osx-arm64                Cached
  + harfbuzz                                          3.4.0  h6efa254_0            conda-forge/osx-arm64                Cached
  + hdf5                                             1.12.1  nompi_hf9525e8_104    conda-forge/osx-arm64                Cached
  + icu                                                69.1  hbdafb3b_0            conda-forge/osx-arm64                Cached
  + imath                                             3.1.5  h1280f1d_0            conda-forge/osx-arm64                Cached
[... 296 lines cut ...]
  + toml                                             0.10.2  pyhd8ed1ab_0          conda-forge/noarch                   Cached
  + unicodedata2                                     14.0.0  py39hb18efdd_1        conda-forge/osx-arm64                Cached
  + unixodbc                                         2.3.10  hb39ff51_0            conda-forge/osx-arm64                 269kB
[... cut ...]
  + yaml                                              0.2.5  h6ce8777_1            robostack-experimental/osx-arm64     Cached
  + yaml-cpp                                          0.6.3  hc88da5d_4            conda-forge/osx-arm64                Cached
  + zstd                                              1.5.2  hd705a24_1            conda-forge/osx-arm64                 383kB
  + zziplib                                         0.13.69  he1e0b03_1            conda-forge/osx-arm64                Cached

```

traversaro, 2022-05-18:
Hello @cmastalli ! From the package list it seems that you are using `osx-arm64` pacakges, i.e. native M1 packages (as opposed to `osx-64` that are macOS/Intel packages). Just as a reference, if you wanted to install packages via Rosetta you would need to follow the instructions in https://conda-forge.org/docs/user/tipsandtricks.html#installing-apple-intel-packages-on-apple-silicon .

For what regards the error, I am not sure, it may be that our packages depend on some system libraries that was installed/enabled by enabling Rosetta? Probably @Tobias-Fischer may know more.

cmastalli, 2022-05-18:
Thanks @traversaro for a prompt response. Just for clarity, I would like to install robotstack using the Apple native chip (i.e., osx-arm64) as I am keen to exploit its benefits.

wolfv, 2022-05-18:
@cmastalli so what you're saying is that – even with only native packages (osx-arm64) – you had to install / enable Rosetta for `roscore` to work?
That's unexpected!

cmastalli, 2022-05-18:
> @cmastalli so what you're saying is that – even with only native packages (osx-arm64) – you had to install / enable Rosetta for `roscore` to work?
> That's unexpected!

Yes, I think so. I had to type this in the terminal: 'arch -x86_64 zsh'.

cmastalli, 2022-05-26:
Is there any update regarding this topic?

Tobias-Fischer, 2022-05-26:
I am not experiencing this issue on my M1 so it's a bit hard to debug.

Could you please try to create a more minimal environment (i.e. `mamba create -n testenv ros-noetic-ros-core`) and see whether the problem occurs there, too (please post again `conda list`)?

Tobias-Fischer, 2022-05-26:
Also, could you please have a look at the ROS log files to see whether there is anything useful contained in there? (/Users/cm2099/.ros/log/)

Tobias-Fischer, 2022-09-17:
Closing due to inactivity

cmastalli, 2022-09-19:
> mamba create -n testenv ros-noetic-ros-core

I apologise if I didn't reply earlier. Please let me re-open this issue. This is the output of `conda list` in this minimal enviroment:
```bash
(testenv) cm2099@ build % conda list
# packages in environment at /Users/cm2099/mambaforge3/envs/testenv:
#
# Name                    Version                   Build  Channel
apr                       1.7.0                h9b22ae9_5    conda-forge
bcrypt                    3.2.2            py39h9eb174b_0    conda-forge
boost-cpp                 1.74.0               h1cb353e_8    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
ca-certificates           2022.9.14            h4653dfc_0    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h04d3946_0    conda-forge
cmake                     3.24.2               h967666c_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
cryptography              37.0.4           py39h4cb6239_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h2804cbe_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h6b3803e_0    conda-forge
gettext                   0.19.8.1          h049c9fb_1008    conda-forge
gmock                     1.11.0               h3e96240_0    conda-forge
gmp                       6.2.1                h9f76cd9_0    conda-forge
gpgme                     1.16.0               hbdafb3b_0    conda-forge
gtest                     1.11.0               h3e96240_0    conda-forge
icu                       70.1                 h6b3803e_0    conda-forge
krb5                      1.19.3               hf9b2bbe_0    conda-forge
libapr                    1.7.0                h3422bc3_5    conda-forge
libapriconv               1.2.2                h3422bc3_5    conda-forge
libaprutil                1.6.1                h3bcc323_5    conda-forge
libassuan                 2.5.5                hbdafb3b_0    conda-forge
libblas                   3.9.0           16_osxarm64_openblas    conda-forge
libcblas                  3.9.0           16_osxarm64_openblas    conda-forge
libcurl                   7.83.1               h2fcd78c_0    conda-forge
libcxx                    14.0.6               h2692d47_0    conda-forge
libedit                   3.1.20191231         hc8eb9b7_2    conda-forge
libev                     4.33                 h642e427_1    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libgfortran               5.0.0           11_3_0_hd922786_25    conda-forge
libgfortran5              11.3.0              hdaf2cc0_25    conda-forge
libglib                   2.72.1               ha1047ec_0    conda-forge
libgpg-error              1.45                 h59f49c1_0    conda-forge
libiconv                  1.16                 h642e427_0    conda-forge
liblapack                 3.9.0           16_osxarm64_openblas    conda-forge
libnghttp2                1.47.0               h232270b_1    conda-forge
libopenblas               0.3.21          openmp_hc731615_3    conda-forge
libsodium                 1.0.18               h27ca646_1    conda-forge
libsqlite                 3.39.3               h76d750c_0    conda-forge
libssh2                   1.10.0               hb80f160_3    conda-forge
libuv                     1.44.2               he4db4b2_0    conda-forge
libzlib                   1.2.12               h03a7124_3    conda-forge
llvm-openmp               14.0.4               hd125106_0    conda-forge
log4cxx                   0.11.0               h357cd66_3    conda-forge
lz4                       4.0.0            py39h049b86e_2    conda-forge
lz4-c                     1.9.3                hbdafb3b_1    conda-forge
ncurses                   6.3                  h07bb92c_1    conda-forge
netifaces                 0.11.0           py39h9eb174b_0    conda-forge
nose                      1.3.7                   py_1006    conda-forge
numpy                     1.23.3           py39hcb4b507_0    conda-forge
openssl                   1.1.1q               ha287fd2_0    conda-forge
paramiko                  2.11.0             pyhd8ed1ab_0    conda-forge
pcre                      8.45                 hbdafb3b_0    conda-forge
pip                       22.2.2             pyhd8ed1ab_0    conda-forge
pkg-config                0.29.2            hab62308_1008    conda-forge
poco                      1.11.1               h4ceb6a2_0    conda-forge
pycparser                 2.21               pyhd8ed1ab_0    conda-forge
[... 72 lines cut ...]
ros-noetic-std-msgs       0.5.13           py39hd384bd2_2    robostack
ros-noetic-std-srvs       1.11.3           py39hd384bd2_2    robostack
ros-noetic-stereo-msgs    1.13.1           py39hd384bd2_2    robostack
ros-noetic-topic-tools    1.15.14          py39hd384bd2_2    robostack
ros-noetic-trajectory-msgs 1.13.1           py39hd384bd2_2    robostack
ros-noetic-visualization-msgs 1.13.1           py39hd384bd2_2    robostack
ros-noetic-xmlrpcpp       1.15.14          py39h541b44f_2    robostack
[... cut ...]
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
zlib                      1.2.12               h03a7124_3    conda-forge
zstd                      1.5.2                h8128057_4    conda-forge
```

The issue is persistent in this minimal environment as well. Additionally, I am not able to see any issue in the ROS log files. I could share with you if this might help you.

cmastalli, 2022-09-20:
Thank you for opening the issue. Let me know if there is something else that I can try from my side.

traversaro, 2022-09-20:
@cmastalli do you have any suggestion on how to reproduce the error once Rosetta is already enabled? A quick google search suggest https://iboysoft.com/news/uninstall-rosetta-2.html as a procedure to re-disable rosetta, but you may have more insight.

jf---, 2023-01-03:
I can't reproduce: `roscore` runs like a kitten on my m1

Tobias-Fischer, 2023-01-24:
Feel free to reopen - can't reproduce either. Note there are new packages on the robostack-staging channel.

closed 2023-01-24 by Tobias-Fischer