RoboStack/ros-noetic #270: Have issues building other peoples projects

tohsin, 2022-06-12:
I am trying to run a simulator from 
https://github.com/yukkysaito/vehicle_sim.git
I admit I am still very new to this stuff but when I do catkin_make after the other steps I get an error

```
[  6%] Linking CXX shared library /Users/emma/dev/robotics/ros_projects/catkin_ws/devel/lib/libgazebo_ros_velodyne_laser.dylib
Undefined symbols for architecture x86_64:
  "tbb::detail::r1::deallocate(tbb::detail::d1::small_object_pool&, void*, unsigned long, tbb::detail::d1::execution_data const&)", referenced from:
      void tbb::detail::d1::small_object_allocator::deallocate<tbb::detail::d1::graph_task>(tbb::detail::d1::graph_task*, tbb::detail::d1::execution_data const&) in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::initialize(tbb::detail::d1::task_arena_base&)", referenced from:
      tbb::detail::d1::task_arena::initialize()::'lambda'()::operator()() const in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::notify_waiters(unsigned long)", referenced from:
      tbb::detail::d1::wait_context::add_reference(long long) in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::cache_aligned_deallocate(void*)", referenced from:
      tbb::detail::d1::graph::~graph() in GazeboRosVelodyneLaser.cpp.o
      tbb::detail::d1::cache_aligned_allocator<tbb::detail::d1::graph_task*>::deallocate(tbb::detail::d1::graph_task**, unsigned long) in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::is_group_execution_cancelled(tbb::detail::d1::task_group_context&)", referenced from:
      tbb::detail::d1::task_group_context::is_group_execution_cancelled() in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::wait(tbb::detail::d1::wait_context&, tbb::detail::d1::task_group_context&)", referenced from:
      tbb::detail::d1::wait(tbb::detail::d1::wait_context&, tbb::detail::d1::task_group_context&) in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::reset(tbb::detail::d1::task_group_context&)", referenced from:
      tbb::detail::d1::task_group_context::reset() in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::destroy(tbb::detail::d1::task_group_context&)", referenced from:
      tbb::detail::d1::task_group_context::~task_group_context() in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::execute(tbb::detail::d1::task_arena_base&, tbb::detail::d1::delegate_base&)", referenced from:
      void tbb::detail::d1::task_arena::execute_impl<void, tbb::detail::d1::graph::wait_for_all()::'lambda'()::operator()() const::'lambda'()>(tbb::detail::d1::graph::wait_for_all()::'lambda'()::operator()() const::'lambda'()&) in GazeboRosVelodyneLaser.cpp.o
  "tbb::detail::r1::terminate(tbb::detail::d1::task_arena_base&)", referenced from:
      tbb::detail::d1::task_arena::terminate() in GazeboRosVelodyneLaser.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/emma/dev/robotics/ros_projects/catkin_ws/devel/lib/libgazebo_ros_velodyne_laser.dylib] Error 1
make[1]: *** [vehicle_sim/vehicle/sensor_model/velodyne/velodyne_gazebo_plugins/CMakeFiles/gazebo_ros_velodyne_laser.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
```

Tobias-Fischer, 2022-06-12:
Is that on OSX? What's the output of conda list?

traversaro, 2022-06-12:
This indeed seems similar to https://github.com/conda-forge/gazebo-feedstock/issues/121 . The output of `conda list` as mentioend by @Tobias-Fischer could help us to debug the problem more.

tohsin, 2022-06-12:
yeah osx sorry forgot to state that, the conda list output is
```
# packages in environment at /Users/emma/opt/miniconda3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.3                hbcca2e3_1    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.2            py39h701faf5_0    conda-forge
blosc                     1.21.1               h97e831e_3    conda-forge
boost                     1.74.0           py39ha1f3e3e_5    conda-forge
boost-cpp                 1.74.0               hdbf7018_7    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
bullet-cpp                3.21                 h428e3ee_1    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.5.18.1          h033912b_0    conda-forge
cairo                     1.16.0            he01c77b_1009    conda-forge
catkin_pkg                0.5.0              pyhd8ed1ab_0    conda-forge
certifi                   2022.5.18.1      py39h6e9494a_0    conda-forge
cffi                      1.15.0           py39he338e87_0    conda-forge
cfitsio                   4.1.0                h2c97ad1_0    conda-forge
cmake                     3.23.1               h35a7dd9_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
cppzmq                    4.8.1                hf09c4ce_0    conda-forge
cryptography              36.0.2           py39h1644bb1_1    conda-forge
curl                      7.83.1               h372c54d_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               h000cc95_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39h6e9494a_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
fcl                       0.7.0                h31ff308_0    conda-forge
ffmpeg                    4.3.2                hbf27d7b_3    conda-forge
flann                     1.9.1             h30321d8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.94              h10f422b_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.33.3           py39h701faf5_0    conda-forge
freeimage                 3.18.0               haf7e230_8    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
freexl                    1.0.6                h0d85af4_0    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.10.2              h08bfaa4_0    conda-forge
gdbm                      1.18                 h8a0c380_2    conda-forge
gdk-pixbuf                2.42.8               hb161b9c_0    conda-forge
geos                      3.10.2               he49afe7_0    conda-forge
geotiff                   1.7.1                had63758_1    conda-forge
gettext                   0.19.8.1          hd1a6beb_1008    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
gmock                     1.11.0               hbb4e6a2_0    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.6.13               h756fd2b_1    conda-forge
gpgme                     1.16.0               he49afe7_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  2.50.0               h2b5222e_2    conda-forge
gst-plugins-base          1.18.5               h63a84d5_3    conda-forge
gstreamer                 1.20.2               hdc08c3f_0    conda-forge
gtest                     1.11.0               hbb4e6a2_0    conda-forge
gtk2                      2.24.33              h7c1209e_2    conda-forge
gts                       0.7.6                hccb3bdf_2    conda-forge
harfbuzz                  3.4.0                h447b35c_0    conda-forge
[... 378 lines cut ...]
xorg-fixesproto           5.0               h0d85af4_1002    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.7.2                h0d85af4_0    conda-forge
xorg-libxau               1.0.9                h35c211d_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.11            h6c3fc93_1014    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

Tobias-Fischer, 2022-06-12:
@traversaro - if you have any ideas that would be great. It seems to be that the downstream packages still need to link to tbb directly?

traversaro, 2022-06-13:
> @traversaro - if you have any ideas that would be great. It seems to be that the downstream packages still need to link to tbb directly?

Yes, basically `tbb` is not exposed in Gazebo Classic's public headers, so now the downstream packages need to link `tbb::tbb` . This should not be a problem if the latest version of Gazebo is used and if the `${GAZEBO_LIBRARIES}` variable is used, but apparently there is something going on, let's check.

traversaro, 2022-06-13:
The user code seems fine: https://bitbucket.org/DataspeedInc/velodyne_simulator/src/9ab05d74438b693b8aad3c3c87ca438e68844409/velodyne_gazebo_plugins/CMakeLists.txt#lines-32 . What had a problem was instead gazebo 11.10.2, a problem that was fixed by https://github.com/osrf/gazebo/pull/3207 that was then released in gazebo 11.11.0 . 

In theory a fix is to run `mamba install gazebo=11.11` or create a new environment in which you can specify `gazebo=11.11`. Unfortunatly, I try to create an env with:
~~~
mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
~~~
but this does not work with error:
~~~
Looking for: ['ros-noetic-desktop', 'gazebo=11.11']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack/linux-64                                            No change
robostack/noarch                                              No change
robostack-experimental/linux-64                               No change
robostack-experimental/noarch                                 No change
Encountered problems while solving:
  - package gazebo-11.11.0-h84b5e5b_0 requires libgdal >=3.4.3,<3.5.0a0, but none of the providers can be installed
~~~

So I am afraid we need a noetic rebuild to update to Gazebo 11.11 and solve the problem? @Tobias-Fischer

traversaro, 2022-06-13:
An alternative may be to create a 10.12 branch in gazebo-feedstock and create a 10.12 build with the backport of https://github.com/osrf/gazebo/pull/3207 , but I am not sure if we are able to get correctly all the old dependencies, because not all of them are tracked in conda-forge-pinnings.

Tobias-Fischer, 2022-06-13:
Yeah agreed we need a rebuild but it's a bit of a pain (not as much as it used to be but still some manual work). If someone else wants to take the lead that'd be great - I won't get around it for quite a few months.

An alternative for this specific issue is to manually add the tbb::tbb linker line in the downstream package.

traversaro, 2022-06-13:
> An alternative for this specific issue is to manually add the tbb::tbb linker line in the downstream package.

Sure, I forgot that. Note that the exact target name is `TBB::tbb`, not `tbb::tbb` (that was exactly my error that was fixed by https://github.com/osrf/gazebo/pull/3207).

tohsin, 2022-06-13:
I wish I could help but I have no idea how any of this works😂. 
And I doubt its something can just learn without like years of experience😭

Tobias-Fischer, 2022-06-13:
In `/Users/emma/dev/robotics/ros_projects/catkin_ws/src/vehicle_sim/vehicle/sensor_model/velodyne/velodyne_gazebo_plugins/CMakeLists.txt` (or something like that, I probably got the path slightly wrong), find 
```
add_library(gazebo_ros_velodyne_laser src/GazeboRosVelodyneLaser.cpp)
target_link_libraries(gazebo_ros_velodyne_laser
  ${catkin_LIBRARIES}
  ${GAZEBO_LIBRARIES}
  RayPlugin
)

add_library(gazebo_ros_velodyne_gpu_laser src/GazeboRosVelodyneLaser.cpp)
target_link_libraries(gazebo_ros_velodyne_gpu_laser
  ${catkin_LIBRARIES}
  ${GAZEBO_LIBRARIES}
  GpuRayPlugin
)
```
and replace with
```
add_library(gazebo_ros_velodyne_laser src/GazeboRosVelodyneLaser.cpp)
target_link_libraries(gazebo_ros_velodyne_laser
  ${catkin_LIBRARIES}
  ${GAZEBO_LIBRARIES}
  TBB::tbb
  RayPlugin
)

add_library(gazebo_ros_velodyne_gpu_laser src/GazeboRosVelodyneLaser.cpp)
target_link_libraries(gazebo_ros_velodyne_gpu_laser
  ${catkin_LIBRARIES}
  ${GAZEBO_LIBRARIES}
  TBB::tbb
  GpuRayPlugin
)
```

Then run `catkin_make` again.

Tobias-Fischer, 2022-09-17:
Can this be closed @tohsin?

tohsin, 2022-09-17:
yh yh

closed 2022-09-17 by tohsin